letscontrolit/ESPEasy #2312: [Feature Request] Bring cmd mode into OLED Framed

Obicom, 2019-02-16:
First of all thank you very much for your effort in ESPEasy ...

It would be great if you could enable cmd mode (http/mqtt) into OLED Framed , comparable with OLED Device. I prefer the text size and scrolling from framed OLED but miss the option to send comand (on/off/clear) and text payloads via mqtt or http commands.  Please consider to implement. Thank you in advance.

Budman1758, 2019-02-16:
Currently supported commands are.
Oledframedcmd,on
Oledframedcmd,off
Oledframedcmd,low
Oledframedcmd,med
Oledframedcmd,high

Documentation on this is a bit behind.  May be other commands supported. Writing to certain lines I can attest is currently NOT supported.

Obicom, 2019-02-16:
Thank you for the explanation but most important part would be exactly the missing part ... send text to specific lines and clear command would be important too.

Budman1758, 2019-02-16:
> but most important part would be exactly the missing part .

Agreed

slawekb, 2020-07-21:
What I think would be very useful is a command to switch to next/previous frame - one could set up button(s) to quickly display data instead of waiting for the interval to pass

TD-er, 2020-07-22:
A bit like implemented in these two pull requests?
- https://github.com/letscontrolit/ESPEasy/pull/3050
- https://github.com/letscontrolit/ESPEasy/pull/3026

slawekb, 2020-07-22:
Yes looks like exactly that, thanks!

TD-er, 2020-07-22:
Can you maybe test the linked test build in PR #3050 ?
Since this is one of the most used plugins, I really want to make sure it doesn't break things before I merge it.

slawekb, 2020-07-22:
Tested on a Wemos D1 with BME280 and  SH1106 OLED and it looks to be working fine
Display works exactly the same as on 20200703 build, commands to switch to next and selecting a particular frame work.
I don't have a way to test the 'step through frames with button' at the moment unfortunately.

TD-er, 2020-07-22:
Well as long as the primary functionality has not been altered, I'm confident to merge it.
Will still try to have a look at its stack usage as that may be a bit too much.

PR #3050 (2020-07-23): [P036] Memory optimization (issue #2799) and other improvements
CHG: Memory optimization (issue #2799)
CHG: Added setting for 'Disable scrolling while WiFi is not connected',
         scrolling is not smooth as long as the ESP tries to connect Wifi (Load 100%!)
BUG: Wifi-Bars not fully right for 128 pixel wide displays
BUG: clearing window for scrolling lines was too high and to narrow
BUG: content of lines with pixel width > 255 had to be shortened
CHG: scrolling lines improved
CHG: variables renamed for better understanding

Fixes: #2312
Closes: #3026 
Fixes: #3127 
Fixes: #2760 
Fixes: #279 [... cut]
closed 2020-07-23 by TD-er