letscontrolit/ESPEasy #2331: Light sensor not configurable as described in WIKI

quorle, 2019-02-18:
Hello, I have the following light sensor:

[https://www.amazon.de/gp/product/B07DP1YM5X/ref=ppx_yo_dt_b_asin_title_o09__o00_s00?ie=UTF8&psc=1](url)

According to Wiki, you can configure this, but this only appears as I2C. My board does not have I2C, but outputs the values ​​as described in the wiki. What's going on here?

[https://www.letscontrolit.com/wiki/index.php?title=Photosensitive_resistor](url)

TD-er, 2019-02-18:
To me the sensor looks like one with a "1 bit" digital output and an analog output.
The analog output is (according to the wiki) 0 ... 3.3 V (probably not reaching 0 and not fully reaching 3.3V)
So you either must use an extra A/D converter, or use an extra resistor (or 2) to make this output in the range 0...1V and then connect it to the A0 analog input pin.

With the extra A/D converter as described in the wiki, you don't need the extra resistor(s).

On a NodeMCU/Wemos D1 you only need a single resistor, but on other boards you may need 2 (there is a voltage divider present on NodeMCU/Wemos d1).

quorle, 2019-02-18:
Ok, why only up to 1V? A Wemos z.b. Can the 0-5V without problems, as far as I know. Here you need then but no more resistance for the analog input. This would then be purely a matter of conversion?
Or am I completely wrong here?

TD-er, 2019-02-18:
Some modules already have a voltage divider, but the analog input of the ESP only allows up-to 1 V, which then results in an output value of 1023.
So put some voltage on it and see what output it will give, then you know the resistor-values used.
Just make sure you don't fry the analog input.

tonhuisman, 2022-04-18:
This seems to be resolved, so can be closed.

closed 2022-04-18 by TD-er