letscontrolit/ESPEasy #2333: no compile with arduino-ide of source and libs v20190216

ozett, 2019-02-18:
i used the fresh source and the libs in my portable arduino-ide.
but i seems not to find this lib.

i can weakly remember, that was a problem with the right place for lib-includes on the arduino-ide,
but maybe anybody with stronger brains can help me out?

[image]

ozett, 2019-02-18:
i made a fresh portable install and went from 1.6.1 to 1.8.8 with the arduino ide.
i put in all the libs from the espeasy source-file, but same error.

any hints, what to put where for including this lib?
error must be dependend on this new ESPEasySerial-thing...

[image]

ozett, 2019-02-18:
if i include the lib with the gui,
the #includes are on the ESPEasy.ino.
directly at the start.
also the #Espsoftwareserial-lib  is also included automatically.

than i disable the include in plugin_helper.ino,
the compile runs nearly to the end.

but differnent serial-lib errors. ...
anybody able to solve this riddle with this lib?

[image]

TD-er, 2019-02-18:
Where do you get this library `ESPeasySerial` from?
Is it the one included in our lib folder?

One of the things which are really different between PlatformIO and Arduino IDE is the way how includes work.
As it is now, it is working in PlatformIO, but I know Arduino IDE is rather picky on when to use `#include <...>` and when `#include "..."`
Also since I made that library myself and I am not aware of how the Arduino IDE handles libraries, it may very well be you either have some outdated version of the library, or there is something wrong with the includes.

clumsy-stefan, 2019-02-19:
@ozett try to do the following:
1) add a line `#include <ESPeasySerial.h>` either in ESPEasy-Globals.h or in your custom.h (if you're using it).
2) in the Arduino IDE library folder (Program folder), remove all libraries, that are also included in ESPEasy/lib. Then copy all libraries from ESPEasy/lib to the Arduine library folder
3) Recompile everything

I had a similar issue and it seems, that Arduino is keeping sometimes already compiled library objects and even if copying new sources over it, it does not recompile... So you need to clean everything first!

Hope that helps... At least my Arduino IDE compiles everything again after above steps...

2bontb, 2019-02-20:
See my notes on https://github.com/letscontrolit/ESPEasy/issues/2251
I had same problem compiling for SONOFF POW. I fixed two things to make it compile.
I reported also my changes to TD-er.

tonhuisman, 2023-09-03:
Compiling the ESPEasy code using Arduino IDE is not endorsed, better use VSCode/PlatformIO as [documented](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html).
This issue can be closed.

closed 2023-09-03 by TD-er