letscontrolit/ESPEasy #2334: Hardware serial not working with Senseair S8

hrikken, 2019-02-18:
Hardware serial appears not to work at all with the CO2 sensor S8 (Senseair). Only -1 values are being returned. Tested with softserial D6/7 and that works fine so far. Using 20190216 with BME680 plugin added.

TD-er, 2019-02-18:
Have you also tried a nightly build? Just to make sure it isn't a build issue. (see other issue where someone has issues building, which is related to the new serial wrapper library I made)

hrikken, 2019-02-19:
I use the ESP_Easy_mega-20190216_normal_ESP8266_1024 and added the BME680 (_P119_BME680). You can [download the complete source HERE.](https://drive.google.com/open?id=1o22sDzAb_aPcc5uBz2vM2navteboy0tB)

TD-er, 2019-02-19:
But did you also try to run a pre-built one? Just to make sure there is actually a bug and we're not trying to find a bug which is somehow a build error?

hrikken, 2019-02-19:
OK, this is what I've tested:
1 - Flashed a 4mb blank
2 - Flashed the pre-built ESP_Easy_mega-20190216_normal_ESP8266_4096.
3 - Disabled serial comms
4 - Configured Sensair device (double checked for 'Sensor = Carbon Dioxide'. 
5 - No further setup other than WiFi.
6 - Tests (swapped the RX/TX wires when no data cama in):
- Tested HWSerial D9/10 -> no data (always -1)
- Tested HWSerial D7/8 -> no data (always -1)
- Tested SWSerial D5/6 -> Working OK (short test)

hrikken, 2019-02-19:
Update: just tested the MH-Z19B: this one seems to be working ok with HWSerial (D9/10)!

TD-er, 2019-02-19:
OK, so it seems to be some issue in the plugin itself and not the serial wrapper I made.
Thanks for testing, this will make it easier to find the issue (I hope)

hrikken, 2019-02-25:
Tested latest nightly (20190225-normal) and hw serial is working with the S8. Top!!

TD-er, 2019-02-25:
Thanks for testing and letting me know.

closed 2019-02-25 by TD-er

hrikken, 2019-02-25:
That's the least I can do!