letscontrolit/ESPEasy #2336: ESP not working after OTA update

michass, 2019-02-19:
Hi, second time, first issue  with ESP_Easy_mega-20190215_test_core_260_alpha_ESP8266_4096,
next today with 20190215 242. After OTA update Sonoff don't work, only flash helps.
I have mounted my unit on electric box. I need disassembly wiring, unscrew unit from DIN rail and try to flash. Today I'm without ventilation for few hours.

TD-er, 2019-02-19:
You are aware of the "alpha" stage on the one you're flashing?
And please try first to just disconnect it (or maybe just power off the group) to help it to reboot. Maybe it is somehow stuck in some undefined state and a power-cycle is enough to help it reboot to a known state.
I've seen it happen sometimes an OTA was failing and after a reboot it was just rebooting the old FW.

michass, 2019-02-19:
My wife is at home, already switch relay off-on. Today I tried to use non "alpha" ROM.

TD-er, 2019-02-19:
It is better to test newer builds first on nodes which are easier to reach.

AxelMilb, 2019-02-19:
ESP_Easy_mega-20190216_test_core_242_ESP8266_4096.bin goes well over OTA but the ESP_Easy_mega-20190219_test_core_242_ESP8266_4096.bin Does not work.
Whether OTA or Direct via USB.

ESP Chip ID: | 1662812 (0x195F5C)

melwinek, 2019-02-19:
For me 20190219-normal-ESP8266-4096 not work.

AxelMilb, 2019-02-19:
test not normal

TD-er, 2019-02-19:
I tested 3 binary files and concluded there is indeed something wrong with it.
So I removed the zip file from download/releases list.

TD-er, 2019-02-24:
There were reports on a number of other projects which had similar issues.
It looks like there was some issue with updates in the toolchain, which also now has partially been reverted to an older version.

I have cleared the caches in Travis and building seems to work OK now.
If tomorrow's build is working fine, this issue can be closed. (please do test first on a well reachable node ;) )

closed 2019-10-10 by TD-er