letscontrolit/ESPEasy #2338: Reboot when node can not connect to AP

melwinek, 2019-02-19:
In one room I have a weak wifi coverage.
Is there any function in the code that triggers a reset on connection problems?

TD-er, 2019-02-19:
In the advanced settings you can change a few things.

For example you may set the WiFi to B/G mode, which will make the connection more stable when you have poor wifi reception.
Also there is this option:
- Connection Failure Threshold

This allows to perform a reboot after the set number of failed connection attempts. (default 0 = disable this feature)

melwinek, 2019-02-19:
Connection Failure Threshold is set to 0.
Since I uploaded mega-20190216 I have reboot every few minutes.
Uploading ESPEasy_mega-20190106, which previously worked well does not help.
B / G also does not help.
I temporarily connected the repeater.
I think only clearing the flash will help.

TD-er, 2019-02-19:
What core version do you use?
core 2.5.0 beta (and 2.6.0 alpha) are known to have some issues with wifi, or at least more than core 2.4.2

Also if possible, you can also try to rotate the ESP to maybe improve WiFi reception.
The antenna used on the ESP has quite a strong orientation preference.
On top of that, it matters if the orientation of the antenna in the access point matches that of the ESP.

For the PCB trace antenna, it is mainly sensitive in the direction where the trace points to. (starting at the 2 lines next to eachother)
For the "chip antenna" (rectangular ceramic antenna), it is mainly sensitive in a 'donut shape' around the antenna. So the antenna bar is like the center axis of the 'donut'.

Also reception may improve when not positioned on the ground and try to move a bit away from metal.

melwinek, 2019-02-19:
I use 2.4.2. My esp is in junction box glued to plexiglass

TD-er, 2019-02-19:
Well, after trying to rotate your house and settling with the neighbors, you can always try the wifi extender ;)

melwinek, 2019-02-20:
:) But it's bad that if you lose WIFI coverage, the node resets. It should normally work offline from time to time trying to connect to WIFI.

jimmys01, 2019-02-20:
This rule will reboot the ESP after 2 minutes that the MQTT is not reachable.

```
On System#Boot do
timerSet,2,120
endon

On MQTT#Connected do
timerSet,2,0
endon

On MQTT#Disconnected do
timerSet,2,120
endon

On Rules#Timer=2 do
Reboot
endon
```

melwinek, 2019-02-20:
@TD-er 
I do not understand anything.
After cleaning and re-uploading, the mega-20190216 works like a charm.
RSSI: -85 dB
Uptime: 0 days 15 hours 23 minutes
Connected: 7 m 35 s
Last Disconnect Reason: (200) Beacon timeout
Number reconnects: 54
Yesterday, with so many disconnections, it would restart every moment.
What is going on?

TD-er, 2019-02-20:
Sometimes I get the impression some left-over settings which are out of our reach, may affect wifi performance.
And the node I have mounted in my car (so rather flaky WiFi to say the least) has an uptime of over 4 days, so bad wifi is no guarantee for crashes.

ESP Easy | Information |
 -----|-----|
Build:⋄|20103  - Mega|
Libraries:⋄|ESP82xx Core 2.6.0-dev, NONOS SDK 3.0.0-dev(c0f7b44), LWIP: 2.1.2 PUYA support|
GIT version:⋄||
Plugins:⋄|76 [Normal] [Testing]|
Build time:⋄|Feb 13 2019 18:22:58|
Binary filename:⋄|ESP_Easy_mega-20190212-73-PR_2235_test_core_260_alpha_ESP8266_40|

melwinek, 2019-02-20:
Only in my case the version is the same, the physical conditions are the same. The difference is in cleaning the flash.

TD-er, 2019-02-20:
Yep, clearing the flash is a good one (I just suggested to do the same on another issue)

melwinek, 2019-02-21:
I do not see any problem to clear flash with every update. But only if I can do it through OTG without removing the ESP from the wall.

TD-er, 2019-02-21:
Clearing the flash will also clear the configuration (SPIFFS filesystem) so then you have to redefine the wifi settings for example.

PR #2559 (2019-08-26): [WiFi] Improve WiFi reconnect stability and fix initial setup
Simplified the WiFi connect state machine.
Fixed setup of initial WiFi configuration.

AP mode is now kept as long as client is connected.
AP mode will be started immediately if there are no valid WiFi credentials, or after 20 seconds of no wifi connection.
AP mode will be disabled 60 seconds after starting it, or after the last client disconnected from the AP.

Fixes #2561 
Fixes #2338 
Fixes #2203 
Fixes #1987 
Fixes #1737
Fixes #1640 
Fixes #1565
Fixes #1479 
Fixes #1098
Fixes #558
closed 2019-08-26 by TD-er