letscontrolit/ESPEasy #2339: IR Raw doesn't work and esp overload since 20190202

dyter, 2019-02-19:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

jimmys01, 2019-02-20:
Give us the raw command you want to run and I will test it.

dyter, 2019-02-20:
http://192.168.0.15/control?cmd=IRSEND,RAW2,^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9,38,60,60

http://192.168.0.15/control?cmd=IRSEND,RAW2,^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9,38,60,60

http://192.168.0.15/control?cmd=IRSEND,RAW2,R9F7CA696AD9*669D9D9696^A3Q9D9D96969D9*669D9D9696,38,90,68 |  
http://192.168.0.15/control?cmd=IRSEND,RAW2,Q969D96969D969696A776969D9D9696^ABQ969D9696AD8*669D9D9696,38,90,68
 http://192.168.0.15/control?cmd=IRSEND,RAW2,^15E9C9CID9CID*69CJDHC9D9^DP^16C9C9CID8DID9C9C8D9E7C9CICID9C9,38,62,51
http://192.168.0.15/control?cmd=IRSEND,RAW2,H99A494999499C7A4A494A499A894A8^9LH999495A794A8A8A*3494A8A8A499,38,138,67
50 | http://192.168.0.15/control?cmd=IRSEND,RAW2,LD5DAEADAE5DBCAE*45DAEAE5DA^DCLD5DBE9EAE5DADAE*45DAEAD5DA,38,112,47

The ESP go to overload or memory full, I do not know why

jimmys01, 2019-02-20:
`IRSEND,RAW2,Q969D96969D969696A776969D9D9696^ABQ969D9696AD8*669D9D9696,38,90,68` works fine. Also you cand use `IRSEND,SONY,240C `

`IRSEND,RAW2,LD5DAEADAE5DBCAE*45DAEAE5DA^DCLD5DBE9EAE5DADAE*45DAEAD5DA,38,112,47`
works fine. Also you cand use `IRSEND,SONY,3B0D`

`IRSEND,RAW2,H99A494999499C7A4A494A499A894A8^9LH999495A794A8A8A*3494A8A8A499,38,138,67 `works fine. the report from the IRrecvDumpV2 is 
```

Timestamp : 004944.783
Encoding  : UNKNOWN
Code      : EBF674B3 (32 bits)
Library   : v2.5.5

Raw Timing[63]:
   +  2420, -   506,    +  1280, -   638,    +   592, -   566,    +   564, -   594, 
   +  1226, -   626,    +   538, -   620,    +  1226, -   802,    +  1038, -   630, 
   +   590, -   638,    +   564, -   570,    +   590, -   660,    +   566, -   592, 
   +  1248, -   670,    +  1120, -   592,    +   566, -   658,    +  1140, - 20684, 
   +  2402, -   580,    +  1280, -   566,    +   564, -   594,    +   676, -   690, 
   +  1042, -   536,    +   564, -   660,    +  1118, -   660,    +  1116, -   660, 
   +   588, -   572,    +   588, -   570,    +   566, -   594,    +   590, -   636, 
   +  1118, -   658,    +  1120, -   660,    +   566, -   594,    +  1226

uint16_t rawData[63] = {2420, 506,  1280, 638,  592, 566,  564, 594,  1226, 626,  538, 620,  1226, 802,  1038, 630,  590, 638,  564, 570,  590, 660,  566, 592,  1248, 670,  1120, 592,  566, 658,  1140, 20684,  2402, 580,  1280, 566,  564, 594,  676, 690,  1042, 536,  564, 660,  1118, 660,  1116, 660,  588, 572,  588, 570,  566, 594,  590, 636,  1118, 658,  1120, 660,  566, 594,  1226};  // UNKNOWN EBF674B3
```

So perhaps there is another problem? Also keep in mind that i used MQTT to send the commands and not http, so perhaps the problem is there.

dyter, 2019-02-21:
_IRSEND,SONY,240C_ ... doesn't work for me since ESP IR exist

So when I send _http://192.168.0.15/control?cmd=IRSEND,RAW2,^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9,38,60,60_ power command on my Sony, in the log I have:

_HTTP: IRSEND,RAW2,^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9,38,60,60
126092: Invalid RAW2 B32 encoding!_

And the TSOP4838 IR crash ! I need to reboot the ESP

jimmys01, 2019-02-21:
try http://192.168.0.15/control?cmd=IRSEND,SONY,540C

dyter, 2019-02-21:
not work

TD-er, 2019-02-25:
Has this ever worked?
And if so, in what version was it working?

jimmys01, 2019-02-25:
My guess is that the HTTP ?cmd=etc is too many chars for the command to get through HTTP? I have tested MQTT up to 250 chars with no problems tomorrow I will also test he http method as well

TD-er, 2019-02-25:
As far as I know, the command parser has some limits on the number of characters on a single argument.

The "Low memory" message is also an indication some plugins may need some attention or it is simply using too many memory hungry plugins.

dyter, 2019-02-25:
Raw recognized or Raw transmit work fine with 20190121. And with no memory crash plugin

TD-er, 2019-02-25:
So it must have been this pull request: https://github.com/letscontrolit/ESPEasy/pull/2232

TD-er, 2019-02-25:
I've just done a ctrl-F on malloc and this is the first hit:
https://github.com/letscontrolit/ESPEasy/blob/fce9acdba30746f8e243a8abbeaac7ca559e47ba/src/_P035_IRTX.ino#L636-L651

I notice there is no check for the value of `count`, so this one can get out of bounds without noticing and that will cause a crash.
And that's just the first call to malloc.

jimmys01, 2019-02-25:
20190202  is the version I use on all my IR ( nodes, and the one I tested the raw commands posted above. Something else is going on. 
@TD-er the code that you mention is about Pronto devices, it is not called in this circumstances.

TD-er, 2019-02-25:
Well, even if it isn't the cause of these issues, it still is a potential crash.
And like I said, it was the first item I looked for.

Also when using large buffers, it is possible a malloc may fail and there should be a check for it (it will return a nullptr then).
Do you have any idea where in the code I should start looking? I don't know the code of the IR plugins and library.

uzi18, 2019-02-25:
@TD-er in case of error buf is not deleted

jimmys01, 2019-02-25:
buf is created at lines 128-131

buf is deleted at lines 244 245

the only check not done is for variable idx (created at line 126). It represents the pulses count and we have 250 as the limit (the size of buf)

uzi18, 2019-02-25:
@jimmys01 No, it is not deleted in 219 and 223

jimmys01, 2019-02-25:
Hmm OK I got it. So I assume this has to be addressed. Also there is another issue, that @dyter can't pass long commands through http. But I will test this tomorrow

TD-er, 2019-02-25:
Can both of you give some link also when discussing the code?
If you paste 'permalinks' to this repository here, it will be placed inline in the post.
That will make it much easier to read what you're talking about.

uzi18, 2019-02-25:
@TD-er #2357 is ready ;)

jimmys01, 2019-02-25:
@uzi18 idx is not checked in case the encoding is Raw it is only checked in case is RAW2. But since RAW2 works for 90+% of remotes not many will be using RAW1

uzi18, 2019-02-25:
@jimmys01 but we talking about problem with RAW2 ;)

dyter, 2019-02-26:
now with 20190226 no crash, but not work:

HTTP: IRSEND,RAW2,^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9,38,60,60
140739: Invalid RAW2 B32 encoding!

and TSOP4838 not reponding, need a reboot

uzi18, 2019-02-26:
@dyter please provide more log and what you receive in browser when you click on this link:
http://192.168.0.15/control?cmd=IRSEND,RAW2,^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9,38,60,60

dyter, 2019-02-27:
Browser say:

Base32Hex RAW Code: ^18BJB9BJB9BJD99*59BJBJB9B9^BD^19AJBBBHAABJB9BAA9B9AAA9BJBJAAA9
kHz: 38
Pulse Len: 60
Blank Len: 60

uzi18, 2019-02-27:
So it is not truncated string.
Ok thanks

jimmys01, 2019-02-27:
#2357 fixed his freeze issues and #2360 fixes the command not beeing decoded properly, my initial assumptions about the HTTP not passing the command was wrong.

PR #2360 (2019-02-28): [IRTX] Fix RAW2 bug introduced at 20190202
Fixes #2339
commit 88975f07 (2019-02-28): [IRTX] Fix RAW2 bug introduced at 20190202

Fixes #2339
closed 2019-02-28 by TD-er