letscontrolit/ESPEasy #2341: mqtt controller domoticz

megamarco833, 2019-02-21:
hi, my configuration is simple: name of esp: wemos_155
number of unit: 155

i set as controller mqtt domoticz and using standard configuration for controller subscive:
domoticz/out
and controller publish:
domoticz/in

if i send a command to esp it not work.

if i set as controller subscrive:
/%sysname%/#

and i send the comand:
sudo mosquitto_pub -t /wemos_155/cmd -m pwm,0,400,1900

the esp not react (this is for all commands)

but if i set as controller openhab and put as controller subscrive the same topic: 
/%sysname%/#

and i send again the message:
sudo mosquitto_pub -t /wemos_155/cmd -m pwm,0,400,1900

now it work
so, is it possible to correct this bug?
i need domoticz mqtt....
thanks

TD-er, 2019-02-21:
The controller is sending data from a plugin to some endpoint (e.g. MQTT)
There is also a plugin called "MQTT import", which can collect (numerical) data.

For Domoticz there are some exceptions, which are plugin specific.
For example the switch plugin can accept some commands via the controller from Domoticz.
But then you have to define a switch and connect that one to Domoticz.

Not sure about PWM, but maybe that's possible when you define the switch as a dimmer.

wdonker, 2019-02-22:
> Not sure about PWM, but maybe that's possible when you define the switch as a dimmer.

Correct, I use that in combination with MQTT Output Helper

megamarco833, 2019-02-24:
sorry i do not understand.
if i set as controller "open hab"  and set as controller subscrive:
/%sysname%/#
and i send a command:
`sudo mosquitto_pub -t /wemos_155/cmd -m pwm,0,400,1900`
it works perfectly
but if i send same command but with setting as controller domoticz mqtt and same controller subscrive as above the esp not respond, why?
it's not a bug?

at the end, if what way do you use mqtt output helper?

and what command i need to send for activate a gpio or pwm using shell with mqtt?

thanks so much

megamarco833, 2019-02-24:
> 
> 
> > Not sure about PWM, but maybe that's possible when you define the switch as a dimmer.
> 
> Correct, I use that in combination with MQTT Output Helper

can you please explain better how do you use mqtt output helper? thanks

megamarco833, 2019-02-24:
> The controller is sending data from a plugin to some endpoint (e.g. MQTT)
> There is also a plugin called "MQTT import", which can collect (numerical) data.
>
> For Domoticz there are some exceptions, [...]

sorry i do not understand.
if i set as controller "open hab" and set as controller subscrive:
/%sysname%/#
and i send a command:
`sudo mosquitto_pub -t /wemos_155/cmd -m pwm,0,400,1900`
it works perfectly
but if i send same command but with setting as controller domoticz mqtt and same controller subscrive as above the esp not respond, why?
it's not a bug?

at the end, if what way do you use mqtt output helper?

and what command i need to send for activate a gpio or pwm using shell with mqtt?

thanks so much

wdonker, 2019-02-24:
I have no access to my config right now but it is important that:
- you define a MQTT client gateway in Domoticz.
  This makes Domoticz subscribe to messages on topic domoticz/in and publish to domoticz/out.
- you must keep this topic names in EspEasy controller to make it work.
- you must set the IDX of the  MQTT Output Helper to the idx of the device in Domoticz. The helper 
   receives all messages from the MQTT-controller and filters on idx.

You can find more info in Domoticz wiki https://www.domoticz.com/wiki/MQTT#Domoticz_to_MQTT

TD-er, 2019-02-24:
@megamarco833 
If it is working fine in OpenHAB MQTT controller and not in Domoticz MQTT controller, then it is "inconsistent" to say the least.
But since there are so many exceptions in the code regarding Domoticz, I cannot yet say if it is a bug or intentional behavior.

As far as I know, it should only be possible to change a GPIO pin via a switch plugin.
And for the switch plugin, there are some exceptions in the controller code to handle incoming updates (via MQTT) from Domoticz. Maybe also from OpenHAB, but for that I am not sure.

The "MQTT Output Helper" is a plugin, also written mainly for use with Domoticz, but I never used it myself.

Like I said, the Domoticz integration is not really well structured and there are way too many loop-holes made in the past to make it work with Domoticz.
Also OpenHab's controller does look a lot more elaborate than the other controllers.
This also means that it is very well possible that things may work differently depending on the MQTT controller used.

megamarco833, 2019-02-25:
> @TD-er 
> If it is working fine in OpenHAB MQTT controller and not in Domoticz MQTT controller, then it is "inconsistent" to say the least.
> But since there are so many exceptions in the code regarding Domoticz, I cannot yet say if it is a bug or intentional behavior.

yes, the same controller subscrive:
`/%sysname%/#`
is working if setted in openhab and not working if setted in domoticz controller.
So i think that is inconsistent...
could be investigate by some one and "correct"?

> Like I said, the Domoticz integration is not really well structured and there are way too many loop-holes made in the past to make it work with Domoticz.

> This also means that it is very well possible that things may work differently depending on the MQTT controller used.
yes, could be improved for the next release?
for light for example is very important to have an integration with mqtt and domoticz :-)
thanks

wdonker, 2019-02-25:
I think the difference between OpenHAB and Domoticz is that Domoticz publishes ALL changes for devices to 1 topic so a sysname as topic will never work for more than 1 ESP-node.
Again, for me the integration with Domoticz and MQTT works well enough as it is.

megamarco833, 2019-02-25:
> 
> I think the difference between OpenHAB and Domoticz is that Domoticz publishes ALL changes for devices to 1 topic so a sysname as topic will never work for more than 1 ESP-node.
> Again, for me the integration with Domoticz and MQTT works well enough as it is.

Ok but in this case if I want to set pwm of gpio 0 at value of 800 in 1,5second what command I have to send?
Now I use :
sudo mosquitto_pub -t /wemos_155/cmd -m pwm,0,800,1500
where my topic is the esp name: wemos_155
if I still using as topi subscrive= domoticz/in
this command is not working, so what command I should use?
Thanks

wdonker, 2019-02-25:
You do not use a command; the Domoticz MQTT Client gateway publishes the event for you and the MQTT Output helper captures it. But you cannot use the 1,5 second because Domoticz does not support that.
Please read my previous comments and the Domoticz wiki!

wdonker, 2019-02-25:
For pwm you need to define a dimmer in Domoticz and set the idx and gpio in the MQTT output helper.
If you than set the dimmer percentage in Domoticz it will be translated into pwm level by the output helper.

megamarco833, 2019-02-26:
> For pwm you need to define a dimmer in Domoticz and set the idx and gpio in the MQTT output helper.
> If you than set the dimmer percentage in Domoticz it will be translated into pwm level by the output helper.
it seams that is not workin.
i create a task with Output - Domoticz MQTT Helper
i set here the GPIO 0
the idx 36 
in domoticz i set idx36 as dimmer switch

now if i move the slider in domoticz i have no response on gpio0

on the espeasy i set as controller domoticz and as topic: 
Subscribe: = domoticz/out
Publish: = domoticz/in

wdonker, 2019-02-26:
And you also defined the MQTT Client Gateway?

megamarco833, 2019-02-26:
yes, sure. 
in domoticz i set up the mqtt
and also in espeasy as i wrote
mqtt brocker is working because if i send message as in the first topic is working

wdonker, 2019-02-28:
Ok, so here is a few screenshots of my working config:

The MQTT client gateway in Domoticz:
[image]

The MQTT Controller in ESPEasy:
[image]

The helper device in ESPEasy:
[image]

The log when I set the dimmer to 48% in Domoticz:
[image]

Hope this is of any use. By the way, I am using ESPEasy mega-20190202.

megamarco833, 2019-02-28:
thanks, that's mine:
[image]

wdonker, 2019-03-01:
I see that you have 3 controllers defeined whereas I have only 1. If I also add 2 it doesn't work anymore so it seems that the Domoticz MQTT controller must by the first controller in the setup to make it work.
This may be a bug.

TD-er, 2019-03-01:
There can be only 1 MQTT controller active.
Even if you have >1, only the first active one will be used.

This is a known issue and a work in progress.
I thought I had once added a warning to appear on screen when you save these settings that only one active MQTT controller is currently supported.

megamarco833, 2019-03-01:
> I thought I had once added a warning to appear on screen when you save these settings that only one active MQTT controller is currently supported.

yes but only one mqtt controller is actived if you see the image.... :-)

wdonker, 2019-03-01:
Maybe the bug is in P029.ino (the Output Helper).
In Line 45 it is looking up a MQTT controller without checking if it is Domoticz or enabled...
````        // We need the index of the controller we are: 0-CONTROLLER_MAX
        byte controllerNr = 0;
          for (byte i=0; i < CONTROLLER_MAX; i++)
          {
//            if (Settings.Protocol[i] == CPLUGIN_ID_002) { controllerNr = i; }   -> error: 'CPLUGIN_ID_002' was not declared in this scope
            if (Settings.Protocol[i] == 2) { controllerNr = i; }
          }

TD-er, 2019-03-01:
Controller ID 2 is Domoticz MQTT.
But I agree it is not checking there to see if it is enabled.

wdonker, 2019-03-01:
You are right, I was comparing to C002.ino which uses:
`byte ControllerID = findFirstEnabledControllerWithId(CPLUGIN_ID_002);`

TD-er, 2019-03-01:
To have a bit more context here:
https://github.com/letscontrolit/ESPEasy/blob/08d0602d81e9804bf41e1a36aaff8434b7797f73/src/_P029_Output.ino#L43-L59
It is in the setup phase on the web page, so there does not need to be a check for the controller to be enabled.

And this is the part in the Domoticz MQTT controller where it is being used.
https://github.com/letscontrolit/ESPEasy/blob/e964d859870cd52741a7ddce403547d1f96b1d94/src/_C002.ino#L91-L124

wdonker, 2019-03-01:
Yes, but still..
The setup page of P029 seems to make a connection between the controller index and the device (don't know why because it is an 'input plugin' and there can be only 1 MQTT controller). So when 2 controllers are defined it may link to the wrong controller and not work. At least, when I add another Domoticz MQTT controller and disable the first the device webpage does not show the IDX anymore (probably because it takes the value from the 2nd controller defined). Remove the 2nd controllere and the IDX is ok again.

But as long as only 1 MQTT controller is defined everything works ok so I do not think it is a big deal right now.

TD-er, 2019-03-01:
The IDX values you set in the plugin settings and you set those per controller you assign to that plugin.
The index of the controller used for this 'output' plugin is the slot of controller definitions, just like those black inverted 1, 2, 3 values you see in the plugin settings.
If you set the IDX value the same for all controller fields for your plugin, then the IDX value should remain the same.

Or am I missing something here?

wdonker, 2019-03-01:
That is the general concept for a plugin that sends data to one or more controllers but in this case the plugin (or in fact the controller itself) **receives** data. And since there can only be one MQTT controller active it would not be necessary to make a link to the controller and store the IDX with the device. Than changing controller definitions would not break the link. But I can understand this is how the setup is structured.
P029 and C029 are quite intermingled anyway...

wdonker, 2019-03-01:
Addition: even when I add a second, active Domoticz MQTT controller the device webpage shows only 1 IDX input box.

pedromes, 2019-03-02:
I had the same problem, the solution I found was: 

_C002.ino, line 95

if (strcasecmp_P(switchtype, PSTR("Dimmer")) == 0) 

_C002.ino, lines 110 to 113

 case 2:
                          pwmValue = map(atol(svalue1), 0, 100, 0, 1023);
                          UserVar[baseVar] = pwmValue;
                          break;

Sorry for the formatting.

pedromes, 2019-03-02:
@wdonker 

I had the same problem, the solution I found was:

_C002.ino, line 95

if (strcasecmp_P(switchtype, PSTR("Dimmer")) == 0)

_C002.ino, lines 110 to 113

case 2:
pwmValue = map(atol(svalue1), 0, 100, 0, 1023);
UserVar[baseVar] = pwmValue;
break;

Sorry for the formatting.

TD-er, 2019-03-02:
I must look into it when I get back home this evening.
Please help me remind

wdonker, 2019-03-02:
The second change pedromes suggests is a better way to calculate PWM from Domoticz percentage; it uses PWM-values 0..1023 instead of current 0..1000.
However, it does not solve the problem of multiple controllers.

And I think strcasecmp_P ignores case so it makes no difference comparing to ''Dimmer' or 'dimmer'.

pedromes, 2019-03-02:
> And I think strcasecmp_P ignores case so it makes no difference comparing to ''Dimmer' or 'dimmer'.

In my case the change from "dimmer" to "Dimmer" was what made the EspEasy receive the command of Domoticz.

> The second change pedromes suggests is a better way to calculate PWM from Domoticz percentage; it uses PWM-values 0..1023 instead of current 0..1000.

As for the change from 0 ... 1000 to 0 ... 1023 was to test with the onboard LED on the Wemos D1 mini that uses inverted logic, when selecting 100 in the Domoticz LED did not turn off, getting turned on with low brightness.

pedromes, 2019-03-02:
[image]

pedromes, 2019-03-02:
> And I think strcasecmp_P ignores case so it makes no difference comparing to ''Dimmer' or 'dimmer'.

I did the test here and it worked with "dimmer", but at the time I was setting up I only succeeded after switching to "Dimmer".

megamarco833, 2019-03-04:
> 
> 
> I see that you have 3 controllers defeined whereas I have only 1. If I also add 2 it doesn't work anymore so it seems that the Domoticz MQTT controller must by the first controller in the setup to make it work.
> This may be a bug.

i set only one controller
[image]

set again the idx of my dimmer in domoticz
[image]

[image]
[image]

but as you can see no change status of my dimmer inside espeasy:
[image]

PR #5059 (2024-05-29): [Controller] Fail gracefully when almost out of memory
One often reported issue is that the ESPEasy node (especially ESP8266 ones) will crash when WiFi is unstable.

This could be due to memory suddenly filling up by the controller queue. When creating a new item for the controller queue, there was no `(std::nothrow)` included on the `new` calls, so if running out of memory this might have thrown a `std::bad_alloc` exception, which are not caught and thus ESPEasy might crash on it.

Fixes: #4726
Fixes: #3649
Fixes: #2341

Before merge, see: https://github.com/letscontrolit/ESPEasy/pull/4770
PR #4770 (2024-12-30): [MQTT] Controller improvements
Resolves #2341 
Resolves #4864 
Resolves #2560 
Resolves #4459 

Affected controllers:
- [C002] Domoticz MQTT
- [C005] Home Assistant (openHAB) MQTT
- [C006] PiDome MQTT
- [C014] Homie MQTT convention (V3/V4)
- [C007] Emoncms

Features:
- MQTT Controllers: generic support for `/cmd` and `/set` topics where applicable
- String/resource usage reductions (most of that in Homie controller)
- Added generic `CPlugin_helper_mqtt` module with MQTT support functions
- [C004] Small code improvements, add documentation and examples
- [C007] Add configurable url, stored in ` [... cut]
closed 2024-05-29 by TD-er