letscontrolit/ESPEasy #2343: Serial Server is not working anymore

DittelHome, 2019-02-21:
I am using ESP_Easy_mega-20190216_normal_ESP8266_4096.bin with serial plugin (Communication - Serial Server). I have connected a serial device called HMUART (for smarthome FHEM),using RX/TX. With the new version i cannot connect to the serial device. No i am back to ESP_Easy_mega-20181120_normal_ESP8266_4096.bin and all is working fine. The GUI of the ESP8266 is reachable. I have already done a cold boot.

TD-er, 2019-02-21:
The serial port wrapper was added around 20190101, so I guess any build made until 20181231 will behave like the 20181120 build you're now running.

I will have a look at it.
(N.B. I have created numerous test build while changing these and no reports regarding this plugin were reported)

DittelHome, 2019-02-25:
Today, i have done a test again..
```
17402: Ser2N: S>: �
17424: MQTT : Intentional reconnect
17462: MQTT : Connected to broker with client ID: HMUART2_6
17463: Subscribed to: /FHEM/#
18026: DHT  : Temperature: 17.10
18026: DHT  : Humidity: 26.50
33234: Ser2N: S>: �
33259: WD   : Uptime 1 ConnectFailures 0 FreeMem 16048 WiFiStatus 3
33260: LoopStats: shortestLoop: 92 longestLoop: 3121249 avgLoopDuration: 154.21 loopCounterMax: 326086 loopCounterLast: 189993 countFi
33261: Scheduler stats: (called/tasks/max_length/idle%) 189990/1409/8/79.10
41537: Ser2N: S>: �
54366: Ser2N: S>: �
62839: Ser2N: S>: �
62862: WD   : Uptime 1 ConnectFailures 0 FreeMem 15952 WiFiStatus 3
62922: LoopStats: shortestLoop: 92 longestLoop: 3277133 avgLoopDuration: 158.29 loopCounterMax: 326086 loopCounterLast: 181410 countFi
62923: Scheduler stats: (called/tasks/max_length/idle%) 181409/1342/8/93.60
92115: Ser2N: S>: �
92144: WD   : Uptime 2 ConnectFailures 0 FreeMem 15808 WiFiStatus 3
92922: LoopStats: shortestLoop: 92 longestLoop: 3616244 avgLoopDuration: 120.48 loopCounterMax: 326086 loopCounterLast: 238599 countFi
92923: Scheduler stats: (called/tasks/max_length/idle%) 238598/1744/8/79.90
108775: MQTT : Connection lost, state: Connection lost
108819: MQTT : Connected to broker with client ID: HMUART2_6
108820: Subscribed to: /FHEM/#
109381: DHT  : Temperature: 17.10
109381: DHT  : Humidity: 26.50
117231: Ser2N: S>: �
122034: WD   : Uptime 2 ConnectFailures 2 FreeMem 17456 WiFiStatus 3
122922: LoopStats: shortestLoop: 92 longestLoop: 6905513 avgLoopDuration: 162.68 loopCounterMax: 326086 loopCounterLast: 178504 countFi
122923: Scheduler stats: (called/tasks/max_length/idle%) 178504/1324/8/93.50
127110: Ser2N: S>: �
134312: Ser2N: S>: �
139459: Ser2N: S>: �
152034: WD   : Uptime 3 ConnectFailures 2 FreeMem 17376 WiFiStatus 3
152922: LoopStats: shortestLoop: 92 longestLoop: 6905513 avgLoopDuration: 160.99 loopCounterMax: 326086 loopCounterLast: 180115 countFi
152923: Scheduler stats: (called/tasks/max_length/idle%) 180115/1328/8/94.00
155937: HTML : Request file /index.htm.gz
155965: HTML : Request file /index.htm.gz
155966: HTML : Request file /index.htm
155994: HTML : Request file /index.htm
160913: Ser2N: S>: �
184127: Ser2N: S>: �
184151: WD   : Uptime 3 ConnectFailures 2 FreeMem 15136 WiFiStatus 3
184153: LoopStats: shortestLoop: 92 longestLoop: 6905513 avgLoopDuration: 140.06 loopCounterMax: 326086 loopCounterLast: 214741 countFi
184154: Scheduler stats: (called/tasks/max_length/idle%) 214739/1596/8/82.90

```

Here is the output with the working Version from 2018/11/20:

```
37801: Ser2N: N>: �
41359: Ser2N: S>: �
41377: Ser2N: N>: �
45518: Ser2N: S>: �
45541: Ser2N: Client disconnected!
46140: Ser2N: Client connected!
47141: Ser2N: N>: �
50230: Ser2N: S>: �
50249: Ser2N: N>: �
53343: Ser2N: S>: �
53358: MQTT : Connection lost, state: Connection lost
53398: MQTT : Connected to broker with client ID: HMUART2_6
53399: Subscribed to: /FHEM/#
53410: Ser2N: N>: �
53414: WD   : Uptime 1 ConnectFailures 2 FreeMem 17216
53415: LoopStats: shortestLoop: 77 longestLoop: 4143656 avgLoopDuration: 1424.11 loopCounterMax: 389610 loopCounterLast: 21966 countFi
53415: Scheduler stats: (called/tasks/max_length/idle%) 23552/363/8/21.00
56940: Ser2N: S>: �
56966: Ser2N: N>: �
61090: Ser2N: S>: �
61134: DHT  : Temperature: 17.00
61134: DHT  : Humidity: 26.60
61174: Ser2N: Client disconnected!
61676: Ser2N: Client connected!
62975: Ser2N: N>: �
66069: Ser2N: S>: �
66089: Ser2N: N>: �
69199: Ser2N: S>: �
69454: Ser2N: N>: �
74415: Ser2N: S>: �
74434: Ser2N: N>: �
78580: Ser2N: S>: �
78818: MQTT : Connection lost, state: Connection lost
78860: MQTT : Connected to broker with client ID: HMUART2_6
78861: Subscribed to: /FHEM/#
78889: Ser2N: Client connected!
79422: DHT  : Temperature: 17.00
79422: DHT  : Humidity: 26.50
79889: Ser2N: N>: �
83002: Ser2N: S>: �
83025: Ser2N: N>: �
86149: Ser2N: S>: �
86170: WD   : Uptime 1 ConnectFailures 4 FreeMem 11496
86171: LoopStats: shortestLoop: 77 longestLoop: 4539104 avgLoopDuration: 969.38 loopCounterMax: 389610 loopCounterLast: 33482 countFin
86171: Scheduler stats: (called/tasks/max_length/idle%) 33474/268/8/33.10
86175: Ser2N: N>: �
89730: Ser2N: S>: �
89754: Ser2N: N>: �
93874: Ser2N: S>: �
93896: Ser2N: Client disconnected!
94101: SaveToFile: free stack: 3200
94134: FILE : Saved config.dat
94135: SaveToFile: free stack after: 3200
94149: SaveToFile: free stack: 3264
94180: FILE : Saved config.dat
94180: SaveToFile: free stack after: 3264
94196: SaveToFile: free stack: 3216
94263: FILE : Saved config.dat
94263: SaveToFile: free stack after: 3216
97129:      : Rebooting...
7639: Ser2N: Client connected!
8639: Ser2N: N>: �
8649: Ser2N: S>: �
8739: Ser2N: N>: �
9171: Ser2N: S>: �
9178: Ser2N: S>: 
9187: Ser2N: S>: �
10240: Ser2N: N>: �
10253: Ser2N: S>: �
10339: Ser2N: N>: �
10355: Ser2N: S>: �
10370: Ser2N: S>: ���
10439: Ser2N: N>: �
10454: Ser2N: S>: �
10470: Ser2N: S>: Y
10540: Ser2N: N>: �
10555: Ser2N: S>: �
10639: Ser2N: N>: �
10654: Ser2N: S>: �
10739: Ser2N: N>: �
10753: Ser2N: S>: �
10840: Ser2N: N>: �
10855: Ser2N: S>: �
10871: Ser2N: S>: 01|�
10939: Ser2N: N>: �
10955: Ser2N: S>: �
11039: Ser2N: N>: �
11053: Ser2N: S>: �
11139: Ser2N: N>: �
11154: Ser2N: S>: �
```

Hopefully it helps ..
regards Klaus

uzi18, 2019-02-25:
@TD-er @DittelHome will prepare fix for this one :)

TD-er, 2019-02-25:
@uzi18 Hmm that's a confusing line. ;)
Do you mean DittelHome will make a fix, or will you make a fix?

uzi18, 2019-02-25:
@TD-er ok if you want - both serial helper and p20 uses CONFIGPIN1, we can move to different pin eg. 3 ?

TD-er, 2019-02-25:
I have a look before I answer that ;)

uzi18, 2019-02-25:
or move plugin helper pins to 2 and 3

TD-er, 2019-02-25:
All other plugins use that, so then we have to change settings.
I guess we can make some kind of transition in the settings, either dynamic or only to be made once to put all plugins using serial to have the same pins defined for Rx and Tx pin

uzi18, 2019-02-25:
I'm not sure if it is needed, if recent build used it it needed to change configuration options on this plugin.
So it is better just to move along with pin to 3 and if any problem point to check configuration.

You can do something like that if cfg_version change, to have clean transition.
As You can see only one report about Ser2Net not working from 3-4 months.

DittelHome, 2019-02-25:
> As You can see only one report about Ser2Net not working from 3-4 months.

And the rest is still waiting, i believe...

uzi18, 2019-02-25:
I mean more like, if someone is forced to configure serial port one more time, can also configure pin3

TD-er, 2019-02-25:
I am now looking at P020_Ser2Net and P044_P1WifiGateway.
Both still use the normal `Serial` they always used and not the serial wrapper I recently made.

Only thing changed is the way they configure the serial port settings.
This is a set of parameters like stopbits, parity, databits etc.
See these functions:
- `serialHelper_convertOldSerialConfig` 
- `serialHelper_serialconfig_webformLoad`
- `serialHelper_serialconfig_webformSave`

DittelHome, 2019-02-25:
Which means ?
I assume, it should work .. ;(
Can a test something ?

DittelHome, 2019-02-25:
I have done done some more tests. It worked for short time period, after deletion and adding the serial server again. After restart, again no serial connection. strange thing

TD-er, 2019-02-25:
OK, that does make sense to me :)

TD-er, 2019-02-25:
Any idea (or logs) how long it is working?

DittelHome, 2019-02-25:
The load keeps very high:
Load: | 100.00% (LC=0)
bad response times ...

after disabling the serial device, the load goes under 10 %

If i get serial connection, it keeps active. It is just a matter of luck to get a connection

TD-er, 2019-02-25:
I've been looking into the different revisions and it is at least not a very obvious mistake.
There has been a single change between the version you tried and the big changes I made recently.
Can you try the 20181231 build?
Just to make sure that single change inbetween doesn't affect this plugin.

DittelHome, 2019-02-25:
> Can you try the 20181231 build?

tried this one: ESP_Easy_mega-20181231_normal_ESP8266_4096.bin
Flashed => working , restarted => working

TD-er, 2019-02-25:
Thanks.
And just to make sure, you did disable logging and the "Serial port" setting in the advanced settings?

DittelHome, 2019-02-25:
[image]

TD-er, 2019-02-25:
Can you disable it as a test? (in both the working and the non working version)

DittelHome, 2019-02-25:
Disabled serial port:
2018/12/31 not working, work with port enabled
2019/02/25 not working

TD-er, 2019-02-25:
Thanks :)

Maybe it will be useful for me to have a test setup here.
Can you describe your setup, so I can replicate it here?
I never used that plugin myself.

As a test, I will make a build too using the sources of that plugin from 20181231. Just to make sure the bug is caused by the plugin code and not something else.
What build do you need? normal 4MB?

DittelHome, 2019-02-25:
4MB

TD-er, 2019-02-25:
Can you test [this build](https://www.dropbox.com/s/xfjso0fpbjieib1/testbuild_issue_2343.rar?dl=0) ?

DittelHome, 2019-02-25:
Ok, test build is working !!

TD-er, 2019-02-25:
Great, then for now, I will merge these two reverts and add a reference in the code to help finding it when I later will try to not break it again :)

TD-er, 2019-02-25:
So for now I will close this one and later we may look into it.
It will be included in tomorrow's build.

closed 2019-02-25 by TD-er