letscontrolit/ESPEasy #2344: I2C generic plugin required

rudiratlos, 2019-02-22:
Hi,
I have an I2C sensor (Pressure and Temperature) AMS5915 ( https://www.amsys.info/products/ams5915.htm ) and havent't found a specific plugin for it. 
I'm looking for a generic plugin, which just do a raw read of <somebytes e.g. 4> from a given <I2addr> and send the raw hex output string via mqtt. This would support a wide range of i2c sensors, which do not have a specific plugin.

TD-er, 2019-02-22:
There is a class for handling some of the basic I2C calls, which is also being used in the BME280.ino plugin.
It may be great for rapid prototyping to have such a generic I2C plugin, but I am not sure how complex the UI will end up to be, since most sensors still need some quite specific actions or delays.

In general you have:
- init procedure
- start reading procedure
- collect data
- end reading procedure (not often used)

Also the error handling may be quite specific for each sensor and some procedures need a delay (or call back later) before or after and this delay may also be settings specific.
Some allow to read some register to see if a new value is ready and some other just have a fixed delay.

It can very easy end up to have some counter intuitive interface not even the author may know how to use it.

ccapublic, 2019-03-27:
Yes it would actually be great to expose via mqtt/websservice/whatever something like readi2c(i2caddress, memaddress, count) and writei2c(i2caddress, memaddress, data[]). A lot of simple sensors would work, not to say "custom made" devices, like my arduino implementing i2c slave and driving my spa filter and heater...

rudiratlos, 2019-04-05:
whould be really nice to have such a generic i2c plugin. 
sensor is on i2c addr 0x07 and read 2 bytes (msb, lsb). 
send these 2 bytes via mqtt and an additional string with 'NG' or 'LQ' (webgui selectable)
I would spend a bottle of champagne.

uzi18, 2019-04-05:
It is possible but we need to think how to expose these data to controller

rudiratlos, 2019-04-05:
what do you mean with controller? esp8266 or mqtt controller/server?
I think about a device plugin website like bme280 sensor which have more than one value (baro, humidity ...). My sensor has 2 rawvalue bytes (msb,lsb) and an additional string with 2 characters which will characterize the sensor type (gas or liquids):

mqtt content:
rawval: <msb><lsb>
type: NG or LQ or XX ....

To have a generic plugin for a a lot of sensors, the plugin webpage should have a field, where I casn type in the i2c hex-addr e.g. 0x07
a second field readnumbytes, where I can select how many bytes (1-6) should be read from i2c hex addr.
And lastly a third field where I can give a 2 char string (NG,LQ,XX...)

uzi18, 2019-04-05:
Yes because controller need to. know how to send string data =)

rudiratlos, 2019-04-05:
here is a link to the datasheet of the FS2012 (NG stand for gas sensor, LQ for liquids):
https://www.idt.com/products/sensor-products/flow-sensors/fs2012-gas-or-liquid-flow-sensor-module-calibration

the second sensor is this (4 databytes):
I2C-addr: 0x70 - 0x77
https://www.amsys.de/downloads/data/ams5915-AMSYS-datasheet.pdf

rudiratlos, 2019-04-05:
if string data is not supported, just use the e.g 2 bytes of ascii code 0x4e 0x47 (NG)

uzi18, 2019-04-05:
@rudiratlos got this one: HPSD3002-027M hybysens

rudiratlos, 2019-04-05:
1.st prio for me is the FS-2012

TD-er, 2019-04-05:
A plugin does deliver either upto 4 floats or 1 unsigned long (for example RFid readers do)
That's what @uzi18 meant with the output format.
We currently do not support string outputs, but you may give a variable a name and use that in the formatting of the data sent by a controller.

Apart from that there still isn't a "generic I2C plugin"

rudiratlos, 2019-04-05:
I think for a generic i2c plugin 1 unsigned long for the raw data whould be the best choice.
for the type info (NG, LQ) which is a static information a variable is a good idea. It's just for the MQTT server, how to handle the raw data. 

Apart from that there still isn't a "generic I2C plugin"
I would spent a bottle of champagne for the developer, who is implementing this as open source.

rudiratlos, 2019-04-09:
it's easier than I thought. we could use a formula (e.g. %value%/10) to get a meassurement out of the 2 raw data bytes <msb><lsb> (NG and LQ is not needed at all for calcualation. %value% should just contain a unsigned word or long.
Is someone willing to program a generic i2c module for a good bootle of champagne?

rudiratlos, 2019-04-10:
good idea, but how to access the i2c device ?
I've created a device 'DummyI2C' with value1. in rules I have:
on Rules#Timer=1 do
DummyI2C#dummy=3
  timerSet,1,60
endon

TD-er, 2019-04-10:
@rudiratlos These commands suggested by @uzi18 are not yet implemented.
It was just a (good) suggestion.

uzi18, 2019-04-10:
we can feed variables v1-v16 with data from i2c device with these commands
next you can use it with taskvalueset to push it into dummy and send wherever you want

uzi18, 2019-04-10:
can try to prepare something like it to test for you

rudiratlos, 2019-04-10:
hi uzi18, that whould be great to have a generic i2c plugin:
pulldown menu (i2cadr): 0-7f
pulldown menu (bytes to read): 1-4
intervall field
value / formula / decimal field

uzi18, 2019-04-10:
first i will try to create new command for i2c

rudiratlos, 2019-04-12:
ok

uzi18, 2019-04-12:
@rudiratlos in facts plugin idea is more simple to use, but with command it will be more elastic

what kind of ESPEasy firmware file you use?

rudiratlos, 2019-04-12:
Git Build: mega-20190315
I like to have a plugin, pls.

uzi18, 2019-04-12:
show me your system info from your esp module webserver

rudiratlos, 2019-04-12:
Firmware
--
Build:⋄ | 20103  - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20190315
Plugins:⋄ | 77 [Normal] [Testing] [Development]
Build Md5: | 8a634edb71d6a875645851fb62f4b6f3
Md5 check: | passed.
Build Time:⋄ | Mar 15 2019 03:29:42
Binary Filename:⋄ | ESP_Easy_mega-20190315_dev_ESP8266_4M.bin

uzi18, 2019-04-12:
this is it, thanks

uzi18, 2019-04-12:
@rudiratlos it is WIP, waiting for your feedback - need INFO log

[image]

[firmware.bin.zip](https://github.com/letscontrolit/ESPEasy/files/3075305/firmware.bin.zip)

rudiratlos, 2019-04-13:
will test asap

uzi18, 2019-04-14:
ok I need from you log level "info" logs, so i can check if all works fine

will add some commands for i2c to this plugin later

rudiratlos, 2019-04-15:
[image]

rudiratlos, 2019-04-15:
excerpt info log:
Build:⋄ | 20103  - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ |  
Plugins:⋄ | 80 [Normal] [Testing] [Development]
Build Time:⋄ | Apr 12 2019 22:01:45
Binary Filename:⋄ | Self built!

I have hook up 3 i2c sensors. a bme280 (out of the bix plugin) -> works
a FS2012 flowsensor, which gives a value of 0 (2byte read)
and a pressure sensor AMS5915 which gives different values (2Byte read 0x6a86 - 0x6F06).
I will check these values if theese make sense. I believe that the MSB has to be swapped with the LSB.

uzi18, 2019-04-15:
ok, but please set in advenced settings weblog to info level, next go to settings/log window and try to catch as many lines as possible (5-10 minutes), copy and paste here for me.

rudiratlos, 2019-04-15:
these are the i2 readouts (5 times every 500msecs) with raspberry:
PSN_ReadRaw(AMS/0x01/0x73/4): 0x06635F00 deci:107175680
PSN_ReadRaw(IDT/0x01/0x07/2): 0x0000 deci:0

PSN_ReadRaw(AMS/0x01/0x73/4): 0x06635F00 deci:107175680
PSN_ReadRaw(IDT/0x01/0x07/2): 0x0000 deci:0

PSN_ReadRaw(AMS/0x01/0x73/4): 0x06635F00 deci:107175680
PSN_ReadRaw(IDT/0x01/0x07/2): 0x0000 deci:0

PSN_ReadRaw(AMS/0x01/0x73/4): 0x06635F00 deci:107175680
PSN_ReadRaw(IDT/0x01/0x07/2): 0x0000 deci:0

PSN_ReadRaw(AMS/0x01/0x73/4): 0x06665F00 deci:107372288
PSN_ReadRaw(IDT/0x01/0x07/2): 0x0000 deci:0

rudiratlos, 2019-04-15:
84133: EVENT: Clock#Time=Mon,12:31
94687: WD   : Uptime 2 ConnectFailures 0 FreeMem 13040 WiFiStatus 3
109135: BME280: dew point 10.50C
109147: BME280 : Address: 0x76
109148: BME280 : Temperature: 24.51
109148: BME280 : Humidity: 41.36
109148: BME280 : Barometric Pressure: 1026.13
109151: EVENT: BME280#Temperature=24.51
109179: EVENT: BME280#Humidity=41.36
109183: EVENT: BME280#Pressure=1026.13
123422: I2C: cfg addr: 7regs: 2
123422: I2C: data: 0,0,0,0 size:2
123425: EVENT: FS2012#Reg=0
123468: I2C: cfg addr: 115regs: 4
123468: I2C: data: 6,102,95,64 size:4
123471: EVENT: AMS5915#Reg=1079993862
124686: WD   : Uptime 2 ConnectFailures 0 FreeMem 13256 WiFiStatus 3
144133: EVENT: Clock#Time=Mon,12:32
154686: WD   : Uptime 3 ConnectFailures 0 FreeMem 13232 WiFiStatus 3
173137: BME280: dew point 10.84C
173149: BME280 : Address: 0x76
173150: BME280 : Temperature: 24.50
173150: BME280 : Humidity: 42.33
173150: BME280 : Barometric Pressure: 1026.11
173153: EVENT: BME280#Temperature=24.50
173181: EVENT: BME280#Humidity=42.33
173185: EVENT: BME280#Pressure=1026.11
183422: I2C: cfg addr: 7regs: 2
183422: I2C: data: 0,0,0,0 size:2
183425: EVENT: FS2012#Reg=0
183472: I2C: cfg addr: 115regs: 4
183472: I2C: data: 134,104,95,80 size:4
183475: EVENT: AMS5915#Reg=1348429958
184686: WD   : Uptime 3 ConnectFailures 0 FreeMem 13208 WiFiStatus 3
204133: EVENT: Clock#Time=Mon,12:33
214686: WD   : Uptime 4 ConnectFailures 0 FreeMem 13192 WiFiStatus 3
237137: BME280: dew point 10.42C
237149: BME280 : Address: 0x76
237149: BME280 : Temperature: 24.51
237150: BME280 : Humidity: 41.14
237150: BME280 : Barometric Pressure: 1026.08
237153: EVENT: BME280#Temperature=24.51
237181: EVENT: BME280#Humidity=41.14
237185: EVENT: BME280#Pressure=1026.08
243422: I2C: cfg addr: 7regs: 2
243423: I2C: data: 0,0,0,0 size:2
243426: EVENT: FS2012#Reg=0
243469: I2C: cfg addr: 115regs: 4
243469: I2C: data: 6,102,95,64 size:4
243472: EVENT: AMS5915#Reg=1079993862
244686: WD   : Uptime 4 ConnectFailures 0 FreeMem 13192 WiFiStatus 3
264135: EVENT: Clock#Time=Mon,12:34
274686: WD   : Uptime 5 ConnectFailures 0 FreeMem 13192 WiFiStatus 3
286134: BMx280 : Unable to detect chip ID
287134: BMx280 : Detected BME280
293137: BME280: dew point 10.42C
293157: BME280 : Address: 0x76
293157: BME280 : Temperature: 24.52
293157: BME280 : Humidity: 41.12
293158: BME280 : Barometric Pressure: 1026.03
293161: EVENT: BME280#Temperature=24.52
293189: EVENT: BME280#Humidity=41.12
293192: EVENT: BME280#Pressure=1026.03
303422: I2C: cfg addr: 7regs: 2
303422: I2C: data: 0,0,0,0 size:2
303425: EVENT: FS2012#Reg=0
303469: I2C: cfg addr: 115regs: 4
303469: I2C: data: 6,104,95,96 size:4
303472: EVENT: AMS5915#Reg=1616865286
304686: WD   : Uptime 5 ConnectFailures 0 FreeMem 13008 WiFiStatus 3
324133: EVENT: Clock#Time=Mon,12:35
334686: WD   : Uptime 6 ConnectFailures 0 FreeMem 12992 WiFiStatus 3
358135: BME280: dew point 10.42C
358147: BME280 : Address: 0x76
358147: BME280 : Temperature: 24.51
358148: BME280 : Humidity: 41.14
358148: BME280 : Barometric Pressure: 1026.01
358151: EVENT: BME280#Temperature=24.51
358181: EVENT: BME280#Humidity=41.14
358185: EVENT: BME280#Pressure=1026.01
363422: I2C: cfg addr: 7regs: 2
363422: I2C: data: 0,0,0,0 size:2
363425: EVENT: FS2012#Reg=0
363468: I2C: cfg addr: 115regs: 4
363468: I2C: data: 134,99,95,112 size:4
363471: EVENT: AMS5915#Reg=1885299590
364686: WD   : Uptime 6 ConnectFailures 0 FreeMem 12992 WiFiStatus 3
384133: EVENT: Clock#Time=Mon,12:36
394686: WD   : Uptime 7 ConnectFailures 0 FreeMem 12992 WiFiStatus 3
422135: BME280: dew point 10.46C
422147: BME280 : Address: 0x76
422147: BME280 : Temperature: 24.54
422148: BME280 : Humidity: 41.17
422148: BME280 : Barometric Pressure: 1026.03
422151: EVENT: BME280#Temperature=24.54
422181: EVENT: BME280#Humidity=41.17
422185: EVENT: BME280#Pressure=1026.03
423422: I2C: cfg addr: 7regs: 2
423422: I2C: data: 0,0,0,0 size:2
423425: EVENT: FS2012#Reg=0
423469: I2C: cfg addr: 115regs: 4
423469: I2C: data: 134,99,95,112 size:4
423472: EVENT: AMS5915#Reg=1885299590
424686: WD   : Uptime 7 ConnectFailures 0 FreeMem 12992 WiFiStatus 3
444133: EVENT: Clock#Time=Mon,12:37
454686: WD   : Uptime 8 ConnectFailures 0 FreeMem 12992 WiFiStatus 3
483419: I2C: cfg addr: 7regs: 2
483419: I2C: data: 0,0,0,0 size:2
483423: EVENT: FS2012#Reg=0
483465: I2C: cfg addr: 115regs: 4
483466: I2C: data: 134,99,95,112 size:4
483469: EVENT: AMS5915#Reg=1885299590
484687: WD   : Uptime 8 ConnectFailures 0 FreeMem 12968 WiFiStatus 3
486135: BME280: dew point 10.48C
486155: BME280 : Address: 0x76
486155: BME280 : Temperature: 24.48
486155: BME280 : Humidity: 41.37
486155: BME280 : Barometric Pressure: 1026.02
486159: EVENT: BME280#Temperature=24.48
486188: EVENT: BME280#Humidity=41.37
486192: EVENT: BME280#Pressure=1026.02
504133: EVENT: Clock#Time=Mon,12:38
514686: WD   : Uptime 9 ConnectFailures 0 FreeMem 12976 WiFiStatus 3
543422: I2C: cfg addr: 7regs: 2
543422: I2C: data: 0,0,0,0 size:2
543426: EVENT: FS2012#Reg=0
543469: I2C: cfg addr: 115regs: 4
543469: I2C: data: 134,99,95,144 size:4
543472: EVENT: AMS5915#Reg=2422170502
544686: WD   : Uptime 9 ConnectFailures 0 FreeMem 12976 WiFiStatus 3
550135: BME280: dew point 10.63C
550147: BME280 : Address: 0x76
550147: BME280 : Temperature: 24.58
550148: BME280 : Humidity: 41.53
550148: BME280 : Barometric Pressure: 1025.99
550151: EVENT: BME280#Temperature=24.58
550182: EVENT: BME280#Humidity=41.53
550185: EVENT: BME280#Pressure=1025.99
564133: EVENT: Clock#Time=Mon,12:39
574686: WD   : Uptime 10 ConnectFailures 0 FreeMem 12976 WiFiStatus 3

rudiratlos, 2019-04-15:
sample time 1sec on generic i2c devs:
1412160: I2C: cfg addr: 7regs: 2
1412160: I2C: data: 0,0,0,0 size:2
1412164: EVENT: FS2012#Reg=0
1412205: ADC  : Analog value: 32 = 32.000
1412208: EVENT: A0#Analog=32.00
1412255: I2C: cfg addr: 115regs: 4
1412255: I2C: data: 134,102,95,240 size:4
1412258: EVENT: AMS5915#Reg=4032784006
1413160: I2C: cfg addr: 7regs: 2
1413161: I2C: data: 0,0,0,0 size:2
1413164: EVENT: FS2012#Reg=0
1413207: ADC  : Analog value: 32 = 32.000
1413210: EVENT: A0#Analog=32.00
1413255: I2C: cfg addr: 115regs: 4
1413255: I2C: data: 134,97,95,240 size:4
1413258: EVENT: AMS5915#Reg=4032782726
1414162: I2C: cfg addr: 7regs: 2
1414162: I2C: data: 0,0,0,0 size:2
1414165: EVENT: FS2012#Reg=0
1414231: ADC  : Analog value: 31 = 31.000
1414234: EVENT: A0#Analog=31.00
1414279: I2C: cfg addr: 115regs: 4
1414279: I2C: data: 6,102,95,224 size:4
1414282: EVENT: AMS5915#Reg=3764348422
1414686: WD   : Uptime 24 ConnectFailures 0 FreeMem 12696 WiFiStatus 3
1415160: I2C: cfg addr: 7regs: 2
1415160: I2C: data: 0,0,0,0 size:2
1415163: EVENT: FS2012#Reg=0
1415205: ADC  : Analog value: 31 = 31.000
1415210: EVENT: A0#Analog=31.00
1415255: I2C: cfg addr: 115regs: 4
1415255: I2C: data: 6,99,95,224 size:4
1415258: EVENT: AMS5915#Reg=3764347654
1416160: I2C: cfg addr: 7regs: 2
1416160: I2C: data: 0,0,0,0 size:2
1416164: EVENT: FS2012#Reg=0
1416205: ADC  : Analog value: 32 = 32.000
1416208: EVENT: A0#Analog=32.00
1416255: I2C: cfg addr: 115regs: 4
1416256: I2C: data: 134,99,95,240 size:4
1416259: EVENT: AMS5915#Reg=4032783238
1417174: EVENT: FS2012#Reg=0
1417213: ADC  : Analog value: 32 = 32.000
1417216: EVENT: A0#Analog=32.00
1417262: I2C: cfg addr: 115regs: 4
1417262: I2C: data: 134,99,95,240 size:4
1417266: EVENT: AMS5915#Reg=4032783238
1418160: I2C: cfg addr: 7regs: 2
1418160: I2C: data: 0,0,0,0 size:2
1418163: EVENT: FS2012#Reg=0
1418234: ADC  : Analog value: 31 = 31.000
1418238: EVENT: A0#Analog=31.00
1418283: I2C: cfg addr: 115regs: 4
1418284: I2C: data: 134,99,95,240 size:4
1418287: EVENT: AMS5915#Reg=4032783238
1419161: I2C: cfg addr: 7regs: 2
1419161: I2C: data: 0,0,0,0 size:2
1419164: EVENT: FS2012#Reg=0
1419203: ADC  : Analog value: 31 = 31.000
1419206: EVENT: A0#Analog=31.00
1419253: I2C: cfg addr: 115regs: 4
1419253: I2C: data: 134,102,95,240 size:4
1419256: EVENT: AMS5915#Reg=4032784006
1420160: I2C: cfg addr: 7regs: 2
1420160: I2C: data: 0,0,0,0 size:2
1420164: EVENT: FS2012#Reg=0
1420205: ADC  : Analog value: 31 = 31.000
1420208: EVENT: A0#Analog=31.00
1420284: I2C: cfg addr: 115regs: 4
[... 127 lines cut ...]
1436280: EVENT: AMS5915#Reg=4032784006
1437160: I2C: cfg addr: 7regs: 2
1437160: I2C: data: 0,0,0,0 size:2
1437164: EVENT: FS2012#Reg=0
1437206: ADC  : Analog value: 32 = 32.000
1437210: EVENT: A0#Analog=32.00
1437256: I2C: cfg addr: 115regs: 4
1437256: I2C: data: 6,102,95,224 size:4
1437259: EVENT: AMS5915#Reg=3764348422
1438160: I2C: cfg addr: 7regs: 2
1438161: I2C: data: 0,0,0,0 size:2
1438164: EVENT: FS2012#Reg=0
1438206: ADC  : Analog value: 32 = 32.000
1438209: EVENT: A0#Analog=32.00
1438256: I2C: cfg addr: 115regs: 4
1438256: I2C: data: 134,99,95,240 size:4
1438259: EVENT: AMS5915#Reg=4032783238
1439160: I2C: cfg addr: 7regs: 2
1439160: I2C: data: 0,0,0,0 size:2
1439164: EVENT: FS2012#Reg=0
1439207: ADC  : Analog value: 32 = 32.000
1439210: EVENT: A0#Analog=32.00
1439256: I2C: cfg addr: 115regs: 4
1439256: I2C: data: 6,99,95,224 size:4
1439259: EVENT: AMS5915#Reg=3764347654

uzi18, 2019-04-15:
last one:
3764347654=0xE05F6306
reg 0 =0x06
reg 1 =0x63
reg 2 =0x5F
reg 3 =0xE0

uzi18, 2019-04-15:
so:
digoutp=0x6306=25350 dec
digoutt=0xe05f=57439 dec

rudiratlos, 2019-04-15:
I think it looks very good. well done. 
Just a wish, can the i2c adr supplied as hex and not decimal?
now we can read data from 2 different i2c sensors with just a simple generic plugin. That will support a lot of others 'simple' i2c sensors with just one plugin.
How to publish that module to the standard package?

uzi18, 2019-04-15:
@TD-er what do you think? I need plugin number for this one, have choose 87 for tests. 

it is limited now for 4 bytes reads and all bytes filled into LONG

uzi18, 2019-04-15:
wrong should be interpreted as: digoutp=0x0663= 1635 dec
digoutt=0x5fe0= 24544dec

TD-er, 2019-04-15:
The first unused one is 84, so you can use that one.
Of leave it at 87. We will fill up the gap :)
Later we can extend the plugin to allow for different output types after I merge my PR regarding dynamic number of output values.

uzi18, 2019-04-15:
@TD-er add plugin to test or dev ?

TD-er, 2019-04-15:
Make it dev first, since I do think we will change a lot about it in the beginning

Christianbvb, 2019-05-05:
This is a great idea. Where can I find the source? I cannot find it on ESPEasyPluginPlayground...

uzi18, 2019-05-05:
did not send sources for this plugin jet, it will. be available when I commit it to my wip PR.

TD-er, 2019-05-05:
It is not yet put here as a pull request, so it is not yet available.

uzi18, 2019-05-05:
In fact have forget about it :P

Renevandenbroek, 2019-10-16:
Any idea when the plugin will be available?

uzi18, 2019-10-16:
@rudiratlos @Renevandenbroek almost forget about it

Renevandenbroek, 2019-10-16:
I have a dimmer running on a attiny86 it needs via I2C data between 5 and 65. Will the plugin run on a esp8266-01?

uzi18, 2019-10-16:
@Renevandenbroek this plugin has only read feature, but I think we can add write feature as well - extra command(s)

Renevandenbroek, 2019-10-16:
Great! If I can be of any help e.g. test it? let me know.

uzi18, 2019-10-16:
@Renevandenbroek first I need to rebase my repo, @TD-er done great job to split all and remove objects duplication in runtime and my repo is before these changes, next will create PR with read feature
so we can think about it later

My time is very limited :/ but will try to sync and PR in this week.

how do you want to send these data to esp or you want just to use rules for dimmer setting?

Renevandenbroek, 2019-10-16:
I want to send the data via Domoticz to the esp8266-01

uzi18, 2019-10-16:
@Renevandenbroek ok but how? via mqtt or url in scripts?

TD-er, 2019-10-16:
I think you should first make a command for the plugin to write data to a register.
So it does need 2 parameters, the register and the value :)
You probably need to have several of these, since writing an 8 bit value differs from writing a 16 bit etc.
See the I2C.ino file we have (not sure about the exact file name)

Then we can later add support in the controllers to call this command when needed.
It could also be time based or event based (from rules)

Renevandenbroek, 2019-10-16:
I must admit that I do not have an idea yet. I want to have a slider in Domoticz to adjust the dimmer. Up till now I only have switches running. Domoticz sends things like this" http://192.168.1.66/control?cmd=GPIO,0,0". It is the simple sfuff:) but I want much more:) Like smooth dimmers who I can activate from my new bought Google nesthub too. (all with/via espeasy en domoticz) But with 70 years I am learning not that fast anymore:)

Renevandenbroek, 2019-10-16:
This is what is running on my attiny8266 (comming from https://arduino.stackexchange.com/questions/32817/attiny85-ac-phase-control-for-dimming-a-light-bulb )
 "
#include <avr/interrupt.h>
#include <avr/io.h>
#include <TinyWireS.h>

#define PULSE 4       //trigger pulse width (counts)
#define I2C_SLAVE_ADDR 0x4 // the 7-bit address (remember to change this when adapting this example)

byte trigger = 3; //triac is pin 2
byte detector = 4;    // zero cross detection is pin 3
byte level = 10;
byte maxLevel = 65;
byte minLevel = 0;

void setup() {
    TinyWireS.begin(I2C_SLAVE_ADDR); // join i2c network is pin 7 en pin 5
    TinyWireS.onRequest(requestEvent); //setup i2c requester
    digitalWrite(detector, HIGH);  //enable pull-up resistor
    pinMode(trigger, OUTPUT);// Set AC Load pin as output
    TCCR1 = 0;     //stop timer
    OCR1A = level;    //initialize the comparator
    TIMSK = _BV(OCIE1A) | _BV(TOIE1); //interrupt on Compare Match A and  enable timer overflow interrupt
    GIMSK = 0b00100000; //Enable pin change interrupt
    PCMSK = 0b00010000; //PB4, physical pin 3 PCINT4
    TCCR1 = B00001011; //Prescale the timer
    sei();  // Turn on interrupts
}

ISR(PCINT0_vect){ //interrupt looking for zero crossing
        if (digitalRead(4) == HIGH){
                TCNT1 = 0;   //reset timer - count from zero
                OCR1A = level;
                TCCR1 = B00001011;// prescaler on 1024, see table 12-5 of the tiny85 datasheet
       }
}
ISR(TIMER1_COMPA_vect) {
    //comparator match
    digitalWrite(trigger,HIGH); //set triac gate to high
    TCNT1 = 255-PULSE;
    //trigger pulse width for a few cycles for triac to latch on. 255 bc the counter can only count up to 255
}

ISR(TIMER1_OVF_vect) {
    //timer1 overflow
    digitalWrite(trigger, LOW); //turn off triac gate
    TCCR1 = 0;  //disable timer stop unintended triggers
}

void loop() {}

void requestEvent() {
    if (TinyWireS.available()) {
        level = TinyWireS.receive();
        if (level > maxLevel) {
            level = maxLevel;
        }
         else if (level < minLevel){
            level = minLevel;
        }
    }
    TinyWireS.send(OCR1A);
} 
"

Renevandenbroek, 2019-10-25:
Is there anything I can do?
René

TD-er, 2019-10-25:
Well, right now I don't have the time to work on it.
As you may have seen from the activity in other issues, I'm spending a lot of time it making things more generic and patching the bugs emerging from it.

NILD01, 2021-12-13:
Hi,
Can i use the  the generic i2c plug in for sending data?
And where can i find it?
I have a MCP4725 with i2c interface (dac).
I would like to implement it to EspEasy.

The only thing i found was a plugin on the playground thats named "Plugin 246: ESP32 DAC"
But it gives error in the custom build.

i'm lost for now.

[3 more replies]

closed 2025-09-16 by TD-er