letscontrolit/ESPEasy #2345: Data is not beign send to multiple controllers - only to one

Grzegorzsa, 2019-02-22:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

Grzegorzsa, 2019-02-22:
EspEasy Mega Dev
I've got SDS dust sensor configured. When Isend data to one controller everything works fine. When I try to send data to multiple controllers only one works.

Protocol: Generic HTTP Advanced[TESTING]

TD-er, 2019-02-22:
What are the used controller settings?

Grzegorzsa, 2019-02-23:
The settings in both cases are the same, but data goes only to one api. I've noticed it long time ago, but recently someone else that is using my api reported problem with sending data to 2 different api's so I assumed that there is some bug in the software.

[image]
[image]

TD-er, 2019-02-23:
So you have 2 controllers using the same type of controller?
That has never been tested as far as I know, so I guess it is quite possible that it will not work for all controller types.