letscontrolit/ESPEasy #2351: Small Request in Data Transformations.

Budman1758, 2019-02-25:
In reference see this issue.
[https://github.com/letscontrolit/ESPEasy/pull/1289](url)

My request is to make a very minor change.  See this section...

> Transformations:

Special:
!: negate (only for 0 or 1)
R: RIGHT JUSTIFY

V: value
O: ON/OFF
C: OPEN/CLOSE  **<<<<< Change this to "CLOSED"**
U: UP/DOWN

Methinks this would make a bit more sense. My use case is a valve that is either "OPEN" or "CLOSED"
CLOSE is not quite as informative and in some cases could be construed as being  "near"

Just a wee little thing...  :)

TD-er, 2019-02-25:
So this is one only for the wiki/[ReadTheDocs](https://espeasy.readthedocs.io/en/latest/) I assume?

Budman1758, 2019-02-25:
No. The output is "CLOSE" as it is now. Would like it to be "CLOSED" Thats what the code outputs now. One would also need to update the Wiki also.  :)

TD-er, 2019-02-25:
Hmm if it will be changed in the code, then it may break some existing rules.

uzi18, 2019-02-25:
CLOSED also use 1 char more on LCD display.

Grovkillen, 2019-02-25:
It's "only" the string for displays and other outputs that will be
effected. So probably rules will work but like Uzi says, it will add one
more character. But for a native American I'd say Bud got the upper hand in
the correctness of the grammar 😎

TD-er, 2019-02-25:
> [...] But for a native American I'd say Bud got the upper hand in the correctness of the grammar 😎 

[Native American?](https://www.google.com/search?q=native+American&rlz=1C1GGRV_enNL751NL751&source=lnms&tbm=isch&sa=X&ved=0ahUKEwiz98mjzdfgAhVEaFAKHahFB_EQ_AUIDigB&biw=1920&bih=1008) ;)

Grovkillen, 2019-02-25:
Yep, I'm obviously not ;)

Budman1758, 2019-02-25:
> [Native American? ;)

Well.... Not exactly. :)  We weren't very nice to those folks..... To put it mildly..

PR #3094 (2020-06-02): [Transformation] Add L/l and c transformations (LEFT/RIGHT, L/R and OPEN/CLOSED)
Just a small addition of what seems to be missing but useful.
(Changed form first using R/r to L/l as R is used for right-justify)
Added c for OPEN/CLOSED in addition to C, resolves #2351
closed 2020-06-02 by TD-er