letscontrolit/ESPEasy #2354: ESP32 Setting won´t Save

Icestan, 2019-02-25:
Since Version 20190215 on ESP32, the Data for Wifi via Console (wifissid and wifikey) wont save.
After Restart i loop back to the AP-Mode.

If I have flash with an older Version till 20190212, I can set the Wifi-Data and can save settings on the Website. After Flashing the newer Versions (Only Appdata) and i want to save settings, i get the Message
Error: Incorrect offset in struct: SettingsStruct

TD-er, 2019-02-25:
Hmm, it looks like I may have to add an extra check here:
```C++
  if (offsetof(SettingsStruct, ResetFactoryDefaultPreference) != 1224) {
```
So this means the ESP32 does store some things a bit differently.
This is also important for the new UI we're working on.

TD-er, 2019-02-25:
Pull request is ready and will very likely be merged for tonight's build

Icestan, 2019-02-26:
Thanks it works fine. The Settings will be save, now.

closed 2019-02-26 by Icestan