letscontrolit/ESPEasy #2355: Cannot load 20190225 on SonoffPow R1 after fix for #2251

2bontb, 2019-02-25:
### Checklist
- [ x] The title of this issue is "to the point" and descriptive.
- [ x] This issue describes what is happening.
- [ x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x ] This issue describes how to reproduce it.
- [ x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x ] given a list of active plugins or controllers when applicable.
- [ x] filled out all applicable fields below.

### Steps already tried...
- [ x] Tried a clean install (empty `.bin` files are included in the ZIP)
Loaded clean4M first
Than added Wifi credentials via arduineo IDE
Could not add device HLW8012: Page crashes

- [ ] Tested previous/other build (mention which one already tested)
Is same problem as reported in #2251 

- [ ] Tested on other node to make sure hardware isn't defective.
No did not do this.

- [ ] Verified if the problem is limited to a single plugin/controller
Yes; Load is running on other devices.

### Summarize of the problem/feature request

1. I loaded 4M blank; Because upgrade from my 20181220 version did not work.
2. Loaded 20190225
3.  See what I did on arduino IDE:

>reset
22844 : Command: reset
22845 : reset
RESET: Resetting factory defaults... using Sonoff POW settings
RESET: Warm boot, reset count: 1
RESET: formatting...
RESET: formatting done...
RESET: Succesful, rebooting. (you might need to press the reset button if you've justed flashed the firmware)

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vbb28d4a3
~ld
⸮⸮⸮U85 : 

INIT : Booting version: mega-20190225 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
86 : INIT : Warm boot #7 - Restart Reason: Software/System restart
88 : FS   : Mounting...
113 : FS   : Mount successful, used 76053 bytes of 957314
398 : CRC  : program checksum       ...OK
435 : CRC  : SecuritySettings CRC   ...FAIL
517 : INIT : Free RAM:30144
519 : INIT : SPI not enabled
533 : INFO : Plugins: 7 [Normal] [Sonoff POW R1/R2] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
534 : EVENT: System#Wake
556 : WIFI : No valid wifi settings
556 : WIFI : Could not connect to AP!
557 : WIFI : Set WiFi to AP
596 : EVENT: WiFi#APmodeEnabled
616 : WIFI : AP Mode ssid will be Sonoff POW R1/R2_0 with address 192.168.4.1
618 : EVENT: System#Boot
[... 170 lines cut ...]

Page displays, so I can update settings

### Actual behavior

System crashes.

### Steps to reproduce

1. See above.

### System configuration

Hardware: Sonoff POW 4M rev 2.0

ESP Easy version:  20190225 hard sonoff 4M

ESP Easy settings/screenshots: 
[image]

### Rules or log data

```

```

TD-er, 2019-02-25:
OK, I will close 'the other one' and we'll continue in this one.
I will also try to run the nightly build, just to make sure it is also failing here.

TD-er, 2019-02-25:
It is confirmed, the one with the nightly build is not working on the POW r1.
And before updating I had confirmed it was possible to add/delete and edit the plugin on the node I have here.

TD-er, 2019-02-25:
Without any changes, I just made this [test build](https://www.dropbox.com/s/kbz76tzjsvd2l79/Issue2355_Sonoff_POW_testbuild.rar?dl=0) Can you please try if it works?
If it does, I will (again) throw away all caches on Travis, since one of the build environments is then apparently broken (again).

2bontb, 2019-02-25:
TD-er,
I wanted to help on where it goes wrong. So I compiled the source in IDE so I could do a stack trace.
But to my wonder: The bin file I created did not create the problem. It worked as you described.
So I do only have the problem if I load the bin file that was in the distribution. This is the boot info.
So we can compare the difference:

INIT : Booting version: (custom) (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
5956 : INIT : Cold Boot - Restart Reason: Software/System restart
5958 : FS   : Mounting...
5983 : FS   : Mount successful, used 76304 bytes of 957314
5993 : CRC  : No program memory checksum found. Check output of crc2.py
6030 : CRC  : SecuritySettings CRC   ...OK 
6032 : CRC  : binary has changed since last save of Settings
6142 : INIT : Free RAM:30256
6144 : INIT : SPI not enabled
6151 : P076: Init object done
6154 : P076: Init Basic Resistor Values done
6161 : P076: Saved Calibration after INIT
6164 : P076: Applied Calibration after INIT
6176 : INFO : Plugins: 7 [Normal] [Sonoff POW R1/R2] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3)
6178 : EVENT: System#Wake
6232 : WIFI : Set WiFi to STA
6265 : WIFI : Connecting DeHeijIOT attempt #0
6268 : WIFI : Not configured in Station Mode!!: xx
6270 : EVENT: System#Boot
6287 : ACT  : timerSet,3,300
6297 : Command: timerset
6388 : SYS  : 0.00,31.00,100.00,0.00
6395 : EVENT: Uptime#Uptime=0.00
6426 : EVENT: Uptime#RSSI=31.00
6449 : EVENT: Uptime#Load=100.00
6471 : EVENT: Uptime#=0.00
7600 : P076: Read values - V=0 - A=0.00 - W=0 - Pf%=0
7604 : EVENT: EnergieWerkkamer#Voltage=0.00
7628 : EVENT: EnergieWerkkamer#Current=0.00
7650 : EVENT: EnergieWerkkamer#Power=0.00
7674 : EVENT: EnergieWerkkamer#Factor=0.00
7729 : WD   : Uptime 0 ConnectFailures 0 FreeMem 27200 WiFiStatus 0
10273 : WIFI : Connected! AP: DeHeijIOT (7A:8A:20:9F:5F:B3) Ch: 6 Duration: 3781 ms
10276 : EVENT: WiFi#ChangedAccesspoint
10302 : WIFI : DHCP IP: 10.2.1.123 (SonoffWerkkamer-123) GW: 10.2.1.138 SN: 255.255.255.0   duration: 34 ms
10305 : EVENT: WiFi#Connected
10323 : ACT  : timerSet,3,0
10331 : Command: timerset

I see that you already commented expecting a wrong build. I think my try does confirm this.
I will try your build also right now.

2bontb, 2019-02-25:
TD-er; I can confirm that your build works,.

TD-er, 2019-02-25:
I already cleared the caches on Travis and made sure there was only one build active.
So there should be a completely clean build cache right now.

Maybe I will change the settings for Travis to keep as little cached as possible, to prevent issues like these.
Tonight there will be a new build, so if you could try that one tomorrow, that would be great.

TD-er, 2019-02-26:
I have absolutely no idea what's wrong here.
The build I made myself is working just fine, but also the build made last night is failing as described here.

2bontb, 2019-02-27:
TD-er,
I have the feeling that my original problem was also due to "the" build error. The symptoms are completely the same with the last build as with my original problem.  I had this already with release 20190116 loading the distributed binary. You may want to try those distributed builds also on your system to see on what build it started.
Let me know if I can help with anything.

Other qustion: To allow builds with Arduino IDE I have to do some code changes; Should I raise also tickets for those two changes I make every time; Alsoforloading the ITHO plugin, I need to make some changes. I never reported that build error.

TD-er, 2019-02-27:
Please make a pull request for those changes.

And I have some idea of what's wrong with the build here and I also tried to build it locally in Linux and it fails the same way as the nightly build.
Only thing is, that if it appears to be the issue I'm thinking about, then we may have some bigger issue at hand and a lot of code to check. (or find a way to detect the issue at run time and report an error)

Barracuda09, 2019-02-28:
@TD-er What can we do to help and were should we look at in the code?

TD-er, 2019-02-28:
I have to change to code in this plugin to no longer use a struct like is now used to keep track of the different settings for different modules.
It seems like the Windows compiler does something different here than the Linux compiler regarding alignment of data stored on the flash.

Barracuda09, 2019-02-28:
@TD-er Yes maybe it is not a good idea to depent on alignment of data in combination with compilers

TD-er, 2019-02-28:
It should be fixed.
Before I merge it, I will build it in Linux and test on my node here, but I see no reason why it shouldn't work like this.

closed 2019-10-10 by TD-er