letscontrolit/ESPEasy #2356: 20180225 pre-release asking for Username and Password

Obicom, 2019-02-25:
### Checklist
- [x ] The title of this issue is "to the point" and descriptive.
- [ x] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ -] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ x] entered the full filename of the used version (e.g. `ESP_Easy_mega-ESP_Easy_mega-20190225_normal_IR_ESP8266_4M.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [- ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x ] Tested previous/other build (mention which one already tested)
- [ x] Tested on other node to make sure hardware isn't defective.
- [- ] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request
After flashing ESP_Easy_mega-20190225_normal_IR_ESP8266_4M.bin via OTA and Flashertool the web site asking for username / password as shown in the screenshot. After downgrading to 20180216 all works as expected.

### Expected behavior
Normal web GUI login with password

### Actual behavior
Popup windows for username /password
[image]

TD-er, 2019-02-25:
Did you have a password set in the previous installed version?
In just any page?

Obicom, 2019-02-25:
I did setup the login password .... in the previous version 

[image]

TD-er, 2019-02-25:
OK, so now it does do extra checks on the password, even on pages than do not need this check, right?

Obicom, 2019-02-25:
It is not possible to login with this issue ... or do you know the username?

TD-er, 2019-02-25:
Nope, I will have a look at it.
Looks like a nice bug which should be fixable before midnight ;) (it is some policy of mine not to merge any commit for a nightly build after midnight)
So I will pop a beer and start lookin' :)

TD-er, 2019-02-25:
You could try the user "admin" by the way.

Obicom, 2019-02-26:
I tried it with "admin" and my password without success ...

pwassink, 2019-02-26:
Suffered same issue in 20190226 Mega version, 

username admin 
passwd (what you had used before that) 

Worked for me .

jimmys01, 2019-02-26:
I can confirm this, admin as username works

wolverinevn, 2019-02-27:
Username: admin
Password: pre-set
NOT worked for me!

TD-er, 2019-02-27:
@wolverinevn Does your password contain any non standard characters? (other than letters and numbers)

Obicom, 2019-02-27:
one addtional question to @TD-er : will this the future behavior in ESPEasy with username/password popup windows or do you plan to go back to the "old" style with only password in the web GUI (what I would prefere)?

wolverinevn, 2019-02-27:
@TD-er None. Only a-z and numbers.

Obicom, 2019-02-27:
With latest version 20190226 the username: admin and my password works for me.

TD-er, 2019-02-27:
@wolverinevn You could go back a version and reset/clear the password and try again by setting one.
I will try to add a description to the dialog to use "admin" as default.
The idea of this change is to later support multiple users.

Could you also try another browser, just to be sure?

wolverinevn, 2019-02-27:
@TD-er It works with Chrome (not Safari). Phew!

TD-er, 2019-02-27:
OK, so it may still need a look at then.
Glad you can now use it again.

TD-er, 2019-02-28:
It looks to be [this issue](https://github.com/esp8266/Arduino/issues/4717)

jimmys01, 2019-03-06:
It is stell asking for username and password for me.
After updating from 20190202 to 20190305 and tryong to connect to x.x.x.x.x/login
I put the password and it still pops up the form that you need to type the admin and the Password. I use firefox if that is of importance

TD-er, 2019-03-06:
I changed it for the core 2.4.x builds to use 'the other' form for authentication. I'm not sure if that one demands a username too, but I guess it would.

The "old way" we used up to like a month ago, was not really a good way of protection.
As soon as a user was authenticated, all access was authenticated. It was not using a session or any other form of proper protection.
So @ppisljar noticed this and changed the authentication.

What core version build are you using? And is it possible to authenticate using Firefox?
If so, can you explain what's the problem you're now facing?

ppisljar, 2019-03-06:
not sure if it was already mentioned above, but username is hardcoded to `admin`, password is the password you setup with previous (or current) version

ppisljar, 2019-03-06:
and just to offer a bit more explanation:
previous versions were sending password over in clear text. esp would then check this password and if correct go into "authenticated" mode. this means that any other user on your network would no longer need to enter password.
current version uses HTTP basic/digest authentication. Its much more secure, really authenticates on per user basis (so if you login, someone else on the network still can't access it) Also its a standard, so espeasy doesn't need to do much, browser handles pretty much everything.
Its possible to create custom login screen in our web app (something that looks better or at least more integrated than what the browser provides), but i would say thats not really a priority at this point.

jimmys01, 2019-03-06:
Ok I understand now the reasons why it has been made the way it is. My problem is that i can't login just by the /login screen (it is still accessible). This breaks scripts i have for updating and doing stuff to nodes. This bugs me because I am approaching 200 nodes on the swarm.

> Its possible to create custom login screen in our web app (something that looks better or at least more integrated than what the browser provides), but i would say thats not really a priority at this point.

This would solve my problem, essentialy restoring the old  /login screen, but with the new session protection stuff.

TD-er, 2019-03-06:
Can you post some parts of this script, which are related to the login?

jimmys01, 2019-03-06:
This is a 3 part command seperated by && markings. The first part logins to the node, the second part updates the rules and the last part updates the node

`wget -q --tries=2 --timeout=20 -O /dev/null "http://172.18.1.231/login" --post-data "password=*******&submit=Submit"&&curl -# -o /dev/null --form datafile=@rules1.txt --connect-timeout 10  http://172.18.1.231/upload&&curl -# -o /dev/null --form update=@ESP_Easy.bin --max-time 40 --connect-timeout 10 --retry 1 http://172.18.1.231/update`

TD-er, 2019-03-06:
Does this help?
`wget --http-user=USERNAME --http-password=PASSWORD http://172.18.1.231/login`

TD-er, 2019-03-06:
I just tested to get the main page using wget and that just fetches the page.
`wget --http-user=admin --http-password=test123 http://192.168.1.123/`

closed 2019-10-10 by TD-er

kimseongki66, 2022-03-18:
I solved it using IE.