letscontrolit/ESPEasy #2364: MQ-Messages are skipped when using deepsleep in a rule

SvnKswttr, 2019-02-27:
According to this thread: http://www.letscontrolit.com/forum/viewtopic.php?p=35576#p35576
Usig wemos D1 mini on ESP_Easy_mega-20190227_normal_ESP8266_4M.bin

Following rule does not send any data to MQ
`on MQTT-IN#AwakeDuration do
  publish,%sysname%/SensorData/H,[SensorData#H] //publish a reading
  publish,%sysname%/SensorData/T,[SensorData#T] //publish a reading
  publish,%sysname%/SensorData/P,[SensorData#P] //publish a reading
  publish,%sysname%/SensorData/PublishTime,%systime% //publish reading time
  if [MQTT-IN#DeepSleepEnabled]=0
    publish,%sysname%/DeepSleep/Status,"Staying awake."
  else
  Publish,%sysname%,DeepSleep/Status,"Need to sleep for [MQTT-IN#DeepSleepDuration] sec"
  deepsleep [MQTT-IN#DeepSleepDuration]
  endif
endon`
It works well when adding a short break before the deepsleep kicks in:
`  on MQTT-IN#AwakeDuration do
  publish,%sysname%/SensorData/H,[SensorData#H] //publish a reading
  publish,%sysname%/SensorData/T,[SensorData#T] //publish a reading
  publish,%sysname%/SensorData/P,[SensorData#P] //publish a reading
  publish,%sysname%/SensorData/PublishTime,%systime% //publish reading time
  if [MQTT-IN#DeepSleepEnabled]=0
    publish,%sysname%/DeepSleep/Status,"Staying awake."
  else
    timerSet,2,1 // wait a sec, please
  endif
endon

on Rules#Timer=2 do
  Publish,%sysname%,DeepSleep/Status,"Need to sleep for [MQTT-IN#DeepSleepDuration] sec"
  deepsleep [MQTT-IN#DeepSleepDuration]
endon
`

TD-er, 2019-02-27:
I edited your post to change the inline code tag (single left tick) into code block tags (3 left ticks)

TD-er, 2019-02-27:
Deepsleep doesn't seem to wait for the buffers to be flushed.
So I guess we should make some flush command to make sure it is flushed.
Ofcourse this command also needs some timeout setting.

tonhuisman, 2022-04-18:
This seems to have been fixed, so can probably be closed.

closed 2022-04-18 by TD-er