letscontrolit/ESPEasy #2368: [IR RX/TX] Bugfixes needed for IR RX/TX plugins

TD-er, 2019-02-28:
As discussed [here](https://github.com/letscontrolit/ESPEasy/pull/2360#issuecomment-468313680) and in this issue: https://github.com/letscontrolit/ESPEasy/issues/2339

After only a short look at the code, a number of bugs were already pinpointed.
This means it is very likely there will be more.

Also the code is quite complex and uses a lot of resources.
Complexity makes it hard to maintain the code and will lead to new (hard to find) bugs when something has to change.

In short, these plugins deserve some proper debugging and maybe some rewrite.

lhshengshi, 2019-03-01:
hard to change

jimmys01, 2019-03-01:
I will add heavy comments in the code in the next days, but for now you can imagine the IR TX and RX plugins as having 2 separate functions.
1 is to accommodate the excellent markszabo/IRremoteESP8266 library and add the functionality of his IRrecvDumpV2.ino (in the RX) and the IRMQTTServer (in the TX)

The second function is a hack, because of the limitations to send RAW timings of that char size:
`uint16_t rawData[63] = {2420, 506,  1280, 638,  592, 566,  564, 594,  1226, 626,  538, 620,  1226, 802,  1038, 630,  590, 638,  564, 570,  590, 660,  566, 592,  1248, 670,  1120, 592,  566, 658,  1140, 20684,  2402, 580,  1280, 566,  564, 594,  676, 690,  1042, 536,  564, 660,  1118, 660,  1116, 660,  588, 572,  588, 570,  566, 594,  590, 636,  1118, 658,  1120, 660,  566, 594,  1226};`

So RAW and RAW2 encodings are trying to do a loosely compression on those timings and produce a much smaller command that is transmittable via MQTT or HTTP.

RAW is calculated via an google excel sheet is guaranteed to have a result and has configurable timings.
RAW2 is calculated in the RX plugin and it is very easy to get a result (directly from the log) from but not guaranteed. I had remotes that the encoding could not find common denominators. Also because you can’t set timings manually but they are calculated the RAW2 produces much longer encoded strings. Having that said RAW2 is a very, very good step forward. 

Both RAW and RAW2 have a place only if the IRrecvDumpV2.ino can give you a code like 
`IRSEND,LG,8055214` but this is more likely to happen for satellite tv boxes and uncommon AC units that the library does not support.

dyter, 2019-03-01:
Since 20190301, Raw2 work again for me

uzi18, 2019-03-01:
any hint how to convert raw to raw2 and vice versa?
link to google sheets file?

Will check these example of. code like irrecvdumpv2 good hint

lhshengshi, 2019-03-04:
HI！   I hope will be a new vision--with pms7003 and IR .The  existing
“IR”Vision without PMS7003,and i dont know how to add PMS7003 in code.

TD-er, 2019-03-06:
But that would essentially double the amount of binary versions we have to build.

jimmys01, 2019-03-16:
> link to google sheets file?

This is the google sheet I found and then I enchanced to fit bigger (usualy AC) signals.
https://docs.google.com/open?id=16icZJWFbHq2iIxuEYxHzKWoP4ZEtra_iegIqDCMVvG4

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er