letscontrolit/ESPEasy #2372: 7-segment plugin to normal builds

fly74, 2019-03-05:
I don't know if I asked before. When will the 7-segement plugin finds it way to normal builds from testing?

fly74, 2019-04-29:
Hi, any news? The only build wich worked is the sdk3. But this version cannot be updates OTA. So it's anoying since a while. Every OTA flash ends up in serial flash with new config.

TD-er, 2019-04-29:
Like I replied in the other issue regarding this plugin... 
I have to get these modules myself to actually test them myself

fly74, 2019-04-30:
I'll would test it for your if you make special builds,

fly74, 2019-07-28:
It's a long time ago, can u plz move de plugin to normal build. It still works only with ESP_Easy_mega-20190630_test_core_260_sdk3_alpha_ESP8266_4M.bin at the moment. The thing with more than one display would have a lower priority for me.

TD-er, 2019-07-28:
Right now, I am working on the issue regarding WiFi not able to connect to the AP.
Along with that I am also looking into how data structures are stored (and retrieved) from flash, which I believe is the issue with the 7-segment plugin.
It is a bit like when using `F()` macro to store strings into flash, but it is also used in this plugin for data structures. In discussions on the esp8266/Arduino repo it became clear this may be the explanation for some code running fine in sdk3 core setups and not always in sdk2.

This is also my search area right now for WiFi related build issues.

TD-er, 2019-07-29:
Can you test this one with a different core version?
I made a [test build](https://www.dropbox.com/s/uknq183mlb8yx9a/ESPEasy_mega-20190716-15-PR_2514.zip?dl=0)

It is not yet moved to the 'normal' build.
If it is working on any version that has not SDK3, I will move it to the normal set of plugins.

N.B. do only test this on a node that's reachable for serial update, since most of the fixes are about WiFi issues.

fly74, 2019-07-29:
Which .bin should I test?

fly74, 2019-07-29:
:(

4357 : WIFI : Entering processGotIP()
4358 : WIFI : Entering processDisconnect()
4459 : EVENT: WiFi#Disconnected
4523 : WIFI : Connecting Wlannotforall24 attempt #2
4525 : IP   : Static IP : 172.25.99.27 GW: 172.25.99.1 SN: 255.255.255.0 DNS: 172.25.99.1
4527 : WIFI : WiFi.status() = WL_DISCONNECTED  SSID: XXXXXXXXXXX4641 : WIFI : Entering processGotIP()
4642 : WIFI : Entering processDisconnect()
4744 : EVENT: WiFi#Disconnected
4812 : WIFI : Entering processGotIP()
4813 : WIFI : Entering processDisconnect()
4914 : EVENT: WiFi#Disconnected
4979 : WIFI : Entering processGotIP()
4980 : WIFI : Entering processDisconnect()
5081 : EVENT: WiFi#Disconnected

TD-er, 2019-07-29:
What bin file is it?

fly74, 2019-07-29:
ESP_Easy_mega-20190716-15-PR_2514_normal_ESP8266_4M.bin

TD-er, 2019-07-29:
It looks like you're using static IP.
Can you also test DHCP?
Then I will later look into the static IP stuff.

fly74, 2019-07-29:
now ESP_Easy_mega-20190716-15-PR_2514_normal_ESP8266_4M.bin running with DHCP. Plugin works like a charm

TD-er, 2019-07-29:
Great, then I will also keep my promise :)

fly74, 2019-07-30:
works with Release mega-20190730, but the static IP WLAN issue still exists

uzi18, 2019-07-30:
static ip configuration also does not work for me, still don't understand why with static ip setted espeasy call with dhcp for configuration

TD-er, 2019-07-30:
@uzi18 It looks to me there is a bug in the core lib regarding the call/activation of the dhcpc.
I will look into it.
Last time I fixed it, it was quite a hack to set the static IP mode, since there was no function to set this static variable.
In short, I created an object that inherits the base class to add a function to set the static variable.
I can imagine something has changed there, or maybe the WiFi object gets destroyed and thus resets the static variable indicating it is using a static IP.

fly74, 2019-07-30:
ESP_Easy_mega-20190730-2-PR_2520_normal_ESP8266_4M.bin work with static IP + 7dn ! TOP

closed 2019-07-30 by fly74

uzi18, 2019-07-30:
@TD-er so it looks like we still need to hack something in core :/

fly74, 2019-07-30:
why?

TD-er, 2019-07-30:
@uzi18 I would also like to know what we need to hack in there?
Not saying we don't have to, just curious which one you're referring to :)

uzi18, 2019-07-30:
@TD-er need to find time to check your patch for static IP.
I just talking about static IP is still officially unsupported in esp8266 core ;/
OT: have found something interesting ... Winner Micro W600 Arm Cortex-M3, really cheap and powerful chip with SSL

TD-er, 2019-07-30:
I looked into the code of core 2.5.2 and it seems the _staticIP variable is now set when it does see a network config before starting wifi session.
That was not there before, so that's why I needed that hack before