letscontrolit/ESPEasy #2373: Question: If there will be a vision that with IR and PMS7003 ?

lhshengshi, 2019-03-06:
Question: If there will be a version that with IR and PMS7003 ? 
I want konw Why the IR version without PMS7003? I think PMS7003 is better than the others DUST sensors.
PMS7003 is more small and exact.

TD-er, 2019-03-06:
As far as I know, there were still some issues with these Powerplant sensors.
So that's why they were not yet moved into the "normal" set of plugins.
The IR build is using the 'normal' set of plugins and controllers to keep the build size low, since the IR library has become quite big.

So as soon as these PMS sensors appear to be working fine, I can move them to the 'normal' set and then they will be included in the IR builds too.

lhshengshi, 2019-03-07:
Thank you for your reply, look forward to the new version and wish ESPEasy be better and better.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er