letscontrolit/ESPEasy #2374: ESP_Easy_mega-20190305_esp32test_1M8_partition keeps rebooting

berbergh, 2019-03-06:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

After uploading the firmware, the ESP32 keeps rebooting
Upload ESP_Easy_mega-20190301_esp32test_1M8_partition.bin works okay

### Steps to reproduce

1. You need a Wemos ESP32 board with ESP32 WROOM 32 on it
2. Upload ESP_Easy_mega-20190305_esp32test_1M8_partition.bin
3. 

### System configuration

Hardware:
You need a Wemos ESP32 board with ESP32 WROOM 32 on it

ESP Easy version: 
ESP_Easy_mega-20190305_esp32test_1M8_partition.bin

ESP Easy settings/screenshots: 

### Rules or log data

```
1193 : WIFI : Not configured in Station Mode!!: Spoon3
1197 : OTA  : Arduino OTA enabled on port 3232
1223 : EVENT: System#Boot
1446 : DS   : Temperature: 15.88 (28-ff-5-f0-a1-15-3-47)
1448 : EVENT: DS1#Temperature=15.88
1487 : DHT  : No Reading
2534 : BMx280 : Detected BME280
2561 : WD   : Uptime 0 ConnectFailures 0 FreeMem 237844 WiFiStatus 0
3386 : assertion "netif->hwaddr_len must be the same as ETH_HWADDR_LEN for etharp!" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/lwip/lwip/src/core/ipv4/etharp.c", line 1155, function: etharp_raw

Backtrace: 0x4008da70:0x3ffb1dc0 0x4008dca1:0x3ffb1de0 0x401767b3:0x3ffb1e00 0x401a77bd:0x3ffb1e30 0x401a78ad:0x3ffb1e60 0x401a7a1c:0x3ffb1e90 0x400f214d:0x3ffb1eb0 0x400f22e3:0x3ffb1ee0 0x40164d0c:0x3ffb1f00 0x40164f26:0x3ffb1f80 0x401698dd:0x3ffb1fb0 0x400905d9:0x3ffb1fd0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
⸮U35 : 
```

TD-er, 2019-03-06:
OK, that looks to be related to the new _Gratuitous ARP_ I added.
I will disable it for ESP32 for now until I found a way it is working on ESP32.

closed 2019-08-27 by TD-er

uzi18, 2019-08-27:
how this feature work?

TD-er, 2019-08-27:
It is an issue I came across while looking for an error I got on ESP32, and it is removed from ESP32 builds, so it is no longer relevant.
So that's why I closed it.

But just to give a very short introduction to Gratuitous ARP...
It is giving an answer to a question never asked.

For normal ethernet traffic, each switch/router etc. needs to know what MAC address is behind what port.
Since these tables in switches etc. are not infinite large, they have to renew their tables every now and then.
This is done via ARP packets.
So someone asks "who has 192.168.1.1" and the answer is also used by every other host or switch on the network to update its MAC tables.
If the ESP does not reply to such ARP requests, it may become impossible to send data (back) to the ESP, since the switches do not know how to route it.
So therefore we send Gratuitous ARP packets to answer "Mac AA:BB:CC:DD:EE:FF has 192.168.1.1" even though no-one asked for it.

PR #2574 (2019-08-29): [Gratuitous ARP] Only send ARP packets when IF is up
See this PR: xoseperez/espurna#1877  and discussion here: esp8266/Arduino#5998 (comment)

Fixes #2374  (ESP32 use of Gratuitous ARP)