letscontrolit/ESPEasy #2379: [HTU21D] Heats up on its own

jimmys01, 2019-03-08:
### Steps already tried...
Got an indepentand Unit-T thermometer fo sanity checking.
Found similar reports on the forum

### Summarize of the problem/feature request
After powering on a "cold" ESPeasy with a HTU21D I got a reading of 18.88c (61% humidity)
The Intependant meter shows 18.8c (59% humidity)
after 20 min the ESP reports 21.85c (47% humidity) , the intepentand meters reports 20.5c (58% humidity) (the esp reports 1.3c higher)
Also over 150+ rooms the temps seem a bit to high (up to 2c higher and this affects humitidy readings also who seem way way low). Checked with another thermometer inside the rooms as well. 

### Steps to reproduce
Power on an ESPeasy and check the temperature. 
Check again after some time. It will be 1-2c higher.
Replace the HTU with a fresh one and the results will be the same, normal temp and humitidy reading and after 10 min it will go up 1-2c

The datasheet mentions a heater that would do that, maybe we can explicitly disable te heater in the code? (even  if te datasheed sais it defaults to 0). So far the plugin does not mention code lines for a heater.

TD-er, 2019-03-08:
Do you have a picture of how the sensor is being mounted in the enclosure of the ESP?
You can also try the new "eco mode" of ESPeasy to see if it lowers the offset.

At this moment I don't have such a sensor here, but I see similar behavior with the BME280.

It is possible to correct for humidity readings if you know the temp offset is constant.
Simply compute the dew point, correct the temp and use temp + dew point to compute the humidity.
This works only if the offset is constant.
e.g. a sensor in a small enclosure with the ESP will produce a fixed amount of heat.
The enclosure can be considered similar to a heatsink, which has a constant heat dissipation characteristic expressed in Kelvin/Watt.
So just to check if it is the ESP which warms the enclosure and thus the sensor, you can test by activating the 'eco mode'.

jimmys01, 2019-03-08:
For the purposes of this test I did the sensor is far from the ESP in the open air.

chemmex, 2019-03-10:
@TD-er, 

> You can also try the new "eco mode"

is this somehow documented or described somewhere?

Domosapiens, 2019-03-10:
@chemmex 
see [https://github.com/letscontrolit/ESPEasy/pull/2370#issuecomment-468863720](url)

chemmex, 2019-03-10:
@Domosapiens, thanks! Only have to note that url links in your comment are broken

TD-er, 2019-03-10:
I will add a link to that documentation on the web interface too, since it was asked a few times before. (so thanks for asking :) )

jimmys01, 2019-03-15:
Each measurment is 30 seconds appart. Those are taken with the use of another library and pretty much mimic the problem I see also wit the EasyESP
```
Temp: 20.37 C		Humidity: 59.10 %
Temp: 20.69 C		Humidity: 57.86 %
Temp: 21.04 C		Humidity: 56.48 %
Temp: 21.40 C		Humidity: 55.47 %
Temp: 21.74 C		Humidity: 54.22 %
Temp: 22.08 C		Humidity: 55.75 %
Temp: 22.36 C		Humidity: 52.18 %
Temp: 22.54 C		Humidity: 51.30 %
Temp: 22.72 C		Humidity: 51.04 %
Temp: 22.93 C		Humidity: 50.18 %
Temp: 23.09 C		Humidity: 50.12 %
Temp: 23.21 C		Humidity: 49.54 %
Temp: 23.32 C		Humidity: 49.02 %
Temp: 23.49 C		Humidity: 48.97 %
Temp: 23.61 C		Humidity: 48.57 %
Temp: 23.70 C		Humidity: 48.36 %
Temp: 23.76 C		Humidity: 48.27 %
Temp: 23.81 C		Humidity: 48.31 %
Temp: 23.90 C		Humidity: 48.02 %
Temp: 23.98 C		Humidity: 47.83 %
Temp: 24.03 C		Humidity: 47.67 %
Temp: 24.09 C		Humidity: 47.69 %
Temp: 24.11 C		Humidity: 47.61 %
Temp: 24.13 C		Humidity: 47.52 %
Temp: 24.13 C		Humidity: 48.04 %
Temp: 24.15 C		Humidity: 47.67 %
Temp: 24.18 C		Humidity: 47.76 %
Temp: 24.19 C		Humidity: 47.57 %
Temp: 24.25 C		Humidity: 47.78 %
Temp: 24.32 C		Humidity: 47.58 %
Temp: 24.35 C		Humidity: 47.59 %
```
So either the library has issues or the modules I bought (200 of them) are problematic..

TD-er, 2019-03-15:
I don't have this sensor yet, so I cannot test it myself.

chemmex, 2019-03-15:
Most of sensors self-heat. I noticed very similar behaviour with DS18B20, BMP/BME series, HDC1080 and HTU21D. Usually, 1-2 degrees shift is expected up because of too frequent measurements or improper casing. DS18B20, for instance, shows more or less real results only being polled once a minute in open the air .

jimmys01, 2019-03-15:
I poll the HTU21 every 5 minutes and still i see the same result. The sensors are not in any kind of case and away from the esp.

chemmex, 2019-03-15:
is there an LDO on the pcb?

bojanul, 2019-03-19:
I get good readings from BME280 and Si7021 only if I put ESP to sleep for 5 minutes and then wake it up for 10 sec.