letscontrolit/ESPEasy #2381: Gyro - MPU 6050 Device  broken after mega-20190121

thomastech, 2019-03-08:
### Problem Summary
Releases after mega-20190121_test_ESP8266_4096 have broken the MPU6050 plugin. 
Verified not working:
 ESP_Easy_mega-20190226_test_core_242_ESP8266_4M
 ESP_Easy_mega-20190301_test_core_242_ESP8266_4M
 ESP_Easy_mega-20190305_test_core_242_ESP8266_4M

Issues:
 MPU 6050 Gyro Movement detection no longer functions. 
 Range Acceleration (X/Y/Z) detection values don't update.

### Expected behavior
Vibration exceeding thresholds should cause the Movement detection state to change.
Acceleration values should update at interval time.
### Actual behavior
Vibration does no cause the Movement detection state to change.
Acceleration values are not updating at interval. Stuck on high value.
But if the device is re-saved the Acceleration value begins to update.
Rebooting does not fix problem. Re-installing mega-20190121 is a reliable workaround.
### System configuration

Hardware: NodeMCU module. 
 ESP Chip ID:  8752631 (0x858DF7)
 Flash Chip ID:  Vendor: 0xEF Device: 0x4016
Working Build: 20103  - Mega
Libraries:  ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support  
GIT version: mega-20190121  
Plugins: 74 [Normal] [Testing]  
Build time: Jan 21 2019 03:16:28  
Binary filename: ESP_Easy_mega-20190121_test_ESP8266_4096.bin 

ESP Easy version: 
Releases after mega-20190121_test_ESP8266_4096 are broken

### ESP Easy settings/screenshots: 
Working version shown (mega-20190121)
[image]
[image]
[image]

TD-er, 2019-03-08:
Hmm that's too bad, since I've made quite some changes to make this plugin use less memory when not used.
I will revert it and markt it for later to have a proper look at it.
I will also order one to be able to test it myself.

TD-er, 2019-03-09:
I added a commit to revert the last changes.
It will be working again in the next nightly build when this is merged.

thomastech, 2019-03-11:
Installed latest version (ESP_Easy_mega-20190311_test_core_242_ESP8266_4M.bin) and issue is resolved. Ticket closed.
- Thomas

closed 2019-03-11 by thomastech