letscontrolit/ESPEasy #2382: Pulse counter get a false pulse every 11h

Kurairaito, 2019-03-09:
Hi,

I use a wemos D1 mini, with ESPEasy_mega-20190301 TESTING.
I have 2 Devices configured : 
- P068 Environment - SHT30/31/35 [TESTING]
- P003 Generic - Pulse counter

The pulse counter use Gpio15 (D8) which have a pull down hardwired on it.
The pin D8 Is set as input in the hardware tab.
The shield with the SHT doesn't use D8 (it use I2C).
No other cable are wired to D8 for testing purposes. 

I have 1 Controler Http where data are sent (Jeedom) and when looking if everything worked fine, I saw that I got false positive very repeatedly, every 11h.

[image]

> "DateTime";"[Placard][ESP_Easy_cptEDF][Total] "
> "2019-03-04 21:00:00";0
> "2019-03-05 04:00:00";0
> "2019-03-05 05:00:00";0,786667
> "2019-03-05 06:00:00";2
> "2019-03-05 17:00:00";2
> "2019-03-05 18:00:00";2,650833
> "2019-03-05 19:00:00";4
> "2019-03-06 05:00:00";4
> "2019-03-06 06:00:00";5,650833
> "2019-03-06 07:00:00";6
> "2019-03-06 18:00:00";6
> "2019-03-06 19:00:00";6,286667
> "2019-03-06 20:00:00";10
> "2019-03-07 04:00:00";10
> "2019-03-07 05:00:00";10,156667
> "2019-03-07 06:00:00";11
> "2019-03-07 17:00:00";11
> "2019-03-07 18:00:00";11,25
> "2019-03-07 19:00:00";15
> "2019-03-08 04:00:00";15
> "2019-03-08 05:00:00";15,166667
> "2019-03-08 06:00:00";16,911667
> "2019-03-08 07:00:00";17
> "2019-03-08 17:00:00";17
> "2019-03-08 18:00:00";17,3125
> "2019-03-08 19:00:00";19
> "2019-03-09 07:55:00";19
> "2019-03-09 08:00:00";20,8125
> "2019-03-09 08:05:00";21,75
> "2019-03-09 08:10:00";22
> "2019-03-09 09:40:00";22
> "2019-03-09 09:50:00";0

I removed duplicate value, Jeedom automatically compact data, so don't pay attention to non integer value, it's just to see hours when steps occurs.

Unfortunately, I unpluged it to update it from scratch to try old version and lost the log. I will update this ticket if I found an old version which works fine, but every test will last at least 11h...

TD-er, 2019-03-09:
Please also try to log the uptime of the node.
Just to be sure this 'false count' is not caused by a reboot of the node.

Kurairaito, 2019-03-09:
I flashed the mega-20190106 TESTING, and added a third device Generic - System Info to report the uptime.

Kurairaito, 2019-03-09:
additionnal note, I saw in the source that the pin used is configured with a pullup. this might be conflicting with my gpio which habe a pulldown wired on the board.

`pinMode(CONFIG_PIN1, INPUT_PULLUP);`

DittelHome, 2019-03-09:
On my side, i get "Hardware Watchdog's". Max uptime is 15 hour...
After disabling the pulse counter, the ESP8266 stays up.

TD-er, 2019-03-09:
> On my side, i get "Hardware Watchdog's". Max uptime is 15 hour...
> After disabling the pulse counter, the ESP8266 stays up.

Can you post the timing stats of one of such nodes?
Make sure to view the timing stats at least once right after boot, so the worst timings are not from right around boot time (viewing does reset the stats) and then again after a few hours or so.
Also some info about the WiFi RSSI and the (new) WiFI settings in the Advanced settings page (at the bottom)

Kurairaito, 2019-03-10:
No pulse detected after 19h uptime. I will test again in same condition the lastest version to confirm it was an hardware bug.

DittelHome, 2019-03-10:
> Can you post the timing stats of one of such nodes?
Make sure to view the timing stats at least once right after boot, so the worst timings are not from right around boot time (viewing does reset the stats) and then again after a few hours or so.
Also some info about the WiFi RSSI and the (new) WiFI settings in the Advanced settings page (at the bottom)

Sorry, but i am back to an older version. At the moment a cant update the esp. The ESP is needed for measurement. (for two weeks)
The WIFI settings were:
Force WiFi B/G was set
Periodical send Gratuitous ARP was set
RSSI -68db

regards Klaus

Kurairaito, 2019-03-10:
After 13h15 uptime, with the new version, still no false positive. I almost sure that the double pullup/pulldown + something that append every 11h on the chip caused my issue. @DittelHome you should open a ticket for your issue, as I didn't get any watchdog in my case.

TLDR: my "bug" was not a bug.

closed 2019-03-10 by Kurairaito