letscontrolit/ESPEasy #2383: LongPulse command output is missing on device page

DittelHome, 2019-03-10:
Software used: 
ESP_Easy_mega-20190305_normal_ESP8266_4M.bin
Hardware used:
ESP8266 Mini pro
Problem:
The command "LongPulse,0,0,900" is not displayed in the device page. In device page a have defined a input switch. Version 20181120 is working correct. The output gpio is working itself, only the display does not work. I propagate the state to my smarthome system, with actuell firmware, i get no more states reported.
The state is always 0.
best regards Klaus

TD-er, 2019-03-10:
Can you add some screenshots?
I don't really understand were on the device page things are not displayed like they should.
So perhaps a screenshot of how it should be, would be perfect.

DittelHome, 2019-03-10:
Here it is a picture:
[image]
They value for example of Lueftung keeps to 0, while longpulse is running.

[image]

regards Klaus

chunter1, 2019-03-13:
Sounds like the issue i posted under #2269.

giig1967g, 2019-03-13:
I will look into it.

giig1967g, 2019-03-14:
@DittelHome @chunter1 
the problem is a known issue: the GPIO/PCF/MCP plugins as are designed now monitor INPUT events (infact the name is SWITCH INPUT), not OUTPUT events. You are using the plugin in a different way as it was designed for. It works but not everything is perfect.
You can easily avoid defining your output pin in the device page and just assign its value in the rules.
You'll save memory.

Long story short, to see the current RELIABLE value of an OUTPUT pin you should go into "Pin State Buffer" page and you will see the state changing.

I will see in the meantime if I can update the value in the Device page too.

DittelHome, 2019-03-15:
> You can easily avoid defining your output pin in the device page and just assign its value in the rules.
You'll save memory.

This does also not work.
I tried the follwing:

```
on System#Boot do
      monitor gpio,13
endon

 on GPIO#13=0 do
   publish /Heizraum-ESP/HZ.wr.vent,0
 endon
 on GPIO#13=1 do
   publish /Heizraum-ESP/HZ.wr.vent,1
 endon
```
The rule does not fire with LongPulse, i only get:

`SW : GPIO 13 Pulse set for 90 sec`

If i use GPIO,13,0 instead of LongPulse i get:

```
908029: ACT : GPIO,13,0
908032: SW : GPIO 13 Set to 0
908090: SW : GPIO=13 State=0 Output value=0
908098: EVENT: HZ.wr.vent#HZ.wr.vent=0.00
908206: EVENT: GPIO#13=0
908257: ACT : publish /Heizraum-ESP/HZ.wr.vent,0
908272: Command: publish
```

regards Klaus

giig1967g, 2019-03-15:
I will have a look then more deeply.

TD-er, 2019-03-15:
Maybe this PR on the playground  is also applicable here?
https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/131

giig1967g, 2019-03-15:
will check

TD-er, 2019-03-15:
I have also been thinking about our "GPIO handler" switch discussion a while ago.
I think we should just create new plugins for the specific use cases and should not focus too much on the existing compatibility.
The existing plugins are way too complex and any change will lead to lots of unforeseen issues.

So maybe we should just start creating a GPIO-out and GPIO-in plugin, which can also use the expanders.
But first just a single GPIO-in and -out plugin based on the internal GPIO pins.
Then we know how to get it right and will know what is  needed from the central managed GPIO-handler we must make.
From that we can then start to extend to external GPIO pins and later add other functionality (in the form of other plugins) which are using the GPIO-handler.

So in short, we should stop focus on repairing all existing uses and mis-uses of the existing P001 and P003, since that will take forever and will only give us something that is even more complex than we have now.

giig1967g, 2019-03-15:
I agree.

DittelHome, 2019-03-18:
> I think we should just create new plugins for the specific use cases and should not focus too much on the existing compatibility.

Is the use of LongPulse a specific use case ?
GPIO command trigger rules, LongPulse not.

TD-er, 2019-03-18:
Not sure yet.
The thing is, it has been bothering me a lot for a while that one of the most basic plugins (P001 switch) has become one of the most complex plugins which thus also lead to various bug reports.
So my first goal is to keep it simple.

The LongPulse command is now integrated into the switch plugin, but I see no reason why it cannot be a special command with a start and end action (to switch some plugin).
So I guess the LongPulse command has to be moved away from the switch plugin, to a more generic place. Then it would also be perfectly capable to act on GPIO extenders.
But first let the current P001_Switch and some new to be created switch input and switch output plugin co-exist.
If that's working we can perhaps move the LongPulse and other commands, or think about keeping the P001 implementation for these commands.

closed 2023-09-03 by TD-er