letscontrolit/ESPEasy #2384: esp32 + lan8720  Ethernet communication

qiudecai, 2019-03-10:
I use esp32 + lan8720 used in Ethernet communication, can realize the webserver access, but you can't and DOMOTICZ communication.

uzi18, 2019-03-10:
please explain what you need?

qiudecai, 2019-03-10:
support esp32 Ethernet communication

TD-er, 2019-03-10:
Do you have some examples of how this is connected, what you already tried and almost as important... where you've bought your LAN adapter for the ESP. (link/type/model/etc)

qiudecai, 2019-03-10:
I am referring to this
https://blog.csdn.net/naisu_kun/article/details/86710990

TD-er, 2019-03-10:
Ah, that looks like a useful page too to see how it must be programmed in Arduino environment.
I'm glad the online translators are working so well these days ;)

qiudecai, 2019-03-10:
/*********************************************************************************************\
 * SETUP
\*********************************************************************************************/
void setup()
{
#ifdef ESP8266_DISABLE_EXTRA4K
  disable_extra4k_at_link_time();
#endif
  WiFi.persistent(false); // Do not use SDK storage of SSID/WPA parameters
  WiFi.setAutoReconnect(false);
  setWifiMode(WIFI_OFF);
  lowestFreeStack = getFreeStackWatermark();
  lowestRAM = FreeMem();

  Plugin_id.resize(PLUGIN_MAX);
  Task_id_to_Plugin_id.resize(TASKS_MAX);

  checkRAM(F("setup"));
  #if defined(ESP32)
    for(byte x = 0; x < 16; x++)
      ledChannelPin[x] = -1;
  #endif

  Serial.begin(115200);
  // Serial.print("\n\n\nBOOOTTT\n\n\n");

  initLog();

#if defined(ESP32)
  WiFi.onEvent(WiFiEvent);
//*****************************
//初始化 以太网接口
  ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); //启用ETH
  while(!((uint32_t)ETH.localIP())) //等待获取到IP
  {

  }
//  Serial.println("Connected");
//  Serial.print("IP Address:");
//  Serial.println(ETH.localIP());
//******************************

#else
  //无线网络事件处理程序 WiFi event handlers
  stationConnectedHandler = WiFi.onStationModeConnected(onConnected);
	stationDisconnectedHandler = WiFi.onStationModeDisconnected(onDisconnect);
	stationGotIpHandler = WiFi.onStationModeGotIP(onGotIP);
  APModeStationConnectedHandler = WiFi.onSoftAPModeStationConnected(onConnectedAPmode);
  APModeStationDisconnectedHandler = WiFi.onSoftAPModeStationDisconnected(onDisonnectedAPmode);
#endif

qiudecai, 2019-03-10:
#define DEFAULT_PROTOCOL    0                   // Protocol used for controller communications
//   0 = Stand-alone (no controller set)
//   1 = Domoticz HTTP
//   2 = Domoticz MQTT
//   3 = Nodo Telnet
//   4 = ThingSpeak
//   5 = OpenHAB MQTT
//   6 = PiDome MQTT
//   7 = EmonCMS
//   8 = Generic HTTP
//   9 = FHEM HTTP

#define DEFAULT_PIN_I2C_SDA              4
#define DEFAULT_PIN_I2C_SCL              5

#define DEFAULT_PIN_STATUS_LED           -1
#define DEFAULT_PIN_STATUS_LED_INVERSED  true

// --- Advanced Settings ---------------------------------------------------------------------------------
#if defined(ESP32)
  #define USE_RTOS_MULTITASKING
  //**************************
  //ESP32以太网
  #include "ETH.h" //引用以使用ETH
  #define ETH_ADDR        1  //LAN8720写0或1、TLK110写31；
  #define ETH_POWER_PIN  -1  //设置ESP32某管脚，该管脚可以用来管理LAN8720供电，使能时输出高电平，失能时为低电平，写-1则不使用；
  #define ETH_MDC_PIN    23  //mdc管脚编号，写-1则不使用；
  #define ETH_MDIO_PIN   18  //mdio管脚编号，写-1则不使用；
  #define ETH_TYPE

philipptrenz, 2019-12-15:
Hey guys, 

I want to use the ethernet connection of the [Olimex ESP32-POE](https://www.olimex.com/Products/IoT/ESP32/ESP32-POE/open-source-hardware). It uses a LAN8710 chip (as the [schematic](https://github.com/OLIMEX/ESP32-POE/blob/master/HARDWARE/ESP32-PoE-hardware-revision-C1/ESP32-PoE_Rev_C_Color.pdf) show and should be compatible to the LAN8720.
 
Olimex provides a Arduino code example [here](https://github.com/OLIMEX/ESP32-POE/tree/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino), so it seems to use the standard `ETH.h` from arduino-esp32 (see [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/ETH.h)).

What has to be done to integrate the ethernet feature into ESPEasy? I'm willing to help 😉

TD-er, 2019-12-15:
It for sure is an interesting option, but I am afraid it may involve quite a huge code change.
Right now we use the `WiFi` object directly in a lot of places.
To make it more transparent, we should introduce an abstraction layer for network.
Compared to when this issue was created (March) the WiFi related code has changed quite a bit, so maybe now it is less work compared to 9 months ago, but still I think it is significant.

On the other hand right now we only have these interacting with the network:
- Webserver
- Some commands (code has changed a lot for the commands)
- Controllers

First things we need to do is have a look where we use the WiFi object directly and where we create WiFiClient objects.
There is also some clients used for TCP and UDP specific code and maybe I forgot something?

Is the eth module you need already supported in the esp32 Arduino library, or do we need other libraries too?
I also saw some activity about adding ethernet adapter support in the esp8266/Arduino repo. Maybe these do use the same interfaces?

philipptrenz, 2019-12-15:
Thanks for the quick replay, @TD-er! An abstraction layer sounds good as a long term goal.

I'm not very into the code base of the esp32 Arduino library, but for me [this](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino) looks like the interface of `ETH.h` is almost identical to `WiFi.h`. It even uses the `WiFi.onEvent()` for events. For the ESP32-POE only the following has to be configured:

```
#define ETH_CLK_MODE ETH_CLOCK_GPIO17_OUT
#define ETH_PHY_POWER 12

#include <ETH.h>

[...]
void setup()
{
  Serial.begin(115200);
  WiFi.onEvent(WiFiEvent);
  ETH.begin();
}
```
For me and probably a majority of other users having either WiFi or ethernet would be just fine. So maybe having the ability to switch from initial WiFi Adhoc network to an ethernet-only mode would be an option without huge code changes?

From my perspective, the things that have to be configurable are `ETH_PHY_ADDR` , `ETH_PHY_TYPE`, `ETH_PHY_POWER`,  `ETH_PHY_MDC`, `ETH_PHY_MDIO` and `ETH_CLK_MODE` and they could be adjustable within the web server before switching to ethernet-only mode.

What do you think, @TD-er?

TD-er, 2019-12-15:
What I'm a bit worried about is whether the uses of `WiFiClient` are still the same.
I don't want almost every use of these be wrapped in lots of `#ifdef` statements.
It can be done in an abstraction layer of course, but still if that layer is almost completely different between ESP82xx and ESP32 then it will result in a HUGE change of code.

philipptrenz, 2019-12-15:
Ah, I understand. I don't know my way around well enough, but maybe you want to take a quick look [at this demo](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_WebServer_Demo/ESP32_PoE_WebServer_Demo.ino) for the ESP32-POE?

philipptrenz, 2019-12-15:
Oh this seems not to use ethernet, my fault!

philipptrenz, 2019-12-15:
I just saw that `WiFiClient` is also used in the [above mentioned example](https://github.com/OLIMEX/ESP32-POE/blob/master/SOFTWARE/ARDUINO/ESP32_PoE_Ethernet_Arduino/ESP32_PoE_Ethernet_Arduino.ino) with ethernet, here's a snippet:

```
void testClient(const char * host, uint16_t port)
{
  Serial.print("\nconnecting to ");
  Serial.println(host);

  WiFiClient client;
  if (!client.connect(host, port)) {
    Serial.println("connection failed");
    return;
  }
  client.printf("GET / HTTP/1.1\r\nHost: %s\r\n\r\n", host);
  while (client.connected() && !client.available());
  while (client.available()) {
    Serial.write(client.read());
  }

  Serial.println("closing connection\n");
  client.stop();
}
```

TD-er, 2019-12-15:
Looks like they have made it quite abstract.
That would mean only the WiFi code has to be changed, which is just in a very limited number of files.

I will order one of these, although I have no idea when I will have time for it.

Do you have any idea what PoE standard is supported?
I do have a few switches with active PoE, but I know not all "PoE capable" devices support this active protocol.

philipptrenz, 2019-12-15:
Awesome! If you need support, I can help you (at least) until New Years Eve.

They say on their website:

> The [used] Si3402-B chip is IEEE 802.3-compliant, including pre-standard (legacy) PoE support.

So active 802.3af should just be fine.

TD-er, 2019-12-15:
> Awesome! If you need support, I can help you (at least) until New Years Eve.

Well I have to order the unit myself and since I am about to move to a new (temporary) address in the 2nd week of 2020, I cannot do a lot until I've moved.
There is a slightly complicating factor, in the fact that I don't know the official address as it was just changed last week (administrative change, the buildings are already finished), so I cannot order stuff to send over there before I know the address.

This means it is very unlikely I have a unit to test in January.

philipptrenz, 2019-12-15:
Oh okay 😅
If you want, you can give me some hints where the changes have to happen, so I can tinker around over christmas and maybe having a PR ready until your unit arrives 😉

uzi18, 2019-12-15:
@TD-er esp32 has got native mac/mdii interface but curious how they want to support it on esp8266

TD-er, 2019-12-15:
> or even to your office/work directly

Office/work address is the same as my home address ;)

Budman1758, 2019-12-16:
> Office/work address is the same as my home address ;)

Does that mean you never leave home or never leave work??  :)

uzi18, 2019-12-16:
or both =)

TD-er, 2019-12-16:
Well, we have to move in the next weeks, so not sure how that qualifies :)

mobrembski, 2020-03-22:
I've created a basic implementation, it shows Ethernet status on Sysinfo and Root page, and currently only DHCP is supported. Based on Olimex ESP32-PoE board.

https://github.com/letscontrolit/ESPEasy/pull/2962

uzi18, 2020-03-22:
@mobrembski what is your lan chip on board? Is it LAN8710 or 20? 
Maybe it is possible to easy connect eth module to esp32.

mobrembski, 2020-03-22:
LAN8710, works pretty well ;)

DerFossiBaer, 2021-06-24:
I will follow up here.

I just received an  Olimex ESP32-PoE rev. It was easy to flash ESP_Easy_mega_20210503_normal_ESP32_4M316k_ETH-factory.bin. Im Able to login into with wifi-connection. The device powers up via poe - fine.

I took a manual eth config and tried also dhcd. the device makes no Connection throug eth. What is my mistake? what i have to check? That there is no connection i saw in the unfi controller, the port doesn't lights up like normal. its just the sign for poe there.

TD-er, 2021-06-24:
On the reset to factory default is a selector to pre-configure it for a device, including the Olimex ESP32 POE.
Or you select these settings on the Hardware tab:
[image]

[image]

Oh and the board is quite stable:
[image]

DerFossiBaer, 2021-06-24:
Thanks a lot. Thats helping.

No its working fine. Did you made some experience with long time usage? The ships semms to be a bit warm?

VG Fossi

DerFossiBaer, 2021-06-24:
Is it also possible with this chip to add some i2c devices - linke bosch 280/680?

TD-er, 2021-06-24:
Well I have it running on several units for roughly a year now and the chip does work fine.
I haven't noticed it to get warm, but I also never touched it as I can remember.

The LAN8720 chip is however not really ESD protected, so a static discharge can kill it. (been there done that)

If you set the I2C pins as I showed, you can for sure use I2C devices with ESPEasy.
A rough estimate is that about half of the plugins is for I2C devices.

TD-er, 2021-08-12:
This issue is no longer an issue as far as I know, so close it now.

closed 2021-08-12 by TD-er

sigma415, 2024-12-17:
For pure ETH functionality of ESP32-POE the ports GPIO13 and GPIO16 are not needed. 
They can be used for other purposes, if I2C is not used.

[image]

TD-er, 2024-12-17:
Not sure why you have arrows drawn pointing at the multiplexer options?

sigma415, 2024-12-17:
Because your screenshot above (https://github.com/letscontrolit/ESPEasy/issues/2384#issuecomment-867817322) looks like, that GPIO13 and GPIO16 have to be configured for ESP32-ETH. 
This has blocked me two ports which I can use now :)

TD-er, 2024-12-17:
But what you're pointing at is about an optional I2C multiplexer, not about the I2C pin definition.
That's at few lines up on the same page (not shown on your screenshot)

I have to look at the schematics of the board discussed, to see if there isn't anything connected to it.
For example I2C pins typically are pulled-up with some 10k resistor (can be lower), so there may still be caveats to these pins if you want to use them for something else.

Also GPIO-16 is a bit tricky as it is also used for PSRAM for those boards that have PSRAM (typically only WROVER modules for the ESP32-classic, not often seen for the ESP32-classic)
However the ESP SDK code still may check for PSRAM presence which may result in quite high frequencies (40 - 80 MHz) every now and then but at least during boot.

sigma415, 2024-12-18:
Oops, sorry, you are completely right, I got the wrong part of the screen.  
I wanted to show, that I2C Interface SDA and SCL must not be configured to GPIO13 and GPIO16 and can be used for something else.
Sorry for the confusion.
Here the correct screen:

[image]

sigma415, 2024-12-18:
After using GPIO16 to control an output and GPIO13 for 1-wire-sensor input I I fully stepped into the pitfall discussed here: https://www.letscontrolit.com/forum/viewtopic.php?t=10532  :(

Should have better read your post above for the tricky GPIO16  ;)

The solution was to change to your daily build with LitteFS filesystem. ->  https://www.letscontrolit.com/forum/viewtopic.php?p=71982#p71982
Thanks for your help :)