letscontrolit/ESPEasy #2385: IRremoteESP8266 was updated

plin2, 2019-03-10:
### Summarize of the problem/feature request
The IRremoteESP8266 was updated to detect/send Samsung 36 bit codes (new type: Samsung36)

### Expected behavior
When will ESPeasy include the new remote type?

TD-er, 2019-03-10:
@jimmys01 is usually the one to add new pull requests for newer versions.

glmnet, 2019-03-10:
I need support for tcl and repeat codes too. I see irremoteesp8266 has option to set repeat but reviewing code espeasy does not
I quite don’t understand the target user. Having an average user send a hex code they have to calculate themselves (with a different code per ac) seems too challenging

TD-er, 2019-03-10:
It seems like there is also an ESPeasy plugin for that [heatpump IR lib](https://github.com/ToniA/arduino-heatpumpir)

jimmys01, 2019-03-10:
I am quite far from my Pc now. The day after tomorrow I will be able to update the library. Code repeats are handled from the library if needed in a transparent way, I haven't came across a situation that they are needed yet. On how and if we should implement code repeats, ido not know..

TD-er, 2019-03-10:
Not sure if I understand the need for 'code repeats' well enough by just reading your posts.
Is it something like what's needed on Panasonic TV's to turn off the television? (you have to keep the power button pressed on the remote)
I've also seen it on some beamers and some need 2 or 3 presses with a certain interval.

Is it something like that?

uzi18, 2019-03-10:
maybe it is about raw2 repeat capability

glmnet, 2019-03-10:
Sorry not being so explicit. 
Coolix needs 1 repeat on my ac in order for it to work reliable. 
May be this should be done by the irremote lib de facto. but they say: send the query with 1 repeat. 
So a dirty (or not so) May be to hard code a repeat for this one. 
I’m working on a custom plugin also not related to ac so not a big issue for me to go on my own but sharing my experiences here. 
The heatpumpir is like doing mostly the same than this “official” plugin so I don’t see any advantage in maintaining both. Beside it doesn’t support rx

uzi18, 2019-03-10:
you can use rules here, and remotely generate just event to trigger rules

jimmys01, 2019-03-12:
@glmnet 
> Coolix needs 1 repeat on my ac in order for it to work reliable.

probably related to markszabo/IRremoteESP8266#579 and fixed. Check when the next nightly will be up

glmnet, 2019-03-12:
@jimmys01  thanks for commenting. I am well aware of that thread.
Please read everything carefully. Let me know if you have any question.
I'll keep watching IR TX and your commits.

plin2, 2019-03-12:
@jimmys01 : When will the current version of IRremoteESP8266 will be included in the nightly build?

jimmys01, 2019-03-13:
> @jimmys01 : When will the current version of IRremoteESP8266 be included in the nightly build?

TD-er is the boss he makes the nightly builds

TD-er, 2019-03-13:
@plin2 Usually PR's are not merged immediately, so people can have a look at the code and make some comments on it.
This often leads to some extra added commits, which thus may prevent new issues and/or better code.

plin2, 2019-03-13:
mmh, ok. How can I compile an ESPeasy binary with infrared support? Is there a prepared script? Which parameters would I need to set if I compile the sketch in Arduino?

glmnet, 2019-03-13:
@plin2 follow this guide https://www.letscontrolit.com/wiki/index.php/Tutorial_building_and_uploading_with_platformio
later when you hit F7 make sure you write build (or upload) also normal, ir and 4m are the keywords needed  (there are too many build configurations)
it works with vscode too, however I still don't know how to make it like the ino file extension.

jimmys01, 2019-03-13:
This is a .ino for you to test with the latest IR library
[ESPEasy.ino.zip](https://github.com/letscontrolit/ESPEasy/files/2962847/ESPEasy.ino.zip)

plin2, 2019-03-13:
@jimmys01: 
a) thanks
b) loaded the binary file
tested sending a couple of the new Samsung36 commands
worked fine with my Bluray player Samsung BD-F6500 (remote control ak59-00167a)
thumbs up from me

jimmys01, 2019-03-14:
@glmnet 
> Coolix needs 1 repeat on my ac in order for it to work reliable.

There is going to be a fix for that, see IRremoteESP8266 commit 637

glmnet, 2019-03-14:
@jimmys01  thanks for the heads up. the fix is for Coolix but it does not address the repeat issue.

jimmys01, 2019-03-14:
IMO repeats should be handled from the library until we find a way to transmit this commands by JSON, like `{"Protocol":"PANASONIC","Bits":32,"Data":"0x100A0A1","REPEATS":"1"} (kinda like the tasmota way)`. 
This will enable us also to do the AC model, on/off, set temperature , mode, later on when someone feels like tackiling thiscoding challange. example for AC command:
` {"Protocol":"GREE","Temp":32,"Mode":"HEAT","POWER":"ON","SWING":"OFF"}`

glmnet, 2019-03-14:
What about adding a hack:
`
if (IrType.equals(F("coolix2")))             Plugin_035_irSender->sendCOOLIX(IrCode, kCoolixBits, 1);`

jimmys01, 2019-03-14:
Yes sure, but this particular problem will be fixed as soon as IRremoteESP8266 commits the changes for having default 1 repeat for the coolix protocol (`sendCOOLIX(IrCode)` will default to `sendCOOLIX(IrCode, kCoolixBits, 1)`). I will also push it to the already overcrowded push i have made.
Also by adding a `coolix2` there will be no documentation for this change... I believe that i should start writing a documentation anyway... But time is scarse I am afraid..

plin2, 2019-03-16:
FYI, IRremoteESP8266 v2.5.6 has just been release which includes the changes/improvements mentioned.

jimmys01, 2019-04-06:
@glmnet the pr #2435 includes the repeats. Also check the PRs at markszabo/IRremoteESP8266 because he has just added the AC abstraction functionality we where talking about, perhaps you want to take over and code that in the IR TX plugin. As I propose to be done above
```
 example for AC command:
{"Protocol":"GREE","Temp":32,"Mode":"HEAT","POWER":"ON","SWING":"OFF"}
```
Any thoughts?

jimmys01, 2019-06-12:
> @glmnet the pr #2435 includes the repeats. Also check the PRs at markszabo/IRremoteESP8266 because he has just added the AC abstraction functionality we where talking about, perhaps you want to take [...]

implemented in #2490

jimmys01, 2019-06-13:
So in IRMQTTServer.ino perhaps this sould change:

```
#if SEND_COOLIX
    case COOLIX:  // 15
      if (bits == 0)
        bits = kCoolixBits;
      irsend->sendCOOLIX(code, bits, repeat);
      break;
#endif
```

To 

#if SEND_COOLIX
    case COOLIX:  // 15
      if (bits == 0)
        bits = kCoolixBits;
      if (repeat== 0)
        bits = kCoolixDefaultRepeat;
      irsend->sendCOOLIX(code, bits, repeat);
      break;
#endif
```

closed 2023-09-03 by TD-er