letscontrolit/ESPEasy #2386: How to turn off automatic plugin disabling?

agsochi, 2019-03-10:
Hi. I use many wemos d1 mini with Generic - MQTT Import, Light/Lux - TSL2561, Switch input - Switch plugins.
Some times this plugins become disabled and i need to turn it again.
How to turn off automatic plugin disabling?

TD-er, 2019-03-10:
It is only disabled when it crashes a lot during startup.
In short:
If the node is crashing > 10 times without completing a full boot sequence, it will then start to disable a single plugin or controller on every new boot.
So it will disable the first plugin on the 11th boot, 2nd plugin on 12th.. and so on, until it will complete a full boot sequence.

This means that if the plugin was disabled and wasn't before, it was very likely the cause of the crashes.
To simply test it, you can send another reboot command to it to check.
This disabling is only done in memory, not in the stored settings. But as soon as you save something, it will be storing this disabled state to the settings.
So if you encounter this, you can try to reboot it without saving anything.

What plugin is being disabled? The MQTT import?

agsochi, 2019-03-11:
> What plugin is being disabled? The MQTT import?
This plugins: Generic - MQTT Import, Light/Lux - TSL2561, Switch input - Switch plugins

TD-er, 2019-03-11:
This plugin is generating 2 events which can be used in rules:

- `MQTTimport#Connected`
- `MQTTimport#Disconnected`

So you might want to start a timer at boot and if it is connected clear the timer.
If the timer reaches its end, you may trigger a reboot.
Just make sure it does have enough time between reboots from the rules or else you may not be able to login and disable it when it keeps rebooting.

Also you may want to look into why it causes all these reboots.

TD-er, 2019-03-11:
What build version are you using?
How much free memory do you have?
Could it be the MQTT broker is unreachable?
What are the MQTT controller settings (you need to have a MQTT controller defined to use the MQTT import since it is using the settings of the first active MQTT controller)

agsochi, 2019-03-11:
I use many versions to test it from mega-20181203 to latest, same behavior. 
This is averave on all modules:
```
Free Mem:	14376 (12856 - sendContentBlocking)
Free Stack:	3584 (784 - LoadTaskSettings)
```
MQTT broker is eternal (Ubuntu x86, Mosquitto).
MQTT controller settings on all modules are same (first and only 1 controller):
[image]

TD-er, 2019-03-11:
Well, for starters, you can tweak the default settings a bit when using a broker on the local network.
For example the 'Minimum Send Interval' can quite often be set to 10 msec without any issues.
Also the client timeout can be set to something like 100 msec.

I see you're calling reboot from the rules.
Is it after such a reboot that the MQTT import is disabled, or is it also crashing for other reasons and thus ending up in some reboot loop?
The reboot-loop is quite fast. It may take less than a minute, so you'll notice the import plugin will stop fetching values quite soon after a reboot.

I can also add some event to signal if things were disabled due to a reboot loop.
That may help testing for issues or at least to fight the symptoms.

agsochi, 2019-03-11:
@TD-er thank you for support.
> Is it after such a reboot that the MQTT import is disabled, or is it also crashing for other reasons and thus ending up in some reboot loop?

Some times modules becomes offline, after several days, and i decide to do reboots.
I will try tune MQTT controller first.

agsochi, 2019-03-12:
Yesterday i turned off reboot in rules, now
‘’’
Uptime:	0 days 13 hours 25 minutes
‘’’
But Mqtt Import plugin was turned off by FW. Why? There is no reboots or disconnects.

TD-er, 2019-03-12:
And the plugin was running since the last reboot?

agsochi, 2019-03-12:
No, i mean that all modules was enabled 13h ago, and no reboots that time, and Mqtt Import plugin was turned off by FW

TD-er, 2019-03-12:
OK, that's strange. Are you sure you saved the settings after enabling it?
The only way it should be disabled is when:
- it is set like that in the settings
- it was in a reboot loop.

Like I said before, when in a reboot loop, the disabled flag is only in memory, not stored.
But as soon as you write something to the settings, it will dump all settings in memory to the flash.
And also you have to make sure to press [submit] when changing things. I know that on some changes it will auto-submit (and thus write) the settings, but not on all. This is not really intuitively consistent, so better save it once too many than not save it at all.

agsochi, 2019-03-12:
I will wait for couple days, let see what happen

TD-er, 2019-03-15:
I think I know what happened at your place.
The boot failed counter is only set to 0 if there has been a successful established network connection.
But in this test, there was also a test for connecting to a MQTT controller (if any MQTT controller is enabled).
This means that if your node was not able to connect to the MQTT controller, it would not decrease this counter.

I will remove this check for MQTT controllers that have to be connected to a broker.

agsochi, 2019-03-16:
Thank you, which build i can test?

TD-er, 2019-03-16:
The one that contains the commits I have not yet made/pushed ;)

TD-er, 2021-08-12:
Is this still an issue?
I think it has been solved a while ago, so I will close it now.
Please let me know if it is still an issue, then I will re-open it.

closed 2021-08-12 by TD-er