letscontrolit/ESPEasy #2390: remove the [TESTING] tag for plugin P079

TungstenE2, 2019-03-12:
Please remove the [TESTING] tag for plugin P079.

https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P079_Wemos_Motorshield.ino

TD-er, 2019-03-12:
Even with this still being an issue?
https://github.com/letscontrolit/ESPEasy/blob/8377429ee54c6fc8fb70ba2ed4699757f84a41b9/src/_P079_Wemos_Motorshield.ino#L12-L15

TungstenE2, 2019-03-12:
I assume thats why I described in the wiki to flash motor_shield.bin first.

see wiki: https://www.letscontrolit.com/wiki/index.php?title=WemosMotorshield

TungstenE2, 2019-03-13:
I also commented the related post:
https://github.com/wemos/Motor_Shield_Firmware/issues/1

TD-er, 2019-03-13:
This wiki link should also be linked in the plugin setup page and then I guess it can be changed to the normal build.

TungstenE2, 2019-03-13:
sure, do you have a link to the plugin setup page? I dont know what you mean by this? I have already linked it on the official plugin list page.

https://www.letscontrolit.com/wiki/index.php/Official_plugin_list

TD-er, 2019-03-13:
I meant here (already working on this issue)
[image]

But as you can see, it is reporting an error as soon as something is added, so that doesn't sound like "stable" to me.
I will have to take a look to see what's wrong here. Maybe it is just because this plugin doesn't have any input/output values and some space in the flash has not been wiped with zeroes.

TungstenE2, 2019-03-13:
no error here
[image]

TungstenE2, 2019-03-13:
I would rename the note to:
Note: Make sure to update the **motor shield** firmware. see wiki

Grovkillen, 2019-03-13:
@TungstenE2 can't you make the plugin capture this error instead of directing it to the wiki?

TungstenE2, 2019-03-13:
no, this beyond my skills. :-(

Grovkillen, 2019-03-13:
Okay, then it's TESTING status still. Sorry but we would need to take care of stuff like that before pushing it to NORMAL.

TungstenE2, 2019-03-13:
the code of the original shield firmware was buggy, and was updaed. Only after the update the I2C scan will find the shield. How to capture this in plugin?

TungstenE2, 2019-03-13:
the testing  for the plugin is fine and working.

TD-er, 2019-03-13:
@Grovkillen The thing is the firmware of the board is quite buggy.
But as I understood it, it is mainly an issue with that firmware, not with the plugin.

Grovkillen, 2019-03-13:
@TD-er aha, then I guess it's only a warning that needs to be added as per @TungstenE2 s suggestion? Move it to NORMAL is fine with me. (Too bad I don't have the hardware...)

TD-er, 2019-03-13:
@TungstenE2 
I will set the note's text to this:
`Make sure to update the Wemos Motorshield firmware, see <a href='https://www.letscontrolit.com/wiki/index.php?title=WemosMotorshield'>wiki</a>`
That's more clear, right?

TungstenE2, 2019-03-13:
fine with me, thx

TungstenE2, 2019-03-15:
updated wiki and plugin list

TungstenE2, 2019-03-15:
updated to build from 15.3.2019, plugin is working with no errors. only small finding: link from note should open in new tab, not the same tab.

TD-er, 2019-03-15:
The text in the note is now just plain text with HTML code.
It should be using a call to some function generating standard HTML wrappers for both the RTD docs and the Wiki.
These wrappers should always open an URL in a new tab.

closed 2019-10-10 by TD-er