letscontrolit/ESPEasy #2394: MH-Z19B stop working when disable ABC - mega-20190305

carck, 2019-03-14:
WD : Uptime 1 ConnectFailures 0 FreeMem 17920 WiFiStatus 3
67685: MHZ19: Unknown response: ff 79 1 0 0 0 0 0 86
67686: MHZ19: Init OK
82708: MHZ19: Sent sensor ABC Disable!
82708: MHZ19: PPM value: 494 Temp/S/U values: 24/0/0.00
82734: Domoticz: Sensortype: 6 idx: 1 values: 494.00;24.00;0.00
91994: WD : Uptime 2 ConnectFailures 0 FreeMem 17888 WiFiStatus 3
121994: WD : Uptime 2 ConnectFailures 0 FreeMem 17856 WiFiStatus 3
142698: MHZ19: Unknown response: ff 79 1 0 0 0 0 0 86
142699: MHZ19: Init OK
151994: WD : Uptime 3 ConnectFailures 0 FreeMem 17792 WiFiStatus 3
157721: MHZ19: Sent sensor ABC Disable!
157721: MHZ19: PPM value: 495 Temp/S/U values: 24/0/0.00
157747: Domoticz: Sensortype: 6 idx: 1 values: 495.00;24.00;0.00
181994: WD : Uptime 3 ConnectFailures 0 FreeMem 17776 WiFiStatus 3
211994: WD : Uptime 4 ConnectFailures 0 FreeMem 17776 WiFiStatus 3
217711: MHZ19: Unknown response: ff ff 79 1 0 0 0 0 0
217712: MHZ19: Init OK
232737: MHZ19: Unknown response: ff 86 1 e9 a0 0 0 a8 ff
232738: MHZ19: Init OK
241994: WD : Uptime 4 ConnectFailures 0 FreeMem 17776 WiFiStatus 3
247772: MHZ19: Sent sensor ABC Disable!
247772: MHZ19: PPM value: 484 Temp/S/U values: 24/0/0.00
247798: Domoticz: Sensortype: 6 idx: 1 values: 484.00;24.00;0.00
271994: WD : Uptime 5 ConnectFailures 0 FreeMem 17760 WiFiStatus 3
301994: WD : Uptime 5 ConnectFailures 0 FreeMem 17712 WiFiStatus 3
307750: MHZ19: Unknown response: ff 79 1 0 0 0 0 0 c3
307751: MHZ19: Init OK
322785: MHZ19: Sent sensor ABC Disable!
322785: MHZ19: PPM value: 480 Temp/S/U values: 24/0/0.00
322810: Domoticz: Sensortype: 6 idx: 1 values: 480.00;24.00;0.00
331997: WD : Uptime 6 ConnectFailures 0 FreeMem 17696 WiFiStatus 3
361994: WD : Uptime 6 ConnectFailures 0 FreeMem 17696 WiFiStatus 3
382763: MHZ19: Unknown response: ff 79 1 0 0 0 0 0 86
382764: MHZ19: Init OK
391994: WD : Uptime 7 ConnectFailures 0 FreeMem 17664 WiFiStatus 3
397786: MHZ19: Sent sensor ABC Disable!
397786: MHZ19: PPM value: 486 Temp/S/U values: 24/0/0.00

TD-er, 2019-03-14:
If you leave the ABC enabled, can you see how often you see a CRC error?

The response is a bit strange, since I would expect:
`ff 01 79 00 00 00 00 00 00 <checksum>`
In your logs it looks like 0x79 and 0x01 are swapped.

According to the [datasheet](https://www.winsen-sensor.com/d/files/infrared-gas-sensor/mh-z19b-co2-ver1_0.pdf) there should be no response.

TD-er, 2019-03-14:
I can reproduce it on my test board too.

TD-er, 2019-03-14:
I could test it on my own board, so it is a quick fix to allow unexpected responses from the sensor.
So no need to reset the sensor when an unexpected reply is given on a command we just sent.

TD-er, 2019-03-14:
OK, not entirely sure what was it that made it appear to work this afternoon.
I've looked into it a bit more and now it is not performing a reset of the sensor immediately when an unknown response is seen.
It will now only reset the sensor when both apply:
- Unknown responses > 10
- time since last reset of sensor was > 3 minutes ago.

netraam3371, 2019-03-15:
According to docs the ABC enable/disable command sends no response. In reality (at least with my units) it does send a response confirming the state change. With the recent change to reset the unit upon invalid responses, this behaviour now resets because the response to ABC setting is unexpected.

It seems like the code reading from serial interface only reads the number of bytes it wants to process, leaving the rest in the buffer. In theory, this could result in readings being one reading interval behind. Not sure if it would be easy to change that (don't know how if serial reads from empty buffer proceed and result 00, would block, or raise errors).
What could help, is flushing the serial input buffer on reception of invalid response, to make sure next reading starts with the next reply's first byte. For example keep reading from input until at least 5 zeros have been read or no response is received/buffer empty.

TD-er, 2019-03-15:
@netraam3371 That may also lead to issues when the read response is actually less than expected.
For example if you're reading 8 bytes, while the message was 7 bytes, then you already have read the first byte of the new message and thus will keep missing data.

If my memory serves me well, the software is reading up-to the next possible start of a message.

closed 2019-03-20 by carck