letscontrolit/ESPEasy #2399: Mega0315 and 311 no data in web-log screen

pwassink, 2019-03-15:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

pwassink, 2019-03-15:
Build:⋄ | 20103  - Mega
-- | --
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20190315
Plugins:⋄ | 46 [Normal]
Build Md5: | 825459376171e63f211c028ecbf1f69
Md5 check: | passed.
Build Time:⋄ | Mar 15 2019 03:15:17
Binary Filename:⋄ | ESP_Easy_mega-20190315_normal_ESP8266_4M.bin

In version mega0311 and mega0315 same build-type binfile  i have no longer data in Tools, Log menu, 
screen appears , all seems fine, but stays in "Fetching log entries..." on both versions.

seeing same behavior on several nodes 

One of them has syslog activated, log data is available inthere, so it is probable something failing in the displaying of that data on the esp-web-console

TD-er, 2019-03-15:
What log settings do you use?
And what browser?
I have been using the web log very intensively last couple of days myself.

pwassink, 2019-03-15:
Debug dev on the syslog, debug more on Web & serial

Browser was firefox on 2 different systems, checking chromium now on third pc which does work .. 

Again cache issue ?

TD-er, 2019-03-15:
Could be.
Please also test using the 'test' build instead of the 'normal' build.
The Normal build has the NO_DEBUG defined, which does silence some logs to keep the memory/flash usage low.

pwassink, 2019-03-15:
Not sure which test version to use, loaded the version ESP_Easy_mega-20190315_test_core_260_sdk3_alpha_ESP8266_4M.bin now on that ESP.

First thing which i found that it does not work with my MHZ19B in this case  Checksum (pass/fail/reset):4/4/1
and no values displayed on that sensor here, after some time still unknown response messages from the mhz19B , Ok this is another issue but might require some investigation too !

Logging on firefox does not work with this version too, in the top opf te page the Logging: also is empty! on another machine with chromium it says Debug More (4) there and works.

What is the proper way to "clear" firefox for this issue ? tried the prefs,privacy,clear data which did not solve it.
cleaning out the cache for a day did not solve it either .. 

Confused now

TD-er, 2019-03-15:
It may be we need to add some extra flags to the served JSON?
Can you see in Firefox in the inspect/console window if there is some error?
For example if Firefox is not allowing to fetch data from a non-HTTPS site?

pwassink, 2019-03-15:
Did some fddling around on that console, i see a 401 error in the networktab debug console Log window 
This is the link it had :
The HTTP 401 Unauthorized client error status response code indicates that the request has not been applied because it lacks valid authentication credentials for the target resource.

This status is sent with a WWW-Authenticate header that contains information on how to authorize correctly.

It also seems the #current_loglevel parameter is not known or at least not displayed at FF here also 

Will do some investigation later, food has higher priority now :-)

TD-er, 2019-03-15:
Hmm this sounds like an issue we had about a week ago with the authentication, but that was on Safari if I'm not mistaken.

See: https://github.com/letscontrolit/ESPEasy/issues/2356#issuecomment-468326003

ShardanX, 2019-03-16:
FYI
Tested it with Opers and firefox today without issues

pwassink, 2019-03-16:
Well,
did some additional testing, narrowed it down to this laptop with firefox, still not sure whats causing it , 
on the other install i've tried with same firefox versions, same os version and evels, same esp it does work.

No firefox cleaning method i am aware of solves this issue on this machine, but it is reasonable to conclude that it is a local problem.

if anyone has a clue please share it with me, but don't spend developing time on this please

closed 2019-03-22 by pwassink