letscontrolit/ESPEasy #2403: user code done ESP32 DEVKITV1

T3rorx, 2019-03-24:
Hi,

Boot error after update / blank with "ESP_Easy_mega-20190315_esp32test_1M8_partition"

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0010,len:4
load:0x3fff0014,len:812
load:0x40078000,len:0
load:0x40078000,len:10164
entry 0x400789f8
user code done

When i use ESPEasy-mega-20181231 my old version same problem now.

Someone have solution ? 

Thanks

T3rorx, 2019-03-24:
With Verbose version

INIT : Booting version:  (ESP32 SDK v3.3-beta1-179-ge931fe9f5-dirty)
48 : INIT : Free RAM:293008
49 : INIT : Cold Boot - Restart Reason: CPU0: Vbat power on reset CPU1: for APP CPU, reseted by PRO CPU
50 : FS   : Mounting...
90 : CRC  : No program memory checksum found. Check output of crc2.py
105 :

TD-er, 2019-03-24:
I have to admit I don't do much testing on the ESP32 myself.
The one running on my ESP32 was built on this date: Feb 25 2019 22:43:33
Maybe you could try a nightly build from about that date, just to check if the current build is somehow corrupt?

T3rorx, 2019-03-24:
i can boot 1/50 times when Press restart button.
 in line 105 i have securitysettings CRC ..OK

TD-er, 2019-03-24:
And what was the 'old' working version?

T3rorx, 2019-03-24:
ESPEasy-mega-20181231

TD-er, 2019-03-24:
Can you test [2019-02-26](https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20190226), just to be sure we don't have to look back for about 4 months of code changes.

T3rorx, 2019-03-25:
With ESP_Easy_mega-20190226_esp32test_1M8_partition.bin Same issue (user code done)

Build With ATOM Work 100%

TD-er, 2019-03-25:
OK, so it seems to be a build issue?
Thanks, that will help a lot actually.

TD-er, 2019-03-25:
Another question, the build you made yourself with Atom, was this in Windows or Linux?

T3rorx, 2019-03-25:
Is in windows.

uzi18, 2019-03-28:
@TD-er @T3rorx have some reports about flashing bootloader and/or boot_app0.bin resolves this issue
did not tested it jet.

```
0x1000 bootloader_dio_40m.bin
0x8000 partitions.bin
0xe000 boot_app0.bin
0x10000 firmware.bin
```

[esp32_test.zip](https://github.com/letscontrolit/ESPEasy/files/3019384/esp32_test.zip)

T3rorx, 2019-03-28:
Tanks a lot that work 100%.

uzi18, 2019-03-28:
so what was exact solution?

T3rorx, 2019-03-28:
the solution is in partitions.bin

i use ESP32 flash tool, just remplace files you send. and flash with  ESP_Easy_mega-20190315_esp32test_1M8_partition.bin

TD-er, 2019-03-28:
And what do I have to change in the build to automate this?

uzi18, 2019-03-28:
we need to prove first it is solution for all of us, and next maybe we need to fill issue on platformio-esp32, not sure if we can use our partition file directly, but maybe something is wrong with conversion from csv file?

TD-er, 2019-03-28:
Right now, I don't really get what's wrong here.
Is it only an issue when running a fresh install on a completely erased flash?

uzi18, 2019-03-28:
user app hang after rebuild and upload from platform.io
when erase flash also hang, did not checked behavior after flashing blank.bin

uzi18, 2019-04-08:
@TD-er @T3rorx #2438 you can find there one commit related, 
it fix for me booting with DOUT as flash communication with 80MHz.

TD-er, 2019-04-08:
Just to be complete, @uzi18 is talking about these two lines:
https://github.com/letscontrolit/ESPEasy/blob/747f11f426eb4df66b9d8fc816db9745531477e9/platformio.ini#L354-L355

closed 2019-10-10 by TD-er