letscontrolit/ESPEasy #2405: C004 ThingSpeak controller

rayellam, 2019-03-25:
C004 controller should have an 8 field buffer so that variables passed to it can ALL be sent to ThingSpeak in ONE go. 

Currently i use 2 x quad variables to hold the data for all 8 ThingSpeak fields with the first quad indexed as 1 and the second quad indexed as 5. When the controller sends the data to ThingSpeak only the first quad's data is sent, occasional the second quad's data is sent but NOT at the same time as the first quad.

Im using release 2019.03.11 with 2.50 core. Built with Ardunio 1.8.9

TD-er, 2019-03-25:
That's currently a problem, since internally ESPeasy can only handle 4 values per plugin.
I think the simplest way to solve this, is to create a separate command for this and then collect the data in rules to be sent all at once.

Maybe this also applies to other controllers?

rayellam, 2019-03-25:
That would certainly be a fix. The problem with ThingSpeak is even if you only write data to 1 field then the other 7 fields are written to with NAN values so when doing post processing of the ThingSpeak data there is a lot of "useless" data points to deal with.

I tried to make a Octal dummy as a separate plugin and this did work although im unsure of any side effects this would have with the core code? Here is a link to what i did.
https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=6545

Regards
Ray

TD-er, 2019-03-25:
One of the things that will probably fail is the way how settings are stored
Actually anywhere in the code where `VARS_PER_TASK` is used, may run into issues.

rayellam, 2019-03-25:
Yes i did find that other "original" quad vars lost there #names. I flashed the ESP with blank bins then reinstalled the build and all "seemed" fine................but my alarm bells were ringing.........lol :-)

TD-er, 2019-03-25:
Maybe we can also "reuse" the IDX value in the plugin and give it special meaning in the ThingSpeak controller.
For example to mark "1st of 4" and "2nd of 4" or something like that.
I have to think about it

rayellam, 2019-03-25:
Yes i get your drift, that would also be a fix, but do bear in mind that it seems some people have a problem understanding the meaning of "IDX" and how this fits with a controller, i noticed several posts a while back when i first started using "controllers" that there was confusion among some users.

rayellam, 2019-03-25:
Or device 1 and device 2?