letscontrolit/ESPEasy #2407: Multiple SHT10 sensors. Could it be a bug ?

Sasch600xt, 2019-03-25:
Here is a link to my message:

[https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6571&p=36071#p36071](url)

TD-er, 2019-03-25:
A lot of plugins are still not programmed with multiple instances of the same plugin in mind.
So they share the same data to keep the status between reads.

It looks like P031_SHT1x is one of them that still has to be adapted to run multiple instances.

Sasch600xt, 2019-03-25:
ok....thank you.

darn :( i use it in a productiv enviroment for controling big gas heaters in a big hall.
i did try with DHT22 but it seems like the industrial power conditions was too "dirty" for them to work properly.

Sascha

Sasch600xt, 2019-04-07:
works lika a charme ! :)

closed 2019-04-07 by TD-er