letscontrolit/ESPEasy #2408: Controller host FQDN in self-built FW

chemmex, 2019-03-26:
I'm trying to predefine most of the default settings to suit my needs in the ESPEasy-Globals.h or Custom.h to make a precomplied custom binary.
However, it seems that Default Controller section in ESPEasy-Globals.h only allows to define DEFAULT_SERVER which is server's IP. Putting FQDN there compiles with empty value. 
I am configuring OpenHAB MQTT which is protocol #5. Tried different builds, all behave the same way.

TD-er, 2019-03-26:
The default server is copied to an IP address parameter:
```c++
#if DEFAULT_CONTROLLER
  MakeControllerSettings(ControllerSettings);
  strcpy_P(ControllerSettings.Subscribe, PSTR(DEFAULT_SUB));
  strcpy_P(ControllerSettings.Publish, PSTR(DEFAULT_PUB));
  strcpy_P(ControllerSettings.MQTTLwtTopic, PSTR(DEFAULT_MQTT_LWT_TOPIC));
  strcpy_P(ControllerSettings.LWTMessageConnect, PSTR(DEFAULT_MQTT_LWT_CONNECT_MESSAGE));
  strcpy_P(ControllerSettings.LWTMessageDisconnect, PSTR(DEFAULT_MQTT_LWT_DISCONNECT_MESSAGE));
  str2ip((char*)DEFAULT_SERVER, ControllerSettings.IP);
  ControllerSettings.HostName[0]=0;
  ControllerSettings.Port = DEFAULT_PORT;
  SaveControllerSettings(0, ControllerSettings);
#endif
```

So right now, it is behaving as designed, but apparently not as desired ;)

I think something like this is needed to fix that:

```c++
// --- Default Controller ------------------------------------------------------------------------------
#define DEFAULT_SERVER_HOSTNAME      "bla.foo.bar"       // Enter your Server host name
#define DEFAULT_CONTROLLER_USEDNS      true
```

Misc.ino Around line 1209:
```c++
  strcpy_P(ControllerSettings.HostName, PSTR(DEFAULT_HOSTNAME));
  ControllerSettings.UseDNS = DEFAULT_CONTROLLER_USEDNS;
```

chemmex, 2019-03-26:
Thanks, seems exactly like the hint I was looking for

closed 2019-11-20 by chemmex