letscontrolit/ESPEasy #2409: ESPEASY is notv work on WeMos D1 NodeMcu Lua V3!

aperoap, 2019-03-26:
I have WeMos D1 NodeMcu Lua V3. 
I tried following ESPEASY MEGA:
- ESP_Easy_mega-20190315_normal_ESP8266_4M.bin
- ESP_Easy_mega-20190311_normal_ESP8266_4M.bin
- ESP_Easy_mega-20190305_normal_ESP8266_4M.bin

i can see "ESP_Easy_0" and conact with this. web AP ist very slowly. I can search my router WIFI. When i choose my wifi and enter my password, starts the countdown but it cant conact with my router and i have to repeat it all again.

with build R120 can i conact und anything works. 

Please help, i have 25 WeMos D1 NodeMcu Lua V3 and i need MQTT-import.

TD-er, 2019-03-26:
have you also tried other core builds of the latest build?
For example core 2.6.0 sdk 2.2.2?

Please also try 20181231 build

HV-NL, 2019-03-27:
My 'Wemos' (i.e. cheap NodeMCU clone) also has no wifi, using any of the current versions
(also tried core 2.4.1, core 2.4.2, core 2.6.0 sdk 2.2.2, core 2.6.0 sdk 3)

But 20181231 does work!

HV-NL, 2019-03-27:
None of the current versions (20190315) does work on my Wemos (cheap nodeMCU clone). I tried them all.

Then I tried 20181231: that one gives quite another output in the debug screen. But also 20181231  does not properly make a connection to my AP.

20181231: Wifi indicates that it is connected to the primary AP, but after a beacon timeout it stops. 

Quick analysis: it can **send**, but does **not receive** (no DHCP address claimed)
> dhcp client start...
> bcn_timout,ap_probe_send_start
> ap_probe_send over, rest wifi status to disassoc
> WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 25 s

Then it tries to connect to the fallback AP, etcetera.
This is what my AP sees:
> attempts to associate
> connected, signal strength -48
> disconnected, received disassoc: no activity (4)
> 
> attempts to associate
> connected, signal strength -48
> disconnected, received disassoc: no activity (4)

So sending does work in 20181231, but no wifi packages are received.

uzi18, 2019-03-27:
In my opinion it is antena issues or faulty module/board.

uzi18, 2019-03-27:
Please post here photo of board

TD-er, 2019-03-27:
Well, it was said to be working with R120, but that one was using a totally different core library and was also do a lot less at startup.

HV-NL, 2019-03-27:
See [/issues/1337/] (april 30 2018) for a picture of this cheap clone.

The antenna should be OK: the device can send, and the message is properly received by the accesspoint.  The device even gets a DHCP address once, so it even receives properly, but finally a 'beacon' is not received, and the connection is broken down. After that no new connections are possible any more.

> connected with LinksysAtHome, channel 1
> dhcp client start...
> ip:192.168.11.71,mask:255.255.255.0,gw:192.168.11.1
> 4567 : WIFI : Connected! AP: LinksysAtHome (B8:69:F4:1C:56:90) Ch: 1 Duration: 3843 ms
> 4568 : WIFI : DHCP IP: 192.168.11.71 (ESP-Easy-20) GW: 192.168.11.1 SN: 255.255.255.0   duration: 41 ms
> 4574 : Webserver: start
> bcn_timout,ap_probe_send_start
> ap_probe_send over, rest wifi status to disassoc
> state: 5 -> 0 (1)
> rm 0
> 13105 : WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 8536 ms

An older version of ESPEasy does work, so the device is probably not completely faulty.

TD-er, 2019-03-27:
See also the discussion here: https://github.com/esp8266/Arduino/issues/5908#issuecomment-475867223
Maybe you can also try this?

Maybe it is also possible to just wipe the flash and try core 2.4.1 or 2.4.2.
Wiping the flash should also clear the RF calibration values.

HV-NL, 2019-03-28:
Wiping the RF calibration data did not help. 
Wiping the flash + reflash 20190315 core 2.4.2 also did not give me a working wifi.

Just flashed v2.0.0-dev13 again: wifi is working again.

So I tried a different approach: I created a very small program, to test this device stand-alone, using the current Arduino IDE. I used the library 'wifiManager' for this test.

> #include <ESP8266WiFi.h>
> #include <DNSServer.h>
> #include <ESP8266WebServer.h>
> #include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
> 
> void configModeCallback (WiFiManager *myWiFiManager) {
>   Serial.println("Entered config mode");
>   Serial.println(WiFi.softAPIP());
>   Serial.println(myWiFiManager->getConfigPortalSSID());
> } //configModeCallback
> 
> void setup() {
>   Serial.begin(115200);
>   WiFiManager wifiManager;
>   wifiManager.setAPCallback(configModeCallback);
>   if(!wifiManager.autoConnect()) {
>     Serial.println("failed to connect and hit timeout");
>     //reset and try again, or maybe put it to deep sleep
>     ESP.reset();
>     delay(1000);
>   } 
>   //if you get here you have connected to the WiFi
>   Serial.println("connected...yeey :)");
> } // setup
> 
> void loop() {
> } // loop

This also does not work on this cheap 4M Wemos clone (I wiped flash before upload): no wifi.
I also uploaded this to a 'normal' 4M nodeMCU clone (with a blank aluminium cap over the wifi part): works OK.

So my impression is that this cheap Wemos clone is currently incompatible with the current IDE and/or ESP8266 library, and this is not a dedicated ESPEasy configuration related item.

TD-er, 2019-03-28:
Thanks for this test with a simple setup using WiFimanager
That's giving me some confidence it is indeed not an ESPeasy issue.

One thing you could try to do is to call lots of delays at boot until connected

aperoap, 2019-03-28:
> have you also tried other core builds of the latest build?
> For example core 2.6.0 sdk 2.2.2?
> 
> Please also try 20181231 build

i have 20181231 build tried. it works very good.
[image]

a pic of my ESP from Wemos

uzi18, 2019-03-29:
interesting it is not covered by metallic cup

TD-er, 2019-03-29:
@uzi18 The Wemos D1 mini pro (16M) is also not covered with a metal shield.
One of the things changed since 2018/12/31 is that it now tries to lessen the load during startup.

jcpy70, 2019-03-29:
Hello everyone,
For more than 6 months, I use EspEasy with happiness, congratulations to all.
I use either classic NodeMCU Lua V3 (with metal shield) or Wemos D1 mini (also with a metal shield) or Wemos D1 mini Pro with ceramic antenna and no metal shield...
It's the D1 mini Pro which poses problem starting from the version ESP_Easy_mega-20190305_normal_ESP8266_4M (idem for test and dev).
Wifi is working but the browser (Firefox or IE) is waiting for a response...
A port scan indicates that ports 25,80,110,119,143,465,563,587,993,995 are open.

jcpy70, 2019-03-31:
I tested several versions with the Wemos D1 Mini Pro (ceramic antenna and no WiFi shielding):
The latest stable release is ESPEasy_mega-20190216.
If it can be useful.

khambrecht, 2019-04-05:
I also have one of these Wemos devices. Mine looks exactly as the one @aperoap has posted. Old firmware R120 work fine. I then stepped through more recent firmware builds, beginning with 20181231 (but skipped some in between). Here are my results:

- 20181231_normal_ESP8266_4096.bin -> OK, stable
- 20190106_normal_ESP8266_4096.bin -> not OK, connect to initial AP mode fails
- 20190212_normal_ESP8266_4096.bin -> not OK, connect to initial AP mode fails
- 20190215_normal_ESP8266_4096.bin -> not OK, connect to initial AP mode fails
- 20190216_normal_ESP8266_4096.bin -> not OK, initial AP mode worked, later WiFi client slow and unstable.
- 20190225_normal_ESP8266_4M.bin -> OK, stable (*)
- 20190226_normal_ESP8266_4M.bin -> OK, stable (*)
- 20190227_normal_ESP8266_4M.bin -> not OK, initial AP mode worked, later WiFi client slow and unstable.
- 20190305_normal_ESP8266_4M.bin -> not OK, connect to initial AP mode fails
- 20190311_normal_ESP8266_4M.bin -> not OK, connect to initial AP mode fails
- 20190315_normal_ESP8266_4M.bin -> not OK, connect to initial AP mode fails
- 20190404_normal_ESP8266_4M.bin -> not OK, connect to initial AP mode fails

(*) these two firmware releases sometimes require several attempts to proceed with the initial AP mode

jcpy70, 2019-04-08:
Good new !
The latest version ESP_Easy_mega-20190406_xxx is well supported by the Wemos D1 Mini Pro card (with ceramic or external antenna).
Thanks to the whole team !

TD-er, 2019-04-08:
@jcpy70 Good to hear.
Not to spoil it, but I want to make sure it isn't a build issue, so let's try again for the next build.

jcpy70, 2019-04-09:
@TD-er :
Tested today... **Ok** with ESP_Easy_mega-20190409_xxx
I use these cards, the soldering for the external antenna is not easy, the 0 ohm resistor  (R9) is small ...
[image]

TD-er, 2019-04-09:
Great, good to hear it isn't just a single build that was working by accident :)
I use these boards too. They are great and I'm also working on getting to use the whole 16M of the flash.
About the 0-Ohm resistors. I know what you mean. I have a strip of them in case I loose one or destroy in the process.

zerog2k, 2019-04-10:
FWIW:
Also having this strange issue. This is a new clone of Wemos D1 R2. But what I notice from the picture above, is that my mcu is also strange: it has "ECO Plugs" label on mcu instead of familiar ESP8266... labeling. I'm suspecting these are some sort of rebadge/rebrand (and maybe rejects/extras/grey-runs :( ).
Anyhow, 
ESP_Easy_mega-20190409_normal_ESP8266_4M.bin also does not work for me: it never picks up an ip address, whereas ESP_Easy_mega-20190226_normal_ESP8266_4M.bin works.

arnemauer, 2019-04-13:
> Good new !
> The latest version ESP_Easy_mega-20190406_xxx is well supported by the Wemos D1 Mini Pro card (with ceramic or external antenna).
> Thanks to the whole team !

I am using also the Wemos D1 mini pro. It works with 20190226. It doesn't work with 20190227 and later.  The card i have is a v1.0.0. Which versions is yours?

sebastianheyn, 2019-04-15:
IMHO, its power supply issues. I've had that with other boards from china before. Its a capacitor/regulator misalignment. try adding some caps to the 5V an 3,3v and you will see the problems shift or disappear. 
For example, for me the at firmware always worked, since the consumption is lower somehow. but then after changing to easyesp, sometimes they would work, sometimes they dont. 
I add a 470µF to the 5V and 100µF to the 3,3V, and 2 or 3 0,1µF to the 3,3V. Helps most of the times. 
Problem is they use cheap regulators, that cannot handle power bursts.

TD-er, 2019-04-15:
I am running now on a few of these Wemos D1 mini pro's here on my desk and they all seem to work.
I am using good (and short) USB cables and all is powered from a 7+3 port USB3 hub from Anker, with a 60 Watt power supply.
Also on the boards I make, I have added a few capacitors and these boards also run stable.

The boards I make have a 470uF capacitor near the ESP node (and the sensors that need 5V) and have thick copper traces for the traces that run power lines.

jcpy70, 2019-04-16:
**@TD-er** & **sebastianheyn**
Yes, of course, these are basic engineering rules ... But how do you explain that the same Wemos on the same PCB (this is my case for  my personal-made PCBs) works correctly with some versions of ESPEasy and not others? The presence of capacitors does not explain everything.

@**arnemauer** : My Wemos are also 1.0.0 versions.
[image]

arnemauer, 2019-04-16:
> **@TD-er** & **sebastianheyn**
> Yes, of course, these are basic engineering rules ... But how do you explain that the same Wemos on the same PCB (this is my case for my personal-made PCBs) works [...]

I found this tread on reddit about a bad voltage regulator on de wemos clone boards: https://www.reddit.com/r/esp8266/comments/9iizx4/warning_clone_wemos_d1_minis_with_only_150ma_33v/. And... yes mine came with a 150ma voltage regulator :(

I removed the regulator and replaced it with an ams1117-3.3v and couple of capacitors but the esp keeps crashing :(

Domosapiens, 2019-04-16:
> I removed the regulator and replaced it with an ams1117-3.3v and couple of capacitors but the esp keeps crashing :(

I confirm that

arnemauer, 2019-04-17:
Today i tried 20190416 on a wemos d1 pro with ams1117 + capacitors and on a nodemcu amica (also ams1117). I used three different cables (logitech 100cm, anker 30cm,  ikea 150cm) and three power supplies (laptop, anker 24w, samsung 2A). I flashed the chips with the blank.bin before flashing the new release. 8/10 times the esp crashes in the initial setup while scanning or after entering de wifi password.

I posted the exceptions and traces on pastebin:
Wemos D1 + AMS1117 + Anker: https://pastebin.com/w56Lf7Pn
Wemos D1 + AMS1117 + Samsung: https://pastebin.com/R6CEv3PR
NodeMCU amica + Anker: https://pastebin.com/t2Lz9gaL

@TD-er Where do you get your wemos D1 pro's? Lolin doesn't make them anymore and alle versions on aliexpress have the 150mA voltage regulator.

TD-er, 2019-04-17:
I recently ordered a few and will look at the regulators they use.

will let you know asap

Domosapiens, 2019-04-17:
I recently bought:
[https://nl.aliexpress.com/item/D1-mini-Mini-NodeMcu-4M-bytes-Lua-WIFI-Internet-of-Things-development-board-based-ESP8266/32529101036.html?spm=a2g0z.10010108.1000016.1.6c3c7962EiBeib&isOrigTitle=true](url)

Has LDO S2RC (or S2RE ... old eyes). It has the Mircone swoosh logo:
[https://www.reddit.com/r/esp8266/comments/9itox8/survey_did_you_get_a_150ma_regulator_on_your/](url)

Under test now

TD-er, 2019-04-18:
OK, I opened the last mini pro's I ordered and indeed they all have some 4Axx marking on them.
So that will become an issue later on.
I also had a Wemos D1 mini (non pro) which I had to toss away last week as not even able to flash it.
Took that one out of the bin and that one even has a B2X marking, which I cannot even find anything about.

I have on my boards always a 100 nF cap over the 3V3 line, near any sensor and also near the ESP itself.
Maybe I should increase the one near the ESP just to be sure, although I cannot predict whether it will boot properly if the cap is too big.
And I will add a separate 3V3 regulator for the external components.

sebastianheyn, 2019-04-18:
I have ordered a Wemos D1 mini for another project, and I can tell you, my nodemcu works fine, but the same firmware doesnt work on wemos d1 mini. The original Wemos D1 mini suggests to use a ME6211 regulator which is able to handle the power surge without capacitor. if you use a knock-off part with the same pinout, the wemos d1 would severely suffer from not having big capacitors, in comparison to the nodemcu (lolin) board

TD-er, 2019-04-18:
What I can do, is to add some delay before initializing the wifi. And then make this delay an option to disable (or even make it dynamic if it crashes)
This then allows to charge some capacitors before requiring the high current peak.

khambrecht, 2019-04-18:
I don't think we should focus that much on power/voltage issue. My device, which is exactly the one that @aperoap has posted a picture of, has indeed the recommended AMS1117 voltage regulator. And this device works perfectly with some of the firmware releases, but does not work with others. See my posting above for working and non-working releases. So, IMHO power/voltage is just a secondary issue.

TD-er, 2019-04-18:
I agree we must not focus on it as if that's the only issue, but I also want to make sure we're not polluting the reported issues with possible power issues.
And I am not entirely sure those not working builds are build issue and not power related.

khambrecht, 2019-04-25:
Just for the records. I tested the latest releases on my (somewhat unknown) Wemos device. 20190413 and 20190416 do not work at all (no initial AP mode connect), while 20190419 and 20190425 will get the initial AP mode (after several attempts), but WiFi is very unstable and slow, so not really usable. Going back to 20190226 the device works fine.

zerog2k, 2019-04-25:
question for those having this issue:
does your mcu have normal "esp8266ex" or somesuch marking on it, or does it have the "eco plugs" marking?

HV-NL, 2019-04-28:
> question for those having this issue:
> does your mcu have normal "esp8266ex" or somesuch marking on it, or does it have the "eco plugs" marking?

@zerog2k: my cheap Wemo clone has the marking 'ESP8266EX', and even so has big problems:

[image]

(and an on-board AMS1117 power regulator)

arnemauer, 2019-04-28:
[image]
[image]

I have a board with "ESP8266EX" and a board with "ECO Plugs B REV 1.1". Both with a AMS1117.

zerog2k, 2019-04-28:
Ok, at least for me, I doubt the issue is power related. 
- I put scope on 3v3 rail near mcu, with 3.2v trigger - never sagged
- did not see any evidence of reset on serial line when starting. (Would expect power issues to manifest as resets?)
- from serial line (w/ `debug 4`), seems like it claims it's connecting to wifi ap, but then never seems to get an ip.

TD-er, 2019-04-29:
So, it looks like the events for WiFi activity is not processed?
The strange thing is, is that it seems to be working in some builds and not in others.
So it seems to be related to the randomness of builds.
Sounds like something not being initialized.

khambrecht, 2019-08-06:
After a long time, I give it a try on my Wemos D1 NodeMcu device again. Release 20190805 (firmware file ESP_Easy_mega-20190805_normal_ESP8266_4M). This one works somewhat. Initial AP configuration works. However I see lots of DHCP request from the device after connecting to regular WiFi network (e.g. every 10-15 sec), making connections very slow due to lots of lost packages. I do not see this DHCP issue with Release 20190226.

TD-er, 2019-08-06:
I don't think the DHCP requests are making it slow, but the fact that it actually thinks it needs to do those requests.
It is an indication the node does reset the network connection a lot.
Can you set it to force B/G mode only?
Also make sure no metal is near the WiFi antenna and try to position the node to get maximum RSSI from the AP. (or rotate the AP if that's easier for the test)

uzi18, 2019-08-06:
@khambrecht what is your rssi in module?

khambrecht, 2019-08-06:
Well, yes. Maybe the connection itself is not slow. However, each time the device sends a DHCP request, existing client connections (e.g. Browser) get dropped and require reconnect. This makes it at least behave slowly and you often have to reload/refresh browser page.

As already mentioned, the device works fine with release 20190226, using same power supply, same location. I do not see these frequent DHCP requests and the connection is reliable.

RSSI is around -58dB, with both, release 20190805 and 20190226. Force B/G does not make a difference.

uzi18, 2019-08-06:
debug log should give us some hints, witch core use 20190226 build?

khambrecht, 2019-08-06:
set serial to debug-dev. attached serial output from Rel. 20190226 (working) and Rel. 20190805 (not working).
[20190226-serial-debug.log](https://github.com/letscontrolit/ESPEasy/files/3473176/20190226-serial-debug.log)
[20190805-serial-debug.log](https://github.com/letscontrolit/ESPEasy/files/3473177/20190805-serial-debug.log)

uzi18, 2019-08-06:
lots of :
`WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 8652 ms`

uzi18, 2019-08-06:
`WIFI : Disconnected! Reason: '(200) Beacon timeout' Connected for 8652 ms`

uzi18, 2019-08-06:
@TD-er  isn't a problem reported by @clumsy-stefan ?

TD-er, 2019-08-06:
Well, Beacon timeout can be caused by a lot of things, but if it happens quite often then I agree you may be on to something.
Let's summon @clumsy-stefan on this one :)

khambrecht, 2019-08-12:
Just to let you know, same issue with release 20190809. DHCP requests exactly every 10 sec.

mrdc, 2019-09-15:
I can confirm a bug also on build `mega-20190903` running on ESP-01S:
> disconnected, received disassoc: no activity (4)
> DHCP requests exactly every 10 sec.

Some investigation shows the following info concerning DHCP:
build `mega-20190903` on NodeMcu v2 works fine and I see in DHCP Server on Mikrotik that device has a client ID and active host name, but for ESP-01S with the same build `mega-20190903` these fields are empty.
Maybe it somehow related.

TD-er, 2019-09-15:
Are you sure the settings are the same on both?

mrdc, 2019-09-16:
Well, they have different sensors attached: CO2+BME280@NodeMcu and DS18B20@ESP-01S.
As I remember, on ESP-01S I have "Periodical send Gratuitous ARP" enabled. 
In the morning I've switched it off - ESP-01S started to work. Will see in the evening how it performed with "Periodical send Gratuitous ARP" set to off.

> And has the ESP-01S a PUYA chip for flash?
Yes, it's PUYA.

TD-er, 2019-09-16:
The last build, using core 2.6.0 should also be able to work (and continue to work) without Gratuitous ARP checked.

Can you also compare some timings on the timingstats page?
For example the `LoadTaskSettings()` line.
That one is called quite often, so it may be a simple indication for the flash speed.
Please also compare it with another node (not using PUYA chips) which is working fine.

As a comparison these timings on 2 different nodes I have here. Both NodeMCU, but the slower one is on my breadboard test node.
So that one probably has a very fragmented file system. 

N.B. The breadboard one is already replaced once, since I had worn out the flash of the first one after a few 1000x flashing a new build on it. So that one is being abused quite a lot :)

Description | Function | #calls | call/sec | min (ms) | Avg (ms) | max (ms)
-- | -- | -- | -- | -- | -- | --
LoadTaskSettings() (clean filesystem) |   | 17402 | 1.59 | 3.296 | 3.755 | 7.227
LoadTaskSettings() (fragmented filesystem)|   | 11872 | 1.05 | 3.572 | 17.845 | 39.134

mrdc, 2019-09-16:
> Can you also compare some timings on the timingstats page?

Here you go:
ESP-01S (PUYA flash)
[image]

NodeMcu v2
[image]

[image]

I've set `Force WiFi No Sleep: true` for tests.

TD-er, 2019-09-16:
So the reading from flash of the Puya ones does seem to be faster, or it is always trying to read the same plugin data (only 1 plugin active or sending to a controller?)

With a RSSI of worse than -75, you may also want to try to force B/G only.
500+ reconnects is a lot.
Can you backup the settings, perform a full clean flash (writing the blank image) and flash it again?
I'm beginning to suspect the RF calibration data on the node could be wrong here.

mrdc, 2019-09-18:
> or it is always trying to read the same plugin data (only 1 plugin active or sending to a controller?)

I have only one sensor attahced to ESP-01S.

> With a RSSI of worse than -75, you may also want to try to force B/G only.

I've tried it - the same behaviour.

> Can you backup the settings, perform a full clean flash (writing the blank image) and flash it again?

Yes, no problem. BTW I've already had flashed black image before flashing `mega-20190903`. Should I do it one more time?

TD-er, 2019-09-18:
> Yes, no problem. BTW I've already had flashed black image before flashing `mega-20190903`. Should I do it one more time?

If it is related to bad RF calibration (hard to say right now), then it may make a difference.
Just make sure you are running it with a proper power supply and maybe even more important a good USB cable.
Maybe also add some capacitors over the 3v3 line and GND to help stabilize the power during high current consumption.

[49 more replies]

closed 2023-09-02 by TD-er