letscontrolit/ESPEasy #2410: libs incompatibility with LDF

uzi18, 2019-03-27:
some libs are not compatible with LDF, they are not used when build firmware

```
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Ignored library /home/users/linaja/ESPEasy-mega_test/lib/AS_BH1750
Ignored library /home/users/linaja/ESPEasy-mega_test/lib/ESP32_ping
Ignored library /home/users/linaja/ESPEasy-mega_test/lib/ESPEasy_ESP8266Ping
Ignored library /home/users/linaja/ESPEasy-mega_test/lib/IRremoteESP8266
Collected 51 compatible libraries
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode

```

used env esp32test

uzi18, 2019-03-27:
@TD-er any ideas why they are incompatible?

TD-er, 2019-03-27:
Those are listed in the lib_ignore list in platformio.ini, right?
Those in that ignore list are the ones I put there to just make it compile ;)

Or am I missing something in your question?

uzi18, 2019-03-28:
ok, right, I had some problems with build process of esp32test and found such information.

closed 2019-03-28 by uzi18