letscontrolit/ESPEasy #2411: P028 (BME280) vs obsolete P030

uzi18, 2019-03-29:
Plugin P028 in compatibility mode with sensor BMP280 connected, 
does not emulate P030 behavior.

is definied as: 
```
VType = SENSOR_TYPE_TEMP_HUM_BARO;
ValueCount = 3;

```
but should be:
```
VType = SENSOR_TYPE_TEMP_BARO;
ValueCount = 2;

```

related to commit:
https://github.com/letscontrolit/ESPEasy/commit/6a955efb8a667d1b9f1ccf55c91a289c3715bb62

@TD-er is it a way to access for current Device structure inside PluginCall ?

TD-er, 2019-03-29:
I have to look into it to see how it can be made more dynamic.
There are several ways to do it and their main difference is the amount of code change is needed.

uzi18, 2019-03-29:
What are your ideas for this?
global int with current device number should be enough but it is not clean solution.
Maybe we can extend event with device number field as int8_t

TD-er, 2019-03-29:
The most clean solution would be to no longer have these values set in the PLUGIN_ADD function, but query them just like we als do for the device names.
It also does not make sense to have them in memory.
But it does probably involve all plugin codes.

It would however also fix the issues with the plugins that allow multiple outputs like the sysinfo plugin.
And also allows some flexibility for the Dallas plugin.

On the other hand the Dummy plugin also allows changing number of variables, so we may also need to look at how it's done there.

tonhuisman, 2022-04-18:
This seems to have been fixed a long time, so the issue can be closed.

closed 2022-04-18 by TD-er