letscontrolit/ESPEasy #2433: Variable %systm_hm% has the wrong format (no leading zero)

klalei, 2019-04-05:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller
- [x] had a look into source of TimeESPeasy.ino

/ returns the current Time separated by the given delimiter
// time format example with ':' delimiter: 23:59:59 (HH:MM:SS)
String getTimeString(const struct tm& ts, char delimiter, bool am_pm, bool show_seconds)
{
...
if (show_seconds) {
sprintf_P(TimeString, PSTR("%02d%c%02d%c%02d"),
ts.tm_hour, delimiter, ts.tm_min, delimiter, ts.tm_sec);
} else {
sprintf_P(TimeString, PSTR("%d%c%02d"),     <<<<------ here is the 02 missing
ts.tm_hour, delimiter, ts.tm_min);
}
}
return TimeString;
}

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

Variable %systm_hm% has no leading zero

### Expected behavior

should show leading zero as described in the docu

### Actual behavior

shows no leading zero.

### Steps to reproduce

1.  look to Tools / system variables before 10 o'clock
2. with lcd connected edit the device and enter %systm_hm% in one of the lines and submit
3. have a look at TimeESPeasy.ino at getTimeString

 YES

### System configuration

Hardware: Wemos D1 with I2C LCD

ESP Easy version: ESP_Easy_mega-20190404_normal_ESP8266_4M.bin

ESP Easy settings/screenshots: 

### Rules or log data

```

```

tonhuisman, 2022-05-06:
Why would a presentation value have a leading 0? If zeroes are required, better use the `%syshour_0%` `%sysmin_0%` and `%syssec_0%` variables that have been available for quite some time (though maybe not when the issue was created)

uzi18, 2022-05-06:
Because it moves time on lcd by one character.

tonhuisman, 2022-05-07:
I suspect that a leading space is being trimmed off somewhere unintended. I'll look into that.

TD-er, 2022-05-07:
Maybe better to add a variable which clearly states it should have leading zeroes.
I am a bit reluctant on changing it as suddenly adding a leading zero may cause issues on working rules where a leading zero may cause the parsing to be seen as octal format.

tonhuisman, 2022-05-07:
Looking at the code, there is no leading space when the hour is < 10 (using a `sprintf_P` format of `"%d%c%02d"`, so it only uses the required space, no prefixes).
I can add `%systm_hm_0%` and `%systm_hm_sp%`, providing a zero and space-prefixed hour/minutes time.
But possibly this should then also be made available for the AM/PM variant? Maybe @thomastech can add an opinion for the part of the world that uses AM/PM time format?

TD-er, 2022-05-07:
> [...] AM/PM time format?

We already have the `%systm_hm_am%`
Not sure if I can recall anybody compaining about that one.

tonhuisman, 2022-05-07:
Those variables would result in something like this:

[image]

NB: `%systime%` is always 0-prefixed, so I didn't add anything to remove the 0 or space-prefix it.

TD-er, 2022-05-07:
Looks like you already have them, so just PR it ;)

PR #4046 (2022-05-31): [Sysvars] Add %systm_hm_0%, %systm_hm_sp%, %systm_hm_am_0%, %systm_hm_am_sp%, %systime_am_0% and %systime_am_sp%
Resolves #2433 

Adds:
`%systm_hm_0%` - Prefix hour <10 with 0.
`%systm_hm_sp%` - Prefix hour <10 with space.
`%systm_hm_am_0%` - Prefix AM/PM hour <10 with 0.
`%systm_hm_am_sp%` - Prefix AM/PM hour <10 with space.
`%systime_am_0%` - Prefix AM/PM hour <10 with 0 (includes seconds).
`%systime_am_sp%` - Prefix AM/PM hour <10 with space (includes seconds).

This offers the time with a constant width/length, for use on displays. On the System variables page the space-prefixes will be ab [... cut]
closed 2022-05-31 by TD-er