letscontrolit/ESPEasy #2436: SONOFF_POW 1M binary in nightly build

Wiki591, 2019-04-06:
I have seen, that the Blitzwolf SHP hardware meanwhile is included as [Testing] in the 4M release for SONOFF POW. 
Due to the fact that the Blitzwolf SHP devices are equipped only with 1M memory it could make sense to include a 1M build in the nightly relases to have the possibility to test it.

TD-er, 2019-04-07:
Ah OK, did not know that.
Do you have other info for that node, so I can also add a predefined config for that one including the factory default setup of the button and relais?
- GPIO for button
- GPIO for relay
- GPIO for LEDs

Wiki591, 2019-04-07:
Sure, no problem, here we go:

Blitzwolf SHP6&SHP2:
- GPIO15 = Relay
- GPIO13 = Button
- GPIO2 = blue LED

Blitzwolf SHP2 only:
- GPIO0 = red LED

Is there any chance to limit the sketch size for 2-step-OTA?

TD-er, 2019-04-07:
I have to check how big it will be.
It is a bit more than the "minimal" build, and that one doesn't have much room left.

Wiki591, 2019-04-14:
Any progress?

TD-er, 2019-04-15:
Yes and no.
No, I have not yet started to make a POW 1M build (should not name it POW, but something with "energy" or "SHPx")  but I have been working in reducing the binary size.

Wiki591, 2019-04-15:
Sounds good, Thanks for the information and sorry for my impatience.

Wiki591, 2019-05-22:
Still no further progress?

TD-er, 2019-05-22:
Nope, but I will make a configuration for it now.

TD-er, 2019-05-22:
Just to make sure, are those using ESP8266 or ESP8285?

TD-er, 2019-05-22:
A quick search suggests they have ESP8285 ones on board.
So I just made a commit + merge for it.
They should be included in tomorrow's build.

Wiki591, 2019-05-23:
Is it included in the current nightly build? I Didn't find it?
Following [this sheet](https://github.com/arendst/Sonoff-Tasmota/files/3038647/sch_BW-SHP2_.SP1-C_V2.4.pdf) - last found hardware release of tem - its a ESP8285. 
I would like to try flashing these devices following [this guide](https://www.heise.de/ct/artikel/Tuya-Convert-Escaping-the-IoT-Cloud-no-solder-need-4284830.html) without the need to solder, thats why I asked for an OTA version.

TD-er, 2019-05-23:
Oh, in the OTA build. I guess it is not suitable for OTA build and I am also not entirely sure I put it in the build script also, now you mention it.
I will have a look at it later this evening.

Wiki591, 2019-05-23:
The OTA would be imho an essential function. Otherwise the effort for opening the devices, gtting it apart without damages, soldering cables, fiddling around with pulling gpio0 to ground, flash, desolder and putting it together again without damages is a pain in the a*s. Flashing them ota using the tuya hack with tasmota works charming good, but I think most of the ESPEasy community would like to stay on one sole system. At least so do I and there have been already similar requests in the forum.

TD-er, 2019-05-24:
I understand.
It looks like a bit more than just a "OK, will do it right now" change, so I will look into it, when I got back from Sweden.

Wiki591, 2019-05-25:
Jo, having a look at the current image sizes I can gues that its no drive-by compilation, but these devices are worth it. If you like, I could provide you with one of these devices (but already flashed with tasmota using the tuya hack). More are on the road. I would like to assist but my coding skills are far away from the point doing this, even I can't get a running coding environment.

uzi18, 2019-05-25:
@Wiki591what it tuya hack?

Wiki591, 2019-05-25:
Just Have a look at the in my post two days ago.

TD-er, 2019-10-10:
Not entirely sure right now, but is this issue still valid?

closed 2019-10-11 by Grovkillen