letscontrolit/ESPEasy #2437: EVENT not generated by PLUGIN_READ if only one response to PluginCall() is true

BNNorman, 2019-04-08:
### Checklist
- [ x] The title of this issue is "to the point" and descriptive.
- [x ] This issue describes what is happening.
- [x ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x ] This issue describes how to reproduce it.
- [ x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I am self building using Mega_20190315 sources.

I am redeveloping the _P053_PMSx003.ino plugin to provide settings which control how long the PMS sensor is on and off in the same way that the SDS dust sensor works in order to extend its (fan) life without using timers. Also to use passive mode to reduce the number of bad checksums I see when it is free wheeling in active mode using the existing sensor plugin. Those parts are working but at the point where I send the sensor to sleep I only want to generate a single EVENT so that a rule to mqtt publish the data fires just once at the end of the warmup period. I am using a JSON payload which includes a timestamp and all the sensor readings (plus a BME280) hence the rule approach.

To limit to a single event call I use a flag. P053_readNotify which is set false after the code returns true. If I don't set that flag to false then I get multiple events generated as expected.

This is the section of code I have.
```
  case PLUGIN_READ: {
        // copy the data to where ESP Easy wants it
        // values could be zero, after a restart, till a valid checksum is seen
        UserVar[event->BaseVarIndex]     = P053_PM1_0;
        UserVar[event->BaseVarIndex + 1] = P053_PM2_5;
        UserVar[event->BaseVarIndex + 2] = P053_PM10_0;

        // if the sensor is permanently on notify immediately
        // the user is responsible for the interval
        if (P053_sleepTime==0) return true;

        // P053_sensorSleep() sets P053_readyNotify when it goes to sleep
        // so create EVENT just once at end of awake period
        if (P053_readNotify){
          P053_readNotify=false;
          if (loglevelActiveFor(LOG_LEVEL_DEBUG)) addLog(LOG_LEVEL_DEBUG,F("PLUGIN_READ returns true"));
          return true;
        }
          if (loglevelActiveFor(LOG_LEVEL_DEBUG)) addLog(LOG_LEVEL_DEBUG,F("PLUGIN_READ returns false"));
        // no EVENT required.
        return false;
    }
```

### Expected behavior

A single EVENT should be generated for the sensor values e.g. PMSx003#pm10

### Actual behavior

No event is generated unless I comment out the line P053_readNotify=false;

### Steps to reproduce

1. try the technique I am using - I could post the whole (new) plugin code if that helps
2. 
3. 

Yes it does persist ofter a power cycle

### System configuration

Hardware:
nodeMCU V1.0

Unit Number: | 3
-- | --
Local Time: | 2019-04-08 16:26:58
Uptime: | 0 days 0 hours 6 minutes
Load: | 9.90% (LC=5533)
CPU Eco Mode: | false
Free Mem: | 16504 (15096 - sendContentBlocking)
Free Stack: | 3600 (2176 - sendContentBlocking)
Boot: | Manual reboot (1)
Reset Reason: | External System
Network ❔
Wifi: | 802.11N (RSSI -63 dB)
IP Config: | Static
IP / Subnet: | 192.168.1.90 / 255.255.255.0
Gateway: | 192.168.1.1
Client IP: | 192.168.1.10
DNS: | 192.168.1.1 / 0.0.0.0
Allowed IP Range: | 192.168.1.0 - 192.168.1.255
STA MAC: | DC:4F:22:18:03:25
AP MAC: | DE:4F:22:18:03:25
SSID: | KC Lightstream 10 (58:8B:F3:3E:CB:3B)
Channel: | 10
Connected: | 5 m 45 s
Last Disconnect Reason: | (1) Unspecified
Number Reconnects: | 0
WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | true
Connection Failure Threshold: | 0
Firmware
Build:⋄ | 20103  - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ |  
Plugins:⋄ | 47 [Normal]
Build Time:⋄ | Apr  8 2019 15:50:16
Binary Filename:⋄ | Self built!
System Status
Syslog Log Level: | None
Serial Log Level: | Debug
Web Log Level: | None
ESP board
ESP Chip ID: | 1573669 (0x180325)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 795 kB (2276 kB free)
SPIFFS Size: | 934 kB (860 kB free)

ESP Easy version: Mega_20190315

ESP Easy settings/screenshots: 

### Rules or log data

```
on PMS7003#pm10 do
logEntry,"Rule PMS7003:#pm10; pm10=[PMS7003#pm10], pm2.5=[PMS7003#pm2.5], pm1.0=[PMS7003#pm1.0]"
//Publish airquality/data,{"dev":"CHASN-180325-1","timestamp":"%sysyear_0%-%sysmonth_0%-%sysday%T%systime%+01:00","PM25":[PMS7003#pm2.5],"PM10":[PMS7003#pm10],"lon":XXXXXX,"lat":YYYYYY,"alt":1.5}
endon
```

TD-er, 2019-04-08:
This is not how it should be done.
The plugin itself knows when it is done with the measurement, so the plugin should put the sensor to sleep again.

Take a look at how the plugin may re-schedule itself by looking at BME280.ino.
The way most in line with how ESPeasy is working internally is this:

- State sensor = off, state = idle
- Call `PLUGIN_READ`
- Plugin will enable sensor, initialize a reading and set its state to "read initiated"
- `FUNCTION_TEN_PER_SECOND` should check if valid value is read and when finished, set state to "value ready" and reschedule `PLUGIN_READ`
- In `PLUGIN_READ` collect the values, set them on the output, set the sensor off and reset the state to idle.  Only this call to `PLUGIN_READ` must return success = true to indicate there is a new value.

Also take into account that some sensors may need several samples after init to get a stable reading.

So I don't think this is a bug, but more misunderstanding how ESPeasy does things.

uzi18, 2019-04-08:
@TD-er please remember about my CSE chip communication fix, we can also use it here.

BNNorman, 2019-04-08:
Thank you for that explanation. I am using TEN_PER_SECOND to check for a complete packet from the sensor - analyse the packet - store the validated data in temp variables for PLUGIN_READ - schedule another sensor read.  And, Yes the sensor needs to stabilise which is why I am interested in the rule event being raised on the last reading taken (success=true) before the sensor is put to sleep.
I have tried,many times, to find detailed documentation on writing plugins for ESPeasy but all examples I have found are only partially explained. I will look at the BME sensor plugin as you suggested but do you know where I can find a complete plugin writing explanation?
In the template plugin the explanation for PLUGIN_READ is this
```
case PLUGIN_READ:
    {
      //code to be executed to read data
      //It is executed according to the delay configured on the device configuration page, only once
      //after the plugin has read data successfuly, set success and break
      success = true;
      break;
    }
```
It is my understanding that that is what I was doing. IE returning true when the data was valid but not getting the Rule event firing.
Regards

TD-er, 2019-04-08:
Well the documentation is rather outdated I'm afraid.

The main problem with the 'old' method of calling things is that it suggests to do a lot of things in a single loop.
This is blocking behavior and will lead to a lot of issues, lost connections, crashes, etc.

The best strategy is to keep some kind of state to keep track of what stage the plugin is in.
For example:
- idle
- initiate measurement & set state to "waiting for data"
- waiting for measurement data
- Measurement ready (schedule another `PLUGIN_READ`)
- collect data & set state to "idle"
- idle

In general the bulk of this is done in the 10/sec calls.
For example the waiting for data and sometimes send other commands to gather the next sample when different units of measure have to be performed sequential. (like the SHT1x)

So for this the `PLUGIN_READ` call will initiate the measurement when state was idle, or collect the data when the state was ready.

When doing things is these little steps, they all take only msec to execute and thus the system can perform a lot of tasks at the same time.
The "old" way of programming the plugins was to do it all in the `PLUGIN_READ` call, which could sometimes take several seconds to execute.
This results in lost data, lost connections, crashes and nodes that hardly react on external commands.

BNNorman, 2019-04-09:
I don't believe my TEN_PER_SECOND is blocking. If the serial buffer does not contain the 32 characters for a packet it returns immediately. If it does the Checksum and Frame length checks only take a few instructions. If checks pass three values are saved - again only a couple of instructions. On an 80mHz machine that shouldn't take long (but I will attempt a timing check). The fact that every sensor read has correct checksum indicates to me that the new code is working - the plugin I am trying to improve on does not process the received packet immediately (sets a flag and waits for next plugin call) and I observe too many checksum errors as a result - not so with my method. 

Meanwhile, whilst resading _082_GPS.ino I have found there's a rulesProcessing() call. A quick investigation shows I can trigger a rule whenever I want so that solves my problem and therefore this issue can be closed from my point of view.

I really do appreciate all the informative comments - many thanks.

TD-er, 2019-04-09:
The GPS plugin is rather new indeed, so that one could also be used as some kind of reference.
It does however contain a bug. (or more like an unfinished new feature)
It does send values whenever a new position is available, but that's partly caused by the special way how the GPS library does work.
It has a rather clever designed parser that does parse a byte when calling its loop function and when a sentence is finished it will do a bit more.

My intention for this unfinished feature is to be able to return a new position as soon as N meters have been traveled since the last update.

And about the timing... see the timing stats page on the tools tab.

BNNorman, 2019-04-10:
Many thanks for all your comments - the rulesProcessing() call is working just fine for me. The PMSx003 sensor requires a theoretical 30s warm up, after it is woken up, before values become valid and at which point the sensor can be turned off and the latest stored readings taken. The sensor life time is less than 1 year if left on continuously.

I trigger a Rule [sensor name]#Sleep which can now be caught and the values published. Using sleep/wake reduces the Serial traffic enormously as it is only 32 bytes each time and only when the new plugin asks for it. As a result, all the data packets I get have valid checksums.

Especially thanks for the mention of the timing stats. Confucious he say 'Man with nose to grindstone, only see the grind stone.'  I've stepped back from the grindstone now LOL.

Re: _082_GPS.ino I see you have hard coded the event name on line 291+
```
        // Fix status changed, send events.
        String event = curFixStatus ? F("GPS#GotFix") : F("GPS#LostFix");
        rulesProcessing(event);
```
but the plugin still allows the user to name the device. If they are looking for a rule 'myName#GotFix' they will be disappointed - when I first tested using hard coded 'PMSx003#Wakeup' it was ignored (My sensor name was PMS7003) so, in the pluginCall() I changed to something like this :-
```
    String instanceName=ExtraTaskSettings.TaskDeviceName;
   rulesProcessing(instanceName+F("#Wakeup"));
```

closed 2019-04-10 by BNNorman

TD-er, 2019-04-10:
Hmm that's a good point you make here, about the event names.
I think it may deserve its own issue to rename events sent by plugins to allow to differentiate which plugin sends the event. Not all plugins can be used in multiple instances at the same time, but I'm working on that one plugin at a time.