letscontrolit/ESPEasy #2443: Can't connect to hotspot / config.dat write error after upgrade (Release mega-20190409)

FabseN234, 2019-04-12:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

I am using about 15 self soldered boards with ESP8266MOD+BME280. I was running Release "Mega 20190311" til yesterday and decided to upgrade to "Release 20190409". After upgrading (web interface) everything seems fine until I tried to change any setting. I was getting config.dat write errors and nothing was working. I thought it's because of my settings which I load up after flashing a new device so I've tried a factory reset. Now I see the hotspot (ESP_Easy_0) but I can't connect. I've tried this with 3 devices right now with same results.

### Expected behavior

Boot up as usual and let me save settings.
Let me log into the hotspot to configure WiFi settings

### Actual behavior

config.dat write errors after upgrade
No connection to hotspot after factory reset

### Steps to reproduce

1. Device running version 20190311
2. Upgrade firmware to 20190409 per webinterface
3. Do a factory reset
4. Try to connect to hotspot

### System configuration

Hardware:
ESP8266MOD (4M)
BME280 (watterott)
a few resistors
dc/dc converter

ESP Easy version: 
Release mega-20190409
ESP_Easy_mega-20190409_normal_ESP8266_4M.bin

ESP Easy settings/screenshots:

TD-er, 2019-04-12:
Have you tried reverting to the previous build that worked?

FabseN234, 2019-04-15:
> Have you tried reverting to the previous build that worked?

Yes, and it works (2019-03-11). As long as I can reach the webinterface (you know... soldered) I can downgrade and it's working. 

Update: After the weekend I could suddenly connect to the hotspot and configure wifi. Everything was working fine (sensors / mqtt) until hard reboot (cut power off). After the reboot I get again "Error while reading/writing config.dat in 739". I did the configuration manually, so there is no imported config.dat. Reproduced this with 3 samples, 1 of them without attached sensors (only generic system data per mqtt).

TD-er, 2019-04-15:
What size of SPIFFS do you have?
How much free space and are there many files on the filesystem?

I got the feeling your filesystem may be fragmented and there may not be any erased blocks left.

FabseN234, 2019-04-16:
> What size of SPIFFS do you have?
> How much free space and are there many files on the filesystem?
> 
> I got the feeling your filesystem may be fragmented and there may not be any erased blocks left.

The size is 4Mbyte flash and 1Mbyte SPIFFS and there are no additional files on the filesystem. I can't check free space because system info shows 0 (0), see below:

>Firmware
--
Build:⋄ | 20103 - Mega
System Libraries:⋄ | ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support
Git Build:⋄ | mega-20190409
Plugins:⋄ | 47 [Normal]
Build Md5: | 5578052fabd6bbd1a219770ebd75a4e
Md5 check: | passed.
Build Time:⋄ | Apr 9 2019 02:14:13
Binary Filename:⋄ | ESP_Easy_mega-20190409_normal_ESP8266_4M.bin

>ESP board
--
ESP Chip ID: | 646922 (0x9DF0A)
ESP Chip Frequency: | 80 MHz
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 1 daily / 1 boot
Sketch Size: | 788 kB (2280 kB free)
**SPIFFS Size: | 0 kB (0 kB free)**
Page size: | 256
Block size: | 8192
Number of blocks: | 0
Maximum open files: | 5
Maximum path length: | 32

Here is some log output:
> 1709255: FS : Error while reading/writing config.dat in 739
1709257: ADC : Analog value: 784 = 0.745
1709263: FS : Error while reading/writing config.dat in 739
1709272: FS : Error while reading/writing config.dat in 739
1709277: ADC : Analog value: 784 = 0.745
1709284: FS : Error while reading/writing config.dat in 739
1709296: FS : Error while reading/writing config.dat in 739
1709299: SYS : 19.00,16944.00,20.50,-70.00
1709304: FS : Error while reading/writing config.dat in 739
1710404: FS : Error while reading/writing config.dat in 739
1711302: BME280: dew point 5.72C
1711315: BME280 : Address: 0x77
1711315: BME280 : Temperature: 26.68
1711315: BME280 : Humidity: 26.22
1711315: BME280 : Barometric Pressure: 1017.31
1711321: FS : Error while reading/writing config.dat in 739
1712014: WD : Uptime 29 ConnectFailures 112 FreeMem 17328 WiFiStatus 3
1712255: FS : Error while reading/writing config.dat in 739
1712257: ADC : Analog value: 784 = 0.745
1712263: FS : Error while reading/writing config.dat in 739
1712272: FS : Error while reading/writing config.dat in 739
1712274: ADC : Analog value: 784 = 0.745
1712279: FS : Error while reading/writing config.dat in 739
1712293: FS : Error while reading/writing config.dat in 739
1712295: SYS : 19.00,16968.00,20.50,-70.00
1712300: FS : Error while reading/writing config.dat in 739
1714315: FS : Error while reading/writing config.dat in 739
1715255: FS : Error while reading/writing config.dat in 739

TD-er, 2019-04-16:
The reason you cannot save is because there was no SPIFFS mounted.
Which leaves the question, why wasn't it mounted?
- No (successful) SPIFFS format?
- Some other mount error?
- Someway the build was wrong, using incorrect flash layout definition?

If you OTA again (now the SPIFFS was created and formatted), is it then possible to use the file system?
That could be useful in determine what may have been the issue here.

FabseN234, 2019-04-16:
Behavior with 20190409 and also with 20190416:
It is possible to use the FS after OTA update. I can also set up deep sleep with mqtt. After power cut off and reconnect I get again these errors (20190416):

>179804: FS : Error while reading/writing config.dat in 781
179807: BME280 : Address: 0x77
179807: BME280 : Temperature: 23.74
179807: BME280 : Humidity: 28.19
179807: BME280 : Barometric Pressure: 1016.40
179812: FS : Error while reading/writing config.dat in 781
181690: FS : Error while reading/writing config.dat in 781
181693: ADC : Analog value: 770 = 0.732
181698: FS : Error while reading/writing config.dat in 781
181712: FS : Error while reading/writing config.dat in 781
181715: ADC : Analog value: 771 = 0.733
181719: FS : Error while reading/writing config.dat in 781
181729: FS : Error while reading/writing config.dat in 781
181733: SYS : 19.00,16920.00,22.40,-68.00
181738: FS : Error while reading/writing config.dat in 781

BINs I was using:
ESP_Easy_mega-20190311_normal_ESP8266_4M.bin (working, also after power off)
ESP_Easy_mega-20190409_normal_ESP8266_4M.bin (working till powered off)
ESP_Easy_mega-20190416_normal_ESP8266_4M.bin (working till powered off)

TD-er, 2019-04-16:
And if you use the 20190311 build, make some adjustments to the settings, power off and then OTA upgrade to 20190416.
Does that work?
What does the sysinfo page of 20190416 show about the SPIFFS size? (before trying to change anything)

FabseN234, 2019-04-16:
> And if you use the 20190311 build, make some adjustments to the settings, power off and then OTA upgrade to 20190416.
> Does that work?
> What does the sysinfo page of 20190416 show about the SPIFFS size? (before trying to change anything)

Looks like it's mounted properly:
>Storage
--
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DIO
Flash Writes: | 0 daily / 0 boot
Sketch Size: | 789 kB (2280 kB free)
SPIFFS Size: | 934 kB (844 kB free)
Page size: | 256
Block size: | 8192
Number of blocks: | 116
Maximum open files: | 5
Maximum path length: | 32

Without changing anything I can also cut off the power and reboot. SPIFFS size is displayed properly. I can also change anything I want and cut off the power. I realized this problem occurs while "interrupting" deep sleep. So I cut off the power during deep sleep, power up again and everything seems fine (SPIFFS size OK). **If I make any change now I get errors and the SPIFFS size swaps to zero (0). It seems this behavior only appears after interrupting deep sleep.**

TD-er, 2019-04-16:
OK, that's an interesting find.
I wonder why that was not happening before.
As far as I know, I did not change anything that could trigger this behavior.

FabseN234, 2019-04-26:
Tested with 04/25. Same behavior :( Downgraded to 03/11 and everything is fine. 

Reproduce:
-Upgrade to 04/25
-Set up sensor and MQTT
-Set up deepsleep
-Wait till device is in deepsleep
-Power off and power on again
-Since now "Error writing to config.dat" while changing any settings

PS: Forget about the hotspot problem, it was related to my phone/wifi.

holgerschillack, 2019-05-03:
I experienced the same problem, when adding ADS1115 Analog Devices. Sometimes when adding the 3rd or 4th devices and changing their settings, I get a config.dat writing error and the file seems to be corrupted. After power loss the ESP is never reachable again.
From times to times it is working, but additional changes later are not possible due to the config.dat error probability (when crashed, one need to reflash the ESP, which is annoying when the ESP is somewhere digged in)

I think, it's maybe a problem when editing items/pages/devices, where a lot of variables need to be saved at the save-button click. So 'the storage-controller' gets overwhelmed and throws an error and leaves the config.dat file corrupted.

TD-er, 2019-05-03:
What size SPIFFS do you have?
Is it on a 1M flash/128k SPIFFS config?

holgerschillack, 2019-05-03:
Yes it's an ESP8285 with 1MB flash size.

`
ESP board
--
ESP Chip ID | 8602393 (0x834319)
ESP Chip Freq: | 80 MHz
Storage
Flash Chip ID | Vendor: 0x51 Device: 0x4014
Flash Chip Real Size: | 1024 kB
Flash IDE Size: | 1024 kB
Flash IDE speed: | 40 MHz
Flash IDE mode: | DOUT
Flash Writes | 0 daily / 0 boot
Sketch Size | 729 kB (144 kB free)
SPIFFS Size | 110 kB (36 kB free)

`

dliebchen, 2019-05-03:
Sure March 2018? If you mean 2019 than the last release working for me is 2019/03/11

holgerschillack, 2019-05-04:
Yes I've also tested the version from pi day 18, 14/3/18. I get the same error when playing around with ADS1115 devices and changing their settings.

TD-er, 2019-05-04:
I guess there are no free blocks available on the filesystem.
With 128k (effective 112k) SPIFFS, the settings file is actually too big to be really useful. (64k)
I guess we can trim the settings file a bit (last 16 - 20k are zeroes) to make it easier to find a free block on the SPIFFS.
The last core 2.6.0 builds do have some garbage collection build in, which is being called at least at boot and some other moments.
But I guess that will not help a lot with so little free space.

tonhuisman, 2023-09-03:
This seems to be solved, so can be closed.

closed 2023-09-03 by TD-er