letscontrolit/ESPEasy #2444: SHT1x Sensors works too good :)

Sasch600xt, 2019-04-12:
Hello TDer,

you made a change for the SHT sensors in the past so multiple sensors will work.
Thank you, it works :)
But i have a sensor with 2m cable and a sensor with 30 meters cable.
With the 2 meter cable no problem, but the 30 meters cable sensor has trouble because of a "too fast clock speed"
With slower clockspeed i probably could save a cable driver.

So question is, could you make the clockspeed somehow as option parameter in the plugin ?
So i could change the speed by my own ?
Right now that would help me a lot :)

Thank you
Sascha

TD-er, 2019-04-12:
30m cable does impose some resistance and signal loss.
But also it may act as an antenna.
What kind of cable do you use? For longer cable I would suggest to use some kind of twisted cable to reduce the noise you may pick up.
And sometimes it helps to add pull-up or pull-down resistors (depending on how the bus is operated) to improve the signal.

Sasch600xt, 2019-04-12:
Hello TDer,

you are totally right with all answeres.
for sure i use cat6 cable for sensors, and pullup aswell.
Today i had time to check all conditions.
So with a slower clock signal we could use sensors at short and long cable without using an external active signal driver.
It would help me a lot right now to have such an option.

Thank you TDer :)
Sascha

TD-er, 2019-04-13:
This is mentioned in the [datasheet](https://www.sparkfun.com/datasheets/Sensors/SHT1x_datasheet.pdf) (page 5):

> To avoid signal contention the microcontroller must only
> drive DATA low. An external pull-up resistor (e.g. 10kΩ) is
> required to pull the signal high – it should be noted that
> pull-up resistors may be included in I/O circuits of
> microcontrollers. See Table 2 for detailed I/O characteristic
> of the sensor. 

Not sure what the current effective clock rate is.
But if the pull-up resistors are not used, I'm afraid the change of timing may not be enough.

TD-er, 2019-04-14:
I looked into the cut off frequency for the cable you mentioned.
Let's assume 50 pF/m (looked at 2 brands and they specify 46 pF/m) and keep the rest at default.
Enter it [in this online calculator](http://www.sengpielaudio.com/calculator-cable.htm) and we're at 5+ MHz for 1 meter and 176 kHz for 30m.
So indeed it is quite a difference here and not only due to pull-up resistors.

TD-er, 2019-04-14:
Can you test [this test build](https://www.dropbox.com/s/s82w1xu69y3co4w/normal_ESP8266_4m_issue2444.rar?dl=0) ?
I added a 10 microsecond delay between clock ticks.
If that does work, then we can make it a configurable setting.

Sasch600xt, 2019-04-23:
could there be a timing problem ?

One SHT is no Problem.

But with 3 SHT sensors and a common clock port i have some issues while i poll sonsor data every 60 seconds. Sometimes the sonsors dont response or at least it is not reconized from controller.

Thanks
Sascha

TD-er, 2019-04-23:
Well if they share a clock line, I guess both their data lines should have equal length or else you may introduce some issues with data signals not in sync with the clock.

If you have them all connected, but only poll 1 sensor, does that work separately for all sensors?
You could then set the interval of the plugins quite high, but call them from the rules with quite some time between them. (at least not at the same time)

Sasch600xt, 2019-04-24:
that sounds intresting with rules.
could you show me an example how i can pull sensordata with rules ?
I can set interval in plugin to "0". So only by rules every minute sensor data should be pulled.

If that works could you merge a automatic delay for multiple SHT sensors in upcomming build then ?
By the way, is in the newest build the 10 or 15 usec delay for clock of SHT already merged ?

Thank you so much for your help :)

Sascha

TD-er, 2019-04-24:
If you set the interval to '0', it will be using the default which is 60 seconds.

To run a plugin from the rules, see [TaskRun](https://espeasy.readthedocs.io/en/latest/Reference/Command.html?highlight=TaskRun)

Sasch600xt, 2019-05-01:
For now i spent a separate clock port for every SHT sensor which works perfect.
So no common clock for the moment.
The adjustable delay for longer cables is awesome and works great in my enviroment.
Thank you so much for this change ! :)

Have a great day
Sascha

closed 2019-05-01 by TD-er