letscontrolit/ESPEasy #2454: Timer - 1sec isn't 1sec

BuuBaa, 2019-04-24:
Hi, 
I have a problem with the sequence of relays in rules. I have to do 20 lamps (currently 8 for tests), which must light up one after the other with an interval of 1 sec.

Unfortunately, using rules, the timer does not last 1 sec as written, but 2 sec, and instead of 3 sec is 4 sec. 
Where to look for this extra 1 sec? How to get rid of it?
I reinstalled the latest version, I checked the connections, I have no idea anymore...

Thank you for your help.

I'm pasting the logs and rules below.

**RULES**
```
On wlaczkolejno do
 pcfgpio,1,1
   timerSet,1,1
endon

On Rules#Timer=1 do
pcfgpio,2,1
   timerSet,2,1
endon

On Rules#Timer=2 do
pcfgpio,3,1
   timerSet,3,1
endon

On Rules#Timer=3 do
pcfgpio,4,1
   timerSet,4,1
endon

On Rules#Timer=4 do
pcfgpio,5,1
   timerSet,5,1
endon

On Rules#Timer=5 do
pcfgpio,6,1
   timerSet,6,1
endon

On Rules#Timer=6 do
pcfgpio,7,1
pcfgpio,8,1
   timerSet,7,1
endon

On Rules#Timer=7 do
 pcfgpio,1,0
 pcfgpio,2,0
 pcfgpio,3,0
 pcfgpio,4,0
 pcfgpio,7,0
 pcfgpio,8,0
   timerSet,8,3
endon

On Rules#Timer=8 do
 pcfgpio,5,0
 pcfgpio,6,0
endon
```

**LOGS**

```
464862: HTTP: event,wlaczkolejno
464937: EVENT: wlaczkolejno
464982: ACT : pcfgpio,1,1
464986: PCF : GPIO OUTPUT 1 Set to 1
464989: ACT : timerSet,1,1
464994: Command: timerset
466700: EVENT: Rules#Timer=1
466748: ACT : pcfgpio,2,1
466752: PCF : GPIO OUTPUT 2 Set to 1
466755: ACT : timerSet,2,1
466761: Command: timerset
468700: EVENT: Rules#Timer=2
468753: ACT : pcfgpio,3,1
468757: PCF : GPIO OUTPUT 3 Set to 1
468760: ACT : timerSet,3,1
468766: Command: timerset
470700: EVENT: Rules#Timer=3
470757: ACT : pcfgpio,4,1
470761: PCF : GPIO OUTPUT 4 Set to 1
470763: ACT : timerSet,4,1
470769: Command: timerset
472700: EVENT: Rules#Timer=4
472762: ACT : pcfgpio,5,1
472766: PCF : GPIO OUTPUT 5 Set to 1
472769: ACT : timerSet,5,1
472774: Command: timerset
474700: EVENT: Rules#Timer=5
474765: ACT : pcfgpio,6,1
474769: PCF : GPIO OUTPUT 6 Set to 1
474771: ACT : timerSet,6,1
474777: Command: timerset
476700: EVENT: Rules#Timer=6
476770: ACT : pcfgpio,7,1
476774: PCF : GPIO OUTPUT 7 Set to 1
476776: ACT : pcfgpio,8,1
476780: PCF : GPIO OUTPUT 8 Set to 1
476782: ACT : timerSet,7,1
476788: Command: timerset
478774: ACT : pcfgpio,1,0
478778: PCF : GPIO OUTPUT 1 Set to 0
478781: ACT : pcfgpio,2,0
478785: PCF : GPIO OUTPUT 2 Set to 0
478787: ACT : pcfgpio,3,0
478790: PCF : GPIO OUTPUT 3 Set to 0
478793: ACT : pcfgpio,4,0
478796: PCF : GPIO OUTPUT 4 Set to 0
478799: ACT : pcfgpio,7,0
478802: PCF : GPIO OUTPUT 7 Set to 0
478805: ACT : pcfgpio,8,0
478809: PCF : GPIO OUTPUT 8 Set to 0
478811: ACT : timerSet,8,3
478817: Command: timerset
482256: WD : Uptime 8 ConnectFailures 0 FreeMem 18088 WiFiStatus 3
482700: EVENT: Rules#Timer=8
482780: ACT : pcfgpio,5,0
482784: PCF : GPIO OUTPUT 5 Set to 0
482787: ACT : pcfgpio,6,0
482790: PCF : GPIO OUTPUT 6 Set to 0
512255: WD : Uptime 9 ConnectFailures 0 FreeMem 18112 WiFiStatus 3
```

uzi18, 2019-04-24:
try this: 

```
On wlaczkolejno do
pcfgpio,1,1
delay(1000)
pcfgpio,2,1
delay(1000)
pcfgpio,3,1
delay(1000)
pcfgpio,4,1
delay(1000)
pcfgpio,5,1
delay(1000)
pcfgpio,6,1
delay(1000)
pcfgpio,7,1
delay(1000)
pcfgpio,8,1
delay(1000)
pcfgpio,1,0
pcfgpio,2,0
pcfgpio,3,0
pcfgpio,4,0
pcfgpio,5,0
pcfgpio,6,0
pcfgpio,7,0
pcfgpio,8,0
endon

```

BuuBaa, 2019-04-24:
Thanks man! 
I had a problem with delay and I thought that it does not work anymore in the newest versions.
A small fix to your comment and everything works as it should.

_delay(1000)_ does not work, the correct version is _delay,1000_

Thanks for your help, you gave me a good tip ;)

uzi18, 2019-04-24:
Racja zapomnialem ze to nie jezyk "C"

correct, just thinking in C too much =)

BuuBaa, 2019-04-24:
Też się na to "nabrałem" przy początkowych próbach, później wyczytałem, że delay nie działa od którejś tam z nowszych wersji więc zacząłem szukać rozwiązania gdzie indziej ;)

uzi18, 2019-04-24:
staramy sie aby wszystko działało, gdzie o tym delay wyczytales?

BuuBaa, 2019-04-24:
Na którymś z anglojęzycznych forum (przeszukałem pół internetu w poszukiwaniu przyczyny opóźnień). 
Niestety nie znajdę źródła tej błędnej informacji.

uzi18, 2019-04-24:
timerset sluzy do opozniej z tolerancja 1s.
delay jest do opoznien krotszych

TD-er, 2019-04-25:
Using the delay command is an option, but it will cause other issues.
It will prevent other tasks to be executed.

The reason it does create an offset is the execution time of the rules itself.
If the delay command is suitable for this use case, please use it, but be aware you may run into other issues.

I think it makes more sense to set the timers all at once and just let a timer set itself.
It will then still not be at an exact interval of 1sec, but the offset will be less.

nonflammable, 2019-04-27:
> 
> 
> Też się na to "nabrałem" przy początkowych próbach, później wyczytałem, że delay nie działa od którejś tam z nowszych wersji więc zacząłem szukać rozwiązania gdzie indziej ;)

pisałem o tym na grupie FB
problem był zgłoszony tutaj:
https://github.com/letscontrolit/ESPEasy/issues/1654
wygląda na to, że gdzieś w międzyczasie zostało to "naprawione" ale składnia zmieniła się z
timer 1000
na 
timer,1000

TD-er, 2019-04-28:
Please post in English, or else I have to keep using translations.
I am not sure if `, ` (<comma><space>) will have the same result as just using a `,`
The argument separator can be either a space or a comma.

uzi18, 2019-04-28:
kurcze na FB, nie kojarze :)

don't remember anything like this on FB

sorry @TD-er

closed 2023-09-03 by TD-er