letscontrolit/ESPEasy #2455: feature request: forecast from BME280 to MQTT message

henkiejan1, 2019-04-24:
What should be more great than have your own weather forecast from your own weather station. The BME280 is very precious but it can't give forecast information in MQTT. It should be great when implemented and see your own weather forecast. :)

TD-er, 2019-04-25:
How should the BME280 make a forecast?
It can only detect the current situation.

So how do you imagine this forecast should be done?

SargonofAssyria, 2019-04-25:
This is my forecast code in NodeRed to get the value into Domoticz.
So simple is a forecast.

`if (pres < 966) {
    forecast = "4";
} else if (pres < 993) {
    forecast = "3";
} else if (pres < 1007) {
    forecast = "2";
} else if (pres < 1013) {
    forecast = "3";
} else if (pres < 1033) {
    forecast = "0";
} else {
    forecast = "1";
}`

henkiejan1, 2019-04-25:
> 
> 
> How should the BME280 make a forecast?
> It can only detect the current situation.
> 
> So how do you imagine this forecast should be done?

What also SargonoAssyria say. You can with the trend of the atmospheric pressure make a forecast. I could better name it barometer i think... ;)
[image]

So in my opnion it should be a great feature! :)

TD-er, 2019-04-25:
Ah OK, I get it :)

SargonofAssyria, 2019-04-28:
There is a small glitch in my code from above, this is the correct code:

if (pres < 966) {
forecast = "4";
} else if (pres < 993) {
forecast = "3";
} else if (pres < 1007) {
forecast = "2";
} else if (pres < 1013) {
forecast = "1"; // <-- 3 should be 1
} else if (pres < 1033) {
forecast = "0";
} else {
forecast = "5"; // Unknown
}

melwinek, 2019-05-04:
I once reported it.
It's more accurate here.
https://github.com/letscontrolit/ESPEasy/issues/1558

SargonofAssyria, 2019-05-04:
I know, but than you have to preserve previous barometer values.

melwinek, 2019-05-04:
Yes. I believe that @TD-er  will do it :)

henkiejan1, 2019-05-04:
It should be great if it is added quickly :) Is there a simple way to add it by myself via webinterface or something?

TD-er, 2019-05-05:
@henkiejan1 Not as far as I know.
Even when implemented in rules, it would be quite some hassle.

henkiejan1, 2020-01-11:
Maybe a nice detail. When in hook up a BME280 to the pins of my OrangePi i get in Domoticz also the forecast. `[url=https://tweakers.net/ext/f/keBc9BCEFKuMfmJeu8jkYWp8/full.jpg][img=100,75,,,"bme280 forecast"]https://tweakers.net/ext/f/keBc9BCEFKuMfmJeu8jkYWp8/thumb.jpg[/img][/url]`

TD-er, 2020-01-11:
Can you see what data is sent to Domoticz via this orange pi?
I can understand if it does keep track of trends in the air pressure, but I doubt things like "bewolkt" can be determined by that.

henkiejan1, 2020-01-12:
I will check. I think i need to enable debug mode or something. It´s most of time really accurate.

henkiejan1, 2020-01-12:
> Can you see what data is sent to Domoticz via this orange pi?
> I can understand if it does keep track of trends in the air pressure, but I doubt things like "bewolkt" can be determined by that.

If i am correct is the sketch based on this MySensors sketch. https://forum.mysensors.org/post/76246  I will try to figure out some logs

rjblake, 2021-01-19:
Was this feature ever added?