letscontrolit/ESPEasy #2456: Feature request: TCP UART2UART connection using 2 EasyESPs

sebastianheyn, 2019-04-25:
Hi,

I want to ask for an extention of the existing Serial2Net implementation.

Example use case:

2 ESPEasy are connected to the same subnet. 
Each ESP has a microcontroller connected to it to perform some task.
Normally, those 2 microcontrollers would  communicate using UART only.

If the controllers want to talk to each other they can establish a connection
The ESP can determine a command from data, when the string starts with "ESPEasy,"

Controller1 -> ESP1 "ESPEasy,192.168.123.2,12345,COMMAND_TO_SEND_VIA_UART"
ESP1 <-> ESP2
ESP2 -> Controller 2 "COMMAND_TO_SEND_VIA_UART"
Controller 2 -> ESP2 "REPLY_TO_COMMAND"
ESP1 <-> ESP2 (still open tcp channel)
ESP1 -> Controller1 "REPLY_TO_COMMAND"
Controller1 -> ESP1 "ESPEasy,0" (Disconnect)

Grovkillen, 2019-04-25:
How much would you pay for it ✌️😎

TD-er, 2019-04-25:
I'm not sure if this should be a "plugin" or a "controller" or some core level command.
It contains some (binary?) data, so it cannot be a plugin and controllers are not really bi-directional right now.
This means it should be a core level command.
It is possible to send commands to other nodes, via the ESPeasy p2p protocol, but that's not yet used very much. (and has room for improvement)
Another thing to consider is the way how data is fed to a command. That's mainly a subset of ASCII, so if the commands are using characters/bytes not in this allowed subset, we have another rather big change ahead.
Next thing is the response time between command and reply. ESPeasy is not a real time environment, so it may take a while to get an answer and the sender must be able to handle this longer latency.
This raises the next question, when is the response finished?
Also we currently have no commands that can receive data, so either the "slave" side must trigger a new command or we must extend the ESPeasy p2p protocol to these kind of messages.

So I guess that's quite a lot to be considered and I think some of these will not be fixable.

You could have a look at the existing ser2net plugin, to see if that can be used in your setup.

Another project which is only intended to make a serial to network bridge, is [ESP-LINK: Wifi-Serial Bridge w/REST&MQTT](https://github.com/jeelabs/esp-link)
I guess that's probably the better choice right now, since you really must take into account some real time aspects when dealing with command/reply communication via UART. And ESPeasy simply isn't a real time environment.

uzi18, 2019-04-25:
it should be more like option to set host and port to connect for ser2net

but why you need this as commands on serial port?

uzi18, 2019-04-25:
esp-link also does not support such serial. connection over tcp natively

but you can introduce there your idea of connection to another esp-link or espeasy with your microcontroller program, will have control of connection/disconnection by ELClient library

sebastianheyn, 2019-04-25:
To be honest, the existing serial to net already does half of the mentioned above, if I understood correctly. However, I can only go server<->espeasy 
so whats lacking is a espeasy<->espeasy extention. 

Timings are not critical. The existing connection can be closed if the timeout is over.
And I would not even consider sending byte based. the espeasy would have a line buffer and send the received before \r\n one by one (including the \r\n) 

Usecase: I'm building a dimmer right now and the frontend has leds to indicate the dimmer level, so I must get feedback from the dimmer itself. 

Without feedback, I can for example use sendtohttp and cmd=serialsend commands in combination to send data from one esp to the other, but the return path is unclear. 

I love ESPEasy and how you can support so many sensors, since this was the plan, to integrate room sensors in this one node. 

I have already downloaded the latest AT firmware *vomit smilie face* since it is able to do somehow what I'm trying to do, but other than that its a waste of cpu cycles.

sebastianheyn, 2019-04-25:
@Grovkillen 

depends on the type of implementation.

@uzi18 Actually this is a good idea from you: Set the target IP and Port in the ser2net plugin. Whenever the UART received something, the connection is established ....