letscontrolit/ESPEasy #2457: SDS011 in lib jkSDS011.cpp is a small bug.

SargonofAssyria, 2019-04-25:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [x] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x] given a list of active plugins or controllers when applicable.
- [x] filled out all applicable fields below.

### Steps already tried...
- [x] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

SDS011 has 2 reading modes.
1 mode is used by ESPEasy (the default) and 1 mode is used by Tasmota.
My system was working fine with the SDS011 under ESPEasy
I loaded an esp8266 with Tasmota, was not satisfied with Tasmota, and want to go back to ESPEasy. Done that and the SDS011 was working fine, but there came no output anymore.
I found out that Tasmota changed the reading mode and ESPEasy did not change it back to the original default mode.
In the lib file jkSDS011.cpp in the function SetWorkingPeriod after "SendCommand(8, 1, minutes);" this should be added: "SendCommand(2, 1, 0);"
This puts the SDS011 back in the original reading mode.

TD-er, 2019-04-25:
Hmm, that will set the sensor in contiguous reading mode and thus burn up the laser in a year. (not sure what the first '2' does)

https://github.com/letscontrolit/ESPEasy/blob/5713900d0faf86d3fbd77d77f9ba218a2fff293d/lib/SerialDevices/jkSDS011.cpp#L85-L96

I will look into what @arendst is changing in the sensor, to see why it isn't working after Tasmota was running on it.

SargonofAssyria, 2019-04-26:
The two modes are polling or get info after measurement.
Tasmota sets the Working period and polls every 3 seconds for data (report mode query)
ESPEasy sets the Working period and wait for the SDS011 for data (report mode active)
Tasmota sets the report mode **query**, but ESPEasy does not set it to report mode **active**, becasue it is the default mode from the device.
So it should be set (back to default) to get it running again after Tasmota.

TD-er, 2019-04-26:
OK, now I get it, so those should be set in the same scope as when we set the internal timer. (thus adding `{}` )

SargonofAssyria, 2019-04-26:
Yep. Just to make sure that the right report mode is set.

TD-er, 2019-04-26:
I will test it here, to make sure it doesn't break existing installations.

tonhuisman, 2023-09-03:
This seems to be solved, and can be closed.

TD-er, 2023-09-03:
Not sure if it has been fixed...