letscontrolit/ESPEasy #2458: unit name with %variable% part ?

rudiratlos, 2019-04-27:
how can i set the unitname to a partly variable name. like ESP_%mac_int%
should result in ESP_12345 or ESP_98765 (the number _xxxx) should depand on the mac adr.
Idea is, to have a common firmware image with predefined config data. but the node name shoud be unique in the network.

rudiratlos, 2019-04-27:
unique node name with hex part like ESP_55EE7F whould more conveniant

rudiratlos, 2019-04-27:
just discovered, that %sysname% and nodename are not the same. Is there a variable for %hostname%

What's / why is there a difference?

TD-er, 2019-04-28:
You can see all the existing variables via _Tools => System Variables_.

The node name will have the unit number appended, unless you disable it, via _Config => Append Unit Number to hostname_

There is currently only a setting for formatting the name using the MAC address for use in MQTT, not for the hostname.
I can see why it can be useful to use some formatting like this on the hostname.

chemmex, 2019-04-30:
I use rules:

On System#Boot do
  name ESP_%mac_int%
endon

rudiratlos, 2019-05-04:
is there a way to set a unique unit number also ?
e.g.
On System#Boot do
  name ESP_%mac_int%
  unit %mac_int%
endon

Grovkillen, 2019-05-04:
I guess the %ip4% is always unique.

rudiratlos, 2019-05-04:
%ip4% is not available On System#Boot

rudiratlos, 2019-05-04:
this works:

On System#Boot do
  name ESP_%mac_int%
  unit 255
  timerSet,1,5
endon

On Rules#Timer=1 do
  unit %ip4%
endon

Grovkillen, 2019-05-04:
Remember to do a `save` if you want it to be persistent. But I guess since
you're setting it at boot it's almost persistent.

Grovkillen, 2019-05-10:
Can we close this?

rudiratlos, 2019-05-10:
yes thank you

closed 2019-05-10 by Grovkillen