letscontrolit/ESPEasy #2461: Fade out when use pwm

Naimadnap, 2019-05-07:
I have flashed ESPEasy_mega-20190225_normal_ESP8266_1M.bin version to my wemosD1. I would to have fade out effect but the LED goes to 0 and ramp up to setted value.

Example rule

on button#d5=1 do
Pwm, 12,1000,1000
Pwm, 12,100,1000
endon

Led goes from 0 to 1023 in 1sec, turn off, goes from 0 to 100 in 1sec instead of dim from 1000 to 100 in 1sec

sebastianheyn, 2019-05-08:
https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6592

uzi18, 2019-05-09:
@TD-er it looks like globalMapPortStatus storing/fetching problem

TD-er, 2019-05-09:
> @sebastianheyn so it's a bug, know bug or it's a feature

yes ;)

@uzi18 Thanks for narrowing it down a bit 👍

ledfreak3d, 2019-08-04:
when will this be fixed its highly anoying

TD-er, 2019-08-04:
Thanks for kicking it. I will now look at the code to see if there's something obviously wrong.

ledfreak3d, 2019-08-04:
Geen probleem

Maakt al me lampen in huis in een keer ontbruikbaar lol

TD-er, 2019-08-04:
Altijd goed idee om ineens alle nodes te upgraden ;)

ledfreak3d, 2019-08-04:
Yup inderdaad alles weer terug naar r120 lol maar das wel heel oud

TD-er, 2019-08-04:
I found what's the bug here.
In short, a while ago there was an update to create a global pin state buffer to monitor the pin states.
But this one is not really suitable to keep the values for PWM.
On top of that we're back at the ever return issue about the p001_switch plugin which has grown into a totally not maintainable complex beast.

In the end we will split the numerous tasks of the switch plugin into separate ones.
I will try to make a quick fix here for now.

ledfreak3d, 2019-08-04:
Thanks Mutch appreciated

closed 2019-08-04 by TD-er

ledfreak3d, 2019-08-04:
Thanks Gijs

 

You are da bomb all working perfectly again just compiled the latest version

 

Great work

PR #2529 (2019-08-04): [Feature] Serial Proxy reading strings from serial
These Strings will then be available to send to the controllers supporting the string value.
Tested with MQTT controllers.

Fixes #2461
commit 813d5fa6 (2019-08-04): [PWM] Fade always starts at 0, not the last set value

Fixes #2461