letscontrolit/ESPEasy #2466: ESP-12F wifi issue

rds76, 2019-05-12:
I bought 2pcs of ESP-12F from ali, both wont work with latest fw (actually with any fw), seems wifi wont turn on. I use AMS1117 power supply, all required pullups and pulldowns on module.
 Any workaround for this?
```

INIT : Booting version: mega-20190511 (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
109 : INIT : Free RAM:29360
110 : INIT : Cold Boot - Restart Reason: External System
112 : FS   : Mounting...
137 : FS   : Mount successful, used 75802 bytes of 957314
486 : CRC  : program checksum       ...OK
497 : CRC  : SecuritySettings CRC   ...OK 
605 : INIT : Free RAM:26560
607 : INIT : I2C
607 : INIT : SPI not enabled
621 : INFO : Plugins: 47 [Normal] (ESP82xx Core 2_4_2, NONOS SDK 2.2.1(cfd48f3), LWIP: 2.0.3 PUYA support)
624 : mode : sta(b4:e6:2d:39:2a:2f)
add if0
WIFI : Set WiFi to STA
657 : WIFI : Connecting XXX attempt #0
659 : WIFI : Not configured in Station Mode!!: XXX
1994 : WD   : Uptime 0 ConnectFailures 0 FreeMem 24632 WiFiStatus 0
scandone
3654 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2894 ms
4440 : WIFI : Connecting XXX attempt #1
4442 : WIFI : Not configured in Station Mode!!: XXX
scandone
7436 : WIFI : Disconnected! Reason: '(201) No AP found' Connected for 2893 ms
7445 : WIFI : Connecting XXX attempt #2
7446 : WIFI : Not configured in Station Mode!!: XXX
>wifiscan
6605 : Command: wifiscan
WIFI : SSID Scan start
scandone
scandone
WIFI : No networks found

Ok
```

On AP mode it wont work as well.

TD-er, 2019-05-12:
Did you also test other firmware versions, like a build from begin 2018? (before event based wifi and using an older core version)
Just to test if the wifi might work?

ShardanX, 2019-05-12:
I've running an ESP-07S on Version ESP_Easy_mega-20190508_normal_ESP8266_4M.
It's working for some hours now.
As the ESP-07S is an ESP-12F without printed antenna it should count ;)

rds76, 2019-05-12:
```

INIT : Booting version: v2.0.0-dev12
INIT : Cold Boot
FS   : Mounting...
FS   : Mount successful, used 75802 bytes of 957314
INIT : Free RAM:26056
INIT : I2C
INIT : SPI not enabled
WIFI : AP Mode disabled
WIFI : AP Mode ssid will be ESP_Easy_0 with address 192.168.4.1
WIFI : Could not connect to AP!
WIFI : AP Mode enabled
WD   : Uptime 0 ConnectFailures 0 FreeMem 20736
WD   : Uptime 1 ConnectFailures 0 FreeMem 20840
>wifiscan
WIFI : SSID Scan start
WIFI : No networks found

Ok
```

TD-er, 2019-05-12:
Can you post a picture of the board?
And perhaps some schematic you're using if you made the board yourself

rds76, 2019-05-12:
Tried to directly flash espressif AT SDK 2.2.1
[image]
```
ready
AT+GMR
AT version:1.6.2.0(Apr 13 2018 11:10:59)
SDK version:2.2.1(6ab97e9)
compile time:Jun  7 2018 19:34:26
Bin version(Wroom 02):1.6.2
OK
AT+CWMODE=3

OK
AT+CWLAP

OK
AT+RST

OK

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 816, room 16 
tail 0
chksum 0x8d
load 0x3ffe8000, len 788, room 8 
tail 12
chksum 0xcf
ho 0 tail 12 room 4
load 0x3ffe8314, len 288, room 12 
tail 4
chksum 0xcf
csum 0xcf

2nd boot version : 1.2
  SPI Speed      : 40MHz
  SPI Mode       : DIO
  SPI Flash Size : 32Mbit
jump to run user1

```
I'm using exactly same board and wiring https://www.instructables.com/id/ESP-12F-ESP8266-Module-Connection-Test/

TD-er, 2019-05-13:
Well, ESPeasy did seem to be doing something, so I guess it wasn't a flash issue.
What you can try is to completely erase the flash and try a known to be working nightly build.

rds76, 2019-05-13:
Completly erased with blank4MB.bin already, also tried erase from ESP Download tool.
Yes I'm using resistors, the unit is solderd on adapter with resistors for EN and GPIO15 onboard. Actually it goes in flash mode pulling down GPIO0 and boots to normal afterwards. It just wont to connect/scan wifi

TD-er, 2019-05-14:
You may also try to get some measurements on the Vcc line.
Just to make sure the voltage doesn't drop too much.
Maybe even add a capacitor (10 - 100 nF) over the Vcc and GND.
During startup, the ESP will perform a RF calibration and that takes quite a bit of current (upto 500 mA).
Whatever voltage is seen during calibration will be used to base its calibration on.

These calibration values are stored in flash, so that's why I also suggested to clear the entire flash. (stored in a place where ESPeasy doesn't do anything on the flash)

rds76, 2019-05-14:
Vcc is solid, capacitor is there already. Tried also big 1mF cap on Vcc line to support high pikes current demand. Using same setup with many of ESP01 with no issues.
Flash has been cleared with 4MBblank.bin file before.

TD-er, 2019-05-14:
I guess maybe you should also try on the forum then, but I guess there is something wrong with the hardware since even other (older) builds won't work.

You can try to set the WiFi credentials using the ESPeasy flasher included in the download zip file.
Just to see if it will connect with properly set WiFi credentials.

Since this doesn't seem to be a software issue, I will close this one for now.

closed 2019-05-14 by TD-er