letscontrolit/ESPEasy #2471: zigbee extender feature

pilot1981, 2019-05-15:
### I have...
all type of ESP

### Steps already tried...
searched on goole and on let's control it forum

### Summarize of the problem/feature request
I would use this firmware to extend zigbee signal in outdoor...about for 30 meters; I want use 220v power supply in my grade to activate this extender.

### Expected behavior
extend zigbee signal of Aqara zigbee network in outside of my house

### Actual behavior
no any other solution

### Steps to reproduce
NA

### System configuration
ESP

### Rules or log data
NA

TD-er, 2019-05-15:
At this moment nothing regarding zigbee is supported in ESPeasy.
What module do you intend to use for Zigbee?

pilot1981, 2019-05-15:
USB Zigbee (CC2531 USB sniffer)

TD-er, 2019-05-15:
How is that going to be interfaced to the ESP8266?

papperone, 2019-05-17:
what is the point? the CC2531 runs his own firmware and it's comepltely self sufficient when run as router (I do have 2 of them just plugged to a USB Phoen wall charger...)

tonhuisman, 2022-05-07:
This doesn't seem related to ESPEasy, so should probably be closed.

For connecting Zigbee to MQTT, there is https://www.zigbee2mqtt.io/ and ESPEasy's plugin `P037 - MQTT Import` can take care of translating any non-numeric values into something numerical to be processed.

TD-er, 2022-05-07:
The specific project you linked can indeed be used along with the very recent changes of ESPEasy.
N.B. there is a project for the Sonoff units with Zigbee and right now you can only use it with Tasmota. So that one for sure could need ESPEasy support ;)

closed 2022-05-07 by TD-er